SmingHub/Sming #1975: Default MQTT Payload Parser can't handle empty messages

fooker, 2019-11-20:
MQTT supports sending messages without a payload (the remaining bytes after the variable headers are zero). If such a message is received by the `MqttClient` and processed by the default `MqttPayloadParser`, the client dies and does not accept any further messages.

At a first sight, it looks like `malloc` is called with `message->publish.content.length` which is `0` in such a case. Thus leading `malloc` to return `nullptr` and the following sanity-check to fail.

slaff, 2019-11-20:
@fooker Can you send a PR with your suggested fix?

fooker, 2019-11-20:
It looks like fixing the `payloadParser` is not enough. The `on_data_payload` and `on_data_end` callbacks will be called way to late (the are called if a follow-up message is received).

I've opened an associated issue in the according library: https://github.com/slaff/mqtt-codec/issues/3

fooker, 2019-11-20:
@slaff If https://github.com/slaff/mqtt-codec/pull/4 gets merged, there is no need for change in Sming.

Another way to solve this is not to skip the callbacks on empty messages, but make them all be called with zero length. In such a case, I will come up with a PR.

closed 2019-11-20 by slaff

fooker, 2019-11-20:
@slaff will this be back-ported to a 4.0.x release?

PR #1976 (2019-11-20): Update mqtt-codec to allow publish messages without payload.
Closes #1975.