SmingHub/Sming #1982: How to get 4.0 running

DonTSan, 2019-11-25:
I've installed Sming 4.0 and esp-open-sdk on 2 different computers and a virtual machine (all Ubuntu type). I tried Wemos d1 mini, Node-MCU and a self-made circuit board. I don't get even Basic_Blink running. Sming 3.8 (esp-open-sdk 2.0) works fine on all boards. Connecting on 74880 BAUD shows following information: 
**_ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 1188, room 16 
tail 4
chksum 0xea
load 0x3ffe8000, len 600, room 4 
tail 4
chksum 0x7c
csum 0x7c

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   4 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz
rBoot Option: Big flash

Booting rom 0 at 2000, load addr 39990.
emapg
map 0 err
system param partition error
ota1 not set
ota2 not set
system_partition_table_regist: failed
size_map = 0
partition[0]: 1, 0x00000000, 0x00001000
partition[1]: 100, 0x00001000, 0x00001000
partition[2]: 101, 0x00002000, 0x00079000
partition[3]: 4, 0x0007b000, 0x00001000
partition[4]: 5, 0x0007c000, 0x00001000
partition[5]: 6, 0x0007d000, 0x00003000_**

However, 'Flashinit' does not make any difference. Does anyone know what's going wrong with flashing? Any help would be highly appreciated.

mikee47, 2019-11-25:
Try SPI_SIZE=4M for the NodeMCU. See https://smingdev.readthedocs.io/en/latest/_inc/Sming/Arch/Esp8266/Components/esptool/index.html and https://smingdev.readthedocs.io/en/latest/_inc/Sming/Arch/Esp8266/Components/esp8266/index.html

kmihaylov, 2019-11-25:
And do not forget https://smingdev.readthedocs.io/en/latest/_inc/Sming/building.html

```make config-clean```

DonTSan, 2019-11-26:
Thanks both of you. I'm going to start a reading session and will come back later.

DonTSan, 2019-11-26:
It's running.
I started with <make config-clean>, then <make flashinit> and at last <make flash SPI_MODE=dio SPI_SIZE=4M>
Thanks again.

DonTSan, 2019-11-26:
I guess my sentence is not shown completely. The sequence I did was:
-  make config-clean

- make flashinit

- make flash SPI_MODE=dio SPI_SIZE=4M

closed 2020-05-24 by slaff