SmingHub/Sming #1990: String memory consumption?

kmihaylov, 2019-12-07:
Hello,

I'm stuck with strange problem. My app has several classes, one for the mqtt client, one for some tasks.

When the app tries to send several messages at once I use:
```
void kulmq::pubMsg(String topic, String data)
{
	if(mqtt.getConnectionState() == eTCS_Connected) {
		topic = MqttConfig.DeviceTopic + topic;
		mqtt.publish(topic, data, 0);
	}
}
```

Usually this works. Recently I had to create mqtt username and password that were not very short (21 chars). And surprisingly my mqtt client failed to connect.

Initially I tried to reduce the username and password lengths and the problem disappeared... Strange. I thought that maybe there are some limitations in the ```Url``` type used in the mqtt client. But then the problem appeared again.

Then maybe there is some escape problem with the JSON string (that I read from a config file) since the device topic is inside it and ends with forward slash.

I tried to remove the forward slash and again it started to work somehow, but after few tests the mqtt client started to fail again.

I had to debugf the deserialized JSON string and to compare with the filled variables:
```
struct MqttConfigContainer {
	String Protocol = "mqtt";
	String Host = "iot.kitove.com";
	int Port = 1883;
	String User = "";
	String Password = "";
	String DeviceTopic = "";
};

MqttConfigContainer MqttConfig;
```
```
	const char* jsonString = fileGetContent(MQTT_CONF_FILENAME).c_str();

	DynamicJsonDocument root(ConfigJsonBufferSize);
	DeserializationError error = deserializeJson(root,jsonString);
	debugf("JSON deserial err: %s", error.c_str());
	//MqttConfig.Host = String( (const char*) root["Host"] );
	MqttConfig.User = String((const char*)root["User"]);
	MqttConfig.Password = String((const char*)root["Password"]);
	MqttConfig.DeviceTopic = String((const char*)root["DeviceTopic"]);
	Serial.printf("%s %s %s\r\n", MqttConfig.User.c_str(), MqttConfig.Password.c_str(), MqttConfig.DeviceTopic.c_str());
	delete[] jsonString;
```
later in startMqttClient()
```
	mqttClientRemoteUrl.Scheme = MqttConfig.Protocol;
	mqttClientRemoteUrl.Host = MqttConfig.Host;
	mqttClientRemoteUrl.Port = MqttConfig.Port;
	mqttClientRemoteUrl.User = MqttConfig.User;
	mqttClientRemoteUrl.Password = MqttConfig.Password;
	Serial.printf("%s %s %s\r\n", MqttConfig.User.c_str(), MqttConfig.Password.c_str(), MqttConfig.Host.c_str());
```

And voila, the MqttConfig that was just filled with the parsed JSON string contained wrong username (sometimes some other part of the string was wrong).

I figured out that there is some app function that tries to send some data even before the mqtt client was initialized (though this shouldn't be a problem because in the pubMsg() I'm checking the connection state.

The f() that tries to send the data calls pubMsg() few times. And It happened so that if I reduce the number of times pubMsg() is called, the mqtt connection variables remain intact. If they're more than (for example 3), then the problem starts to appear.
Here is the app f() that sends the data:
```
void heatMeterClass::publishData()
{
	mqttClient->pubMsg("topla", String(tempSensor1->getTemperature(), 2));
	mqttClient->pubMsg("studena", String(tempSensor2->getTemperature(), 2));
	mqttClient->pubMsg("debit", String(flowMeter->getFlow(), 3));

	float waterPower = calculateWaterPower(flowMeter->getFlow(), tempSensor1->getTemperature(), tempSensor2->getTemperature());
	mqttClient->pubMsg("P", String(waterPower, 2));

	mqttClient->pubMsg("kWh", String(kWh, 2));
}
```

It looks exactly like some overflow, since I don't use special memory allocation or char arrays and etc..

So I replaced the pubMsg() with another function that takes reference to String as its arguments:
```
void kulmq::pubMsgRef(String& topic, const String& data)
{
	if(mqtt.getConnectionState() == eTCS_Connected) {
		topic = MqttConfig.DeviceTopic + topic;
		mqtt.publish(topic, data,
					 0x00); //flags: 1st bit - retain, 2nd and 3rd - QoS, 4th - DUP; message.h mqtt-codec/src/message.h
	}
}
```

And my app f() that sends the data now looks like:
```
void heatMeterClass::publishData()
{
	String topic = "topla";
	String data;
	data = String(tempSensor1->getTemperature(), 2);
	mqttClient->pubMsgRef(topic, data);

	topic = "studena";
	data = String(tempSensor2->getTemperature(), 2);
	mqttClient->pubMsg(topic, data);

	topic = "debit";
	data = String(flowMeter->getFlow(), 3);
	mqttClient->pubMsg(topic, data);

	float waterPower = calculateWaterPower(flowMeter->getFlow(), tempSensor1->getTemperature(), tempSensor2->getTemperature());
	topic = "P";
	data = String(waterPower, 2);
	mqttClient->pubMsg(topic, data);

	topic = "kWh";
	data = String(kWh, 2);
	mqttClient->pubMsg(topic, data);
}
```

And now the app works as expected, the mqtt variables thah have been filled with the deserialized JSON data are intact. Also I tried to increase the body of heatMeterClass::publishData() with few more mqtt messages to be send in order to verrify if the problem is really gone, and it worked well.

-------

So my question is "If there are fast multiple calls to function that utilizes String arguments, is it possible that memory usage increases and for some reason overflows other variables?"

mikee47, 2019-12-07:
What version of Sming are you building against?

kmihaylov, 2019-12-07:
d1a1efa4

mikee47, 2019-12-07:
OK, this line is suspicious:

```
const char* jsonString = fileGetContent(MQTT_CONF_FILENAME).c_str();
```

You're taking a pointer to a temporary string which gets destroyed immediately. Do this instead:

```
String jsonString = fileGetContent(MQTT_CONF_FILENAME).c_str();
```

Additionally, you can simplify your code using the Json helper functions:

```
DynamicJsonDocument root(ConfigJsonBufferSize);
if(!Json::loadFromFile(root, MQTT_CONF_FILENAME)) {
  debugf("JSON load failed");
}
//MqttConfig.Host = String( (const char*) root["Host"] );
MqttConfig.User = String((const char*)root["User"]);
MqttConfig.Password = String((const char*)root["Password"]);
MqttConfig.DeviceTopic = String((const char*)root["DeviceTopic"]);
Serial.printf("%s %s %s\r\n", MqttConfig.User.c_str(), MqttConfig.Password.c_str(), MqttConfig.DeviceTopic.c_str());
```

kmihaylov, 2019-12-07:
Excuse me for bothering you with complex code. I just prepared the example:

https://github.com/kmihaylov/Sming_String_Overflow

kmihaylov, 2019-12-07:
Just updated the example. It is my fault, sorry! I'll look your notes now.
My test now calls function with String argument each millisecond and nothing happens to a global variable, so there is no reason for blaming String...

mikee47, 2019-12-07:
See also https://arduinojson.org/v6/api/json/deserializejson/, specifically:

*If the input is a char\*, ensure the input buffer stays in memory until the JsonDocument is destructed.*

Have a read of https://arduinojson.org/v6/doc/upgrade/.

kmihaylov, 2019-12-07:
I did what you suggested and the device now works correctly. You're always of great help! In acknowledgement of your help I'll set a picture of my cat for my avatar.
Respect!

closed 2019-12-07 by kmihaylov