SmingHub/Sming #1991: ModbusMaster in Sming

kmihaylov, 2019-12-08:
Hello. During OpenFest I had quick conversation with @slaff about modbus that I used in Sming.

I was able to use ModbusMaster (https://github.com/nomis/ModbusMaster/tree/fixes-2.0.1) without any special modifications.

However during my research I also found that @mikee47 has his own implementation of modbus master (https://github.com/SmingHub/Sming/compare/develop...mikee47:feature/modbus)

What opinion have the core developers regarding modbus master implementation (mike47's and/or ModbusMaster)?

PS: for modbus slave I use modbusino with 1-3 major changes that I can prepare for PR.
PS2: Sming ModbusMaster firmware is currently working on my desk together with Kinco HMI for video shooting and PR.

mikee47, 2019-12-09:
I think it's down to what your application needs. Here are some specifics about my own RTU implementation:

* Fully asynchronous. Timeouts and response handling are handled via callback. No polling, loops, WDT issues.
* Minimal memory usage. Uses hardware serial buffer.
* Does not attempt to provide streaming support: modbus-RTU is a control protocol, not a streaming protocol.

I use this as part of a device control stack which talks to a load of other devices as well. That includes DMX512 controllers sitting on the same wires (using a different driver), plus devices on other busses. The approach is closer to how device stacks work in a real OS so it's capable of handling the load without affecting system responsiveness. Other implementations are just too generic and when you start to load up the application it just doesn't work well enough.

At some point I'll probably include this as part of a seprate `IO Control` library.

kmihaylov, 2019-12-09:
I would like to see your ```IO Control``` as part of Sming. I guess it could be used for modbus slave also?

mikee47, 2019-12-09:
Doesn't have slave at present but definitely something to add. I'll get around to it eventually! Things have changed a lot since #1431.

mikee47, 2019-12-09:
I like this PR as it gives an easy introduction to modbus and can try stuff out, and port existing applications over more easily. Then we can have a more integrated approach if it's required. Lots of options, good.

slaff, 2019-12-13:
> At some point I'll probably include this as part of a separate IO Control library.

Go @mikee47 Go! If the IO Control Library makes it into this release then I already have a code name for the release :)

kmihaylov, 2020-04-12:
Serial.onTransmitComplete() issues, detals here: #2043 

Meanwhile I'm not sure about the UART status register values. I tried to copy->paste @mikee47's solution, but the TX FIFO is always empty.

```
void uartCallback(uart_t* uart, uint32_t status) {
	debugf("cb status: %d", status);

	// Tx FIFO empty
	if(status & _BV(UIFE)) {
		// Set MAX485 back into read mode
		digitalWrite(RS485_RE_PIN, LOW);
	}
}
```

In ```init()```
```
modbusComPort.setUartCallback(uartCallback);
```

kmihaylov, 2020-06-01:
I'm closing this since it's already commited: #1992 #2038 #2059
Future work to involve interrupts...?

closed 2020-06-01 by kmihaylov