SmingHub/Sming #1993: ECDHE-RSA for lightweight TLS handshake

cometurrata, 2019-12-09:
Hello,
it seems that Sming could benefit from enabling ECDHE-RSA instead of DHE-RSA. (Faster and lighter TLS).

https://security.stackexchange.com/questions/14731/what-is-ecdhe-rsa

Unfortunately axtls does not provide support for ECDHE, but mbed-tls does !

Any particular reason for not using mbed-tls ?
Has it already been discussed ?
Any hint of what I should start with ?
Do you think it could bring something useful ?

Thanks.

slaff, 2019-12-09:
> Any particular reason for not using mbed-tls ?

We have already started discussing this in #1713. What we lack is an abstraction that would allow us to switch during linking type between the different SSL implementations. 

> Any hint of what I should start with ?

@cometurrata if you want you can start with building that abstraction and adding axtls and mbed-tls implementations?

slaff, 2020-01-02:
@cometurrata PR #1999 will be merged soon. Once that is done you can add the new SSL adapter based on mbed-tls.

slaff, 2020-01-03:
@cometurrata the SSL adapters are now [merged](https://github.com/SmingHub/Sming/commit/41d03e8747b37e93f3f7d9e8068e3788bc2ee3ee)  in the `develop` branch. Bearssl supports ECDHE https://bearssl.org/support.html

closed 2020-01-06 by slaff