SmingHub/Sming #1995: Cannot compile

teicors, 2019-12-10:
Hi, 
after a git pull I cannot compile anymore.
I did aa make dist-clean, but with no luck.
My make list-config:

```
make list-config

Cronotermostato_nonos: Invoking 'list-config' for Esp8266 (debug) architecture

** Sming build configuration **

CUSTOM_TARGETS
- check-ssl
- out/Esp8266/debug/firmware/rboot.bin
- out/Esp8266/debug/firmware/rom0.bin
- out/Esp8266/debug/firmware/spiff_rom.bin
LIBS
- crypto
- hal
- main-c0c9ae36232daaefc9ab707b79649940
- microc
- microgcc
- net80211
- phy
- pp
- setjmp
- wpa
ARDUINO_LIBRARIES
- ArduinoJson5
- DS18S20
- LiquidCrystal
- OneWire
- SolarCalculator
- Timezone
SUBMODULES
- /opt/sming/Sming/Arch/Esp8266/Components/driver/new-pwm
- /opt/sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip
- /opt/sming/Sming/Arch/Esp8266/Components/esptool/esptool
- /opt/sming/Sming/Components/FlashString
- /opt/sming/Sming/Components/http-parser
- /opt/sming/Sming/Components/libyuarel
- /opt/sming/Sming/Components/mqtt-codec
- /opt/sming/Sming/Components/rboot/esptool2
- /opt/sming/Sming/Components/rboot/rboot
- /opt/sming/Sming/Components/spiffs/spiffs
- /opt/sming/Sming/Components/ws_parser
- /opt/sming/Sming/Libraries/ArduinoJson5/ArduinoJson
CONFIG_VARS
- COM_SPEED_SERIAL = $(COM_SPEED)
- DEBUG_PRINT_FILENAME_AND_LINE = 0
- DEBUG_VERBOSE_LEVEL = 3
- ENABLE_CMD_EXECUTOR = 1
- ENABLE_CUSTOM_HEAP = 0
- ENABLE_CUSTOM_LWIP = 1
- ENABLE_CUSTOM_PWM = 1
- ENABLE_ESPCONN = 0
- ENABLE_GDB = 
- ENABLE_HTTP_SERVER_MULTIPART = 
- ENABLE_LWIPDEBUG = 0
- ENABLE_SMART_CONFIG = 
- ENABLE_SSL = 
- ENABLE_TASK_COUNT = 
- ENABLE_WPS = 
- GDB_UART_SWAP = 
- HTTP_SERVER_EXPOSE_NAME = 1
- HTTP_SERVER_EXPOSE_VERSION = 0
- LOCALE = 
- MQTT_NO_COMPAT = 
- PROJECT_DIR = /home/angelo/Software/esp8266/project/Cronotermostato_nonos
- RBOOT_GPIO_ENABLED = 0
- RBOOT_GPIO_SKIP_ENABLED = 0
- RBOOT_ROM0_ADDR = 0x002000
- RBOOT_ROM1_ADDR = 
- RBOOT_ROM2_ADDR = 
- RBOOT_RTC_ENABLED = 0
- RBOOT_SILENT = 0
- RBOOT_SPIFFS_0 = 0x100000
- RBOOT_SPIFFS_1 = 0x300000
- SDK_BASE = 
- SPIFF_FILEDESC_COUNT = 7
- SPIFF_SIZE = 511024
- SPI_MODE = dio
- SPI_SIZE = 4M
- SPI_SPEED = 40
- STRING_OBJECT_SIZE = 12
- TASK_QUEUE_LENGTH = 10
- USER_CFLAGS = 
- USE_US_TIMER = 1
- WIFI_PWD = 
- WIFI_SSID = 
CACHE_VARS
- COM_OPTS = --raw --encoding ascii
- COM_PORT = /dev/ttyUSB0
- COM_PORT_ESPTOOL = $(COM_PORT)
- COM_PORT_GDB = $(COM_PORT)
- COM_SPEED = 115200
- COM_SPEED_ESPTOOL = 1000000
- DISABLE_SPIFFS = 0
- ENABLE_LEGACY_BUILD = 0
- FULL_COMPONENT_BUILD = 
- GDB_CMDLINE = trap '' INT; $(GDB) -x $(GDBSTUB_DIR)/gdbcmds -b $(COM_SPEED_GDB) -ex "target remote $(COM_PORT_GDB)"
- KILL_TERM = pkill -9 -f "$(COM_PORT) $(COM_SPEED_SERIAL)" || exit 0
- SERVER_OTA_PORT = 9999
- SPIFF_BIN = spiff_rom
- SPIFF_FILES = web
- STRICT = 
- TERMINAL = python -m serial.tools.miniterm $(COM_OPTS) $(COM_PORT) $(COM_SPEED_SERIAL)
- TRACE = 
RELINK_VARS
- RBOOT_LD_TEMPLATE = $(ARCH_BASE)/Compiler/ld/rboot.rom0.ld
- RBOOT_ROM_0 = rom0
- RBOOT_ROM_1 = rom1
DEBUG_VARS
- APP_CFLAGS =  -DCOM_SPEED_SERIAL=$(COM_SPEED_SERIAL) -DRBOOT_SPIFFS_0=$(RBOOT_SPIFFS_0) -DRBOOT_SPIFFS_1=$(RBOOT_SPIFFS_1) -DRBOOT_INTEGRATION -DSPIFF_SIZE=$(SPIFF_SIZE)
- APP_LIBDIR = out/Esp8266/debug/lib
- ARCH_BASE = /opt/sming/Sming/Arch/Esp8266
- AWK = POSIXLY_CORRECT= awk
- BUILD_BASE = $(OUT_BASE)/build
- CLANG_FORMAT = clang-format
- CMAKE = cmake
- COM_SPEED_GDB = $(COM_SPEED_SERIAL)
- ESPTOOL = /opt/sming/Sming/Arch/Esp8266/Components/esptool/esptool/esptool.py
- ESPTOOL2 = /opt/sming/Sming/out/Esp8266/debug/tools/esptool2
- ESP_HOME = /opt/esp-open-sdk
- FW_BASE = $(OUT_BASE)/firmware
- GCC_VERSION = 4.8.5
- GDB = /opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gdb
- GDBSTUB_DIR = /opt/sming/Sming/Arch/Esp8266/Components/gdbstub
- GIT = git
- GLOBAL_CFLAGS = $(USER_CFLAGS) -DPROJECT_DIR=\"$(PROJECT_DIR)\" -DSMING_HOME=\"$(SMING_HOME)\" -DENABLE_CMD_EXECUTOR=$(ENABLE_CMD_EXECUTOR) -DHTTP_SERVER_EXPOSE_NAME=$(HTTP_SERVER_EXPOSE_NAME) -DHTTP_SERVER_EXPOSE_VERSION=$(HTTP_SERVER_EXPOSE_VERSION) -DDEBUG_PRINT_FILENAME_AND_LINE=$(DEBUG_PRINT_FILENAME_AND_LINE) -DCUST_FILE_BASE=$$* -DDEBUG_VERBOSE_LEVEL=$(DEBUG_VERBOSE_LEVEL) -DSTRING_OBJECT_SIZE=$(STRING_OBJECT_SIZE)  -DUSE_US_TIMER
- LIBMAIN_COMMANDS = $(APP_LIBMAIN_COMMANDS)
- OS = 
- OUT_BASE = out/Esp8266/debug
- RBOOT_BIG_FLASH = 0
- RBOOT_TWO_ROMS = 0
- SDK_INCDIR = /include
- SDK_INTERNAL = 0
- SDK_LIBDIR = /lib
- SMING_ARCH = Esp8266
- SMING_CPP_STD = c++11
- SMING_HOME = /opt/sming/Sming
- SMING_RELEASE = 
- TOOLS_BASE = $(SMING_HOME)/$(OUT_BASE)/tools
- UNAME = Linux
- USER_LIBDIR = $(SMING_HOME)/$(OUT_BASE)/lib

make: Nessuna operazione da eseguire per "list-config".
```

and here's the error

```
C+ /opt/sming/Sming/Arch/Esp8266/Components/spi_flash/iram_precache.cpp
AR /opt/sming/Sming/out/Esp8266/debug/lib/clib-spi_flash.a

Building /opt/sming/Sming/out/Esp8266/debug/lib/clib-esp-open-lwip-4a6822d010dd70a4fb667b785ccea5a3.a
CC /opt/sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/def.c
CC /opt/sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/dhcp.c
CC /opt/sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/dns.c
CC /opt/sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/init.c
CC /opt/sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/mem.c
CC /opt/sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/memp.c
CC /opt/sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/netif.c
CC /opt/sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/pbuf.c
CC /opt/sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/raw.c
CC /opt/sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/sntp.c
/opt/sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/sntp.c:302:1: error: unknown type name 'LOCAL'
 LOCAL os_timer_t sntp_timer;
 ^
/opt/sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/sntp.c:302:18: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'sntp_timer'
 LOCAL os_timer_t sntp_timer;
                  ^
/opt/sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/sntp.c: In function 'sntp_process':
/opt/sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/sntp.c:682:20: error: 'sntp_timer' undeclared (first use in this function)
   os_timer_disarm(&sntp_timer);
                    ^
/opt/sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/sntp.c:682:20: note: each undeclared identifier is reported only once for each function it appears in
/opt/sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/sntp.c: In function 'sntp_stop':
/opt/sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/lwip/core/sntp.c:1014:20: error: 'sntp_timer' undeclared (first use in this function)
   os_timer_disarm(&sntp_timer);
                    ^
/opt/sming/Sming/component-wrapper.mk:155: recipe for target 'esp-open-lwip/lwip/core/sntp.o' failed
make[1]: *** [esp-open-lwip/lwip/core/sntp.o] Error 1
/opt/sming/Sming/project.mk:357: recipe for target 'esp-open-lwip-build' failed
make: *** [esp-open-lwip-build] Error 2
```

mikee47, 2019-12-10:
For some reason, SDK_BASE isn't set:
```
- SDK_BASE = 
```
Try `export SDK_BASE='$(SMING_HOME)'`.

teicors, 2019-12-10:
A mistery...
I fixed issuing a

`make SDK_BASE='$(SMING_HOME)'`

and after I did again a make dist-clean and the SDK_BASE was set !

but I found another problem

```
AR /opt/sming/Sming/out/Esp8266/debug/lib/clib-Timezone.a

Building /home/angelo/Software/esp8266/project/Cronotermostato_nonos/out/Esp8266/debug/lib/clib-App-460c12fb1423baf1c104e8c750b5b2cd.a
C+ /home/angelo/Software/esp8266/project/Cronotermostato_nonos/app/webserver.cpp
C+ /home/angelo/Software/esp8266/project/Cronotermostato_nonos/app/orologio.cpp
C+ /home/angelo/Software/esp8266/project/Cronotermostato_nonos/app/configuration.cpp
C+ /home/angelo/Software/esp8266/project/Cronotermostato_nonos/app/NtpClientDemo.cpp
C+ /home/angelo/Software/esp8266/project/Cronotermostato_nonos/app/application.cpp
make[1]: ***  Nessuna regola per generare l'obiettivo "/opt/sming/Sming/Arch/Esp8266/Components/esp8266/include/c_types.h", necessario per "/opt/sming/Sming/Components/rboot/appcode/rboot-overrides.o".  Arresto.
/opt/sming/Sming/project.mk:357: recipe for target 'App-build' failed
make: *** [App-build] Error 2
```

mikee47, 2019-12-10:
Does `make config-clean` help?

mikee47, 2019-12-10:
Note: `make dist-clean` won't touch ypur project config, that only affects Sming framework and samples.

mikee47, 2019-12-10:
Oh, and `make clean` for your project.

teicors, 2019-12-10:
No and I don't understand were the problem lies. Just as a check I compiled Basic_Blink and two my other projects, so the first issue is solved. Now I must find  where the dirty chars in the code.

teicors, 2019-12-10:
is correct the file /opt/sming/Sming/Components/rboot/appcode/rboot-overrides.c.d ,
I found 
```
angelo@angelo-thinkpad-x1-carbon-4th:~/Software/esp8266/project/Cronotermostato_nonos$ grep c_types.h /opt/sming/Sming/Components/rboot/appcode/rboot-overrides.c.d 
 /opt/sming/Sming/Arch/Esp8266/Components/esp8266/include/c_types.h \
 /opt/esp-open-sdk/sdk/include/c_types.h \
 /opt/sming/Sming/Arch/Esp8266/Components/esp8266/include/c_types.h \
```

because the error belongs to the compiling of the file /opt/sming/Sming/Components/rboot/appcode/rboot-overrides.o

teicors, 2019-12-10:
Hi, I solved.
I'm not sure, but I think that you changed something in the logic of the file component.mk: until this morning I compiled with my libs put in different lines, now all libs in one line.

mikee47, 2019-12-10:
Would you post the old and new files/lines so I can try and reproduce the problem?

teicors, 2019-12-11:
Hi,
I've found two problems in my config:

1. the file /opt/sming/Sming/Components/rboot/appcode/rboot-overrides.c.d dated 22 Oct: 
inside this file I found few references to nonexistent files:

```
angelo@angelo-thinkpad-x1-carbon-4th:/opt/sming/Sming/Components/rboot/appcode$ diff rboot-overrides.c.d_ko rboot-overrides.c.d
8d7
<  /opt/sming/Sming/Arch/Esp8266/Components/esp8266/include/c_types.h \
10d8
<  /opt/sming/Sming/Arch/Esp8266/Components/esp8266/include/esp_attr.h \
28d25
<  /opt/sming/Sming/Arch/Esp8266/Components/esp8266/include/c_types.h \
```

Now the file don't contains the dirty lines, but I don't know if is related at my editing or not.

2. different managing of lines component.mk:
As I told before, until yesterday I had my libs in the component.mk distributed over several lines
ie
```
COMPONENT_DEPENDS := Default 
COMPONENT_DEPENDS := CronLibrary
```
but now to compile my project I put all libs on a single line

`COMPONENT_DEPENDS := Default CronLibrary`

and this behavior is confirmed also by 

`make list-components` 

I cannot revert to the previous version, but before i can compile my project flawless.

slaff, 2019-12-11:
>  /opt/sming/Sming/Components/rboot/appcode/rboot-overrides.c.d

The '.d' files should be removed before compiling with newer version. This can be done by running:
```
make dist-clean # to clean the Sming dependency files
make clean # to clean your application dependency files.
```

Most probably you had an old dep file and that's why the compilation did not work as expected.

mikee47, 2019-12-11:
1. That file `/opt/sming/Sming/Components/rboot/appcode/rboot-overrides.c.d` shouldn't be there! Please delete it. All `.d` files are compiler-generated and should not be in the source tree - please run `git status` from your Sming directory to check there aren't any other odd files around.

2. As written above `CronLibrary` would overwrite `Default`. This will work:
```
COMPONENT_DEPENDS := Default 
COMPONENT_DEPENDS += CronLibrary
```
or
```
COMPONENT_DEPENDS := \
  Default \
  CronLibrary
```

Does this make sense?

teicors, 2019-12-11:
I did both make dist-clean and clean, but the file .d is still there.
`-rw-r--r-- 1 angelo angelo 3732 dic 10 21:18 rboot-overrides.c.d`

 and git status show the following lines

```
	Components/
	Sming/Arch/Esp8266/Components/gdbstub/appcode/gdb_hooks.cpp.d
	Sming/Arch/Esp8266/Components/gdbstub/appcode/gdb_hooks.o
	Sming/Arch/Esp8266/Components/gdbstub/appcode/gdbstub-entry.o
	Sming/Arch/Esp8266/Components/pwm_open/
	Sming/Components/rboot/appcode/rboot-overrides.c.d
	Sming/Components/rboot/appcode/rboot-overrides.c.d_ko
	Sming/Components/rboot/appcode/rboot-overrides.o
	Sming/component.mk_old
```

I know the story about "Sming/component.mk_old" and "Sming/Components/rboot/appcode/rboot-overrides.c.d_ko", but the other lines can be saftley removed (git checkout)?

mikee47, 2019-12-11:
> other lines can be safely removed (git checkout)?

Yes, please remove them. The new build system will only write to the `out` directory, there should be nothing added to source directories.

teicors, 2019-12-11:
Regarding 2. For me still a mystery: it's important to fix the issue and knowing how solve it.
You're right showing me the correct definition, but I can assure you that until yesterday I used a different (and may be wrong) configuration. And it worked!
And once again lesson learned ...

teicors, 2019-12-11:
git status shows
`Sming/Arch/Esp8266/Components/pwm_open/` 

Should I erase it or not ?

mikee47, 2019-12-11:
Yes, it's moved into Components/drivers .

closed 2020-07-09 by teicors