SmingHub/Sming #1997: $(AWK) is undefined

kmihaylov, 2019-12-11:
Good morning,

This shouldn't be a thing since Sming compiles successfully despite the error, but when I do ```make dist-clean``` I'm getting this warning:
```
$ make dist-clean 

general: Invoking 'dist-clean' for Esp8266 (debug) architecture
Cleaning all submodules...
Cleaning all samples...
/bin/bash: BEGIN { big=0; for(i = 1; i < ARGC; ++i) if(ARGV[i] > 1024*1024) big=1; print big; }: command not found
../samples/HttpServer_FirmwareUpload/component.mk:10: warning: overriding recipe for target 'web-pack'
../samples/HttpServer_ConfigNetwork/component.mk:5: warning: ignoring old recipe for target 'web-pack'
../samples/HttpServer_FirmwareUpload/component.mk:13: warning: overriding recipe for target 'web-upload'
../samples/HttpServer_ConfigNetwork/component.mk:9: warning: ignoring old recipe for target 'web-upload'
```

Setting a variable ```export AWK=/usr/bin/awk``` solves the problem.

slaff, 2019-12-11:
@kmihaylov Did you pull the latest `develop` version? That should have been fixed.

mikee47, 2019-12-11:
Try a git pull, this is fixed in latest commit 3a99771e

kmihaylov, 2019-12-11:
Thanks! The warning is gone.

closed 2019-12-11 by kmihaylov