SmingHub/Sming #2000: cls

mikee47, 2019-12-13:


mikee47, 2019-12-19:
So far, no luck in finding a solution to this one. Any suggestions?

HanSolo72, 2020-01-18:
Today, I was able to build the Blink sample one or two times, but now I get the following error:

```
C:\tools\Sming\samples\Basic_Blink>make

Basic_Blink: Invoking 'all' for Esp8266 (debug) architecture

Building /c/tools/Sming/samples/Basic_Blink/out/Esp8266/debug/lib/clib-App-fc053ef1152a6b983c66a394c1d8a077.a
C+ /c/tools/Sming/samples/Basic_Blink/app/application.cpp
In file included from c:/tools/Sming/Sming/Arch/Esp8266/Components/esp8266/include/mem.h:18:0,
                 from c:/tools/Sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/include/arch/cc.h:41,
                 from c:/tools/Sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/include/lwip/arch.h:43,
                 from c:/tools/Sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/include/lwip/debug.h:35,
                 from c:/tools/Sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/include/lwip/opt.h:46,
                 from c:/tools/Sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/include/lwip/ip_addr.h:35,
                 from c:/tools/sming/Sming/Arch/Esp8266/Components/esp8266/ESP8266_NONOS_SDK/include/user_interface.h:30,
                 from c:/tools/Sming/Sming/Arch/Esp8266/Components/esp8266/include/esp_systemapi.h:47,
                 from c:/tools/Sming/Sming/Arch/Esp8266/System/include/user_config.h:7,
                 from c:/tools/Sming/Sming/System/include/gdb/gdb_hooks.h:22,
                 from c:/tools/Sming/Sming/Core/SmingCore.h:15,
                 from c:/tools/Sming/samples/Basic_Blink/app/application.cpp:1:
c:/tools/sming/Sming/Arch/Esp8266/Components/esp8266/ESP8266_NONOS_SDK/include/mem.h:42:60: error: conflicting declaration of C function 'void* pvPortMalloc(size_t, const char*, unsigned int, bool)'
 void *pvPortMalloc (size_t sz, const char *, unsigned, bool);
                                                            ^
In file included from c:/tools/Sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/include/arch/cc.h:41:0,
                 from c:/tools/Sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/include/lwip/arch.h:43,
                 from c:/tools/Sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/include/lwip/debug.h:35,
                 from c:/tools/Sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/include/lwip/opt.h:46,
                 from c:/tools/Sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/include/lwip/ip_addr.h:35,
                 from c:/tools/sming/Sming/Arch/Esp8266/Components/esp8266/ESP8266_NONOS_SDK/include/user_interface.h:30,
                 from c:/tools/Sming/Sming/Arch/Esp8266/Components/esp8266/include/esp_systemapi.h:47,
                 from c:/tools/Sming/Sming/Arch/Esp8266/System/include/user_config.h:7,
                 from c:/tools/Sming/Sming/System/include/gdb/gdb_hooks.h:22,
                 from c:/tools/Sming/Sming/Core/SmingCore.h:15,
                 from c:/tools/Sming/samples/Basic_Blink/app/application.cpp:1:
c:/tools/Sming/Sming/Arch/Esp8266/Components/esp8266/include/mem.h:10:14: note: previous declaration 'void* pvPortMalloc(size_t, const char*, uint32)'
 extern void *pvPortMalloc(size_t xWantedSize, const char *file, uint32 line);
              ^
In file included from c:/tools/Sming/Sming/Arch/Esp8266/System/include/user_config.h:7:0,
                 from c:/tools/Sming/Sming/System/include/gdb/gdb_hooks.h:22,
                 from c:/tools/Sming/Sming/Core/SmingCore.h:15,
                 from c:/tools/Sming/samples/Basic_Blink/app/application.cpp:1:
c:/tools/Sming/Sming/Arch/Esp8266/Components/esp8266/include/esp_systemapi.h:86:103: error: conflicting declaration of C function 'void ets_timer_arm_new(ETSTimer*, uint32_t, bool, int)'
 extern void ets_timer_arm_new(ETSTimer *ptimer, uint32_t milliseconds, bool repeat_flag, int isMstimer);
                                                                                                       ^
In file included from c:/tools/Sming/Sming/Arch/Esp8266/Components/esp8266/include/esp_systemapi.h:44:0,
                 from c:/tools/Sming/Sming/Arch/Esp8266/System/include/user_config.h:7,
                 from c:/tools/Sming/Sming/System/include/gdb/gdb_hooks.h:22,
                 from c:/tools/Sming/Sming/Core/SmingCore.h:15,
                 from c:/tools/Sming/samples/Basic_Blink/app/application.cpp:1:
c:/tools/sming/Sming/Arch/Esp8266/Components/esp8266/ESP8266_NONOS_SDK/include/osapi.h:65:6: note: previous declaration 'void ets_timer_arm_new(ETSTimer*, uint32_t, bool, bool)'
 void ets_timer_arm_new(os_timer_t *ptimer, uint32_t time, bool repeat_flag, bool ms_flag);
      ^
make[1]: *** [app/application.o] Error 1
make: *** [App-build] Error 2
```

But it seems that my SMING_HOME variable has the correct case.
```
C:\tools\Sming\samples\Basic_Blink>echo %SMING_HOME%
C:\tools\Sming\Sming
```

mikee47, 2020-01-18:
Try `make config-clean` then rebuild.