SmingHub/Sming #2001: Sming `No Wifi` build option

mikee47, 2019-12-13:
I recently came across https://basicengine.org/.

> The BASIC Engine is a very low-cost single-board home computer with advanced 2D color graphics and sound capabilities, roughly comparable to late-1980s or early-1990s computers and video game consoles. It can be built at home without special skills or tools and using readily available components for under 10 Euros in parts, or mass-produced for even less."

It's an extraordinary project, but of interest here is its use of the ESP8266 without WiFi stack to release a ton of RAM. It uses `https://github.com/pvvx/SDKnoWiFi.git` to achieve this. There's also an [Arduino Port](https://github.com/uli/Arduino_nowifi.git).

I think it'd be rather handy to have this as a build option for Sming, but of course it would have to be done properly. SDKnoWiFi has a ton of low-level definitions and essentially re-implements the low-level support code for task queues, timers, etc. which the SDK normally does. I may end up doing that but as we'd need 100% compatibility with the existing SDK it seems like a lot of work and a little risky.

I've investigated some alternative approaches with the aim of hooking or patching the offending startup code within `libmain.a` itself, so that the linker discards all the wifi stuff we don't need and gives us the RAM back.

This seems simple enough - just replace calls to `wifi_station_start` and so on with no-ops or calls to dummy functions, but patching ELF object files is rather more involved and I'm disinclined to do that manually; I want a scripted approach.

slaff, 2019-12-14:
> just replace calls to wifi_station_start and so on with no-ops

So I guess you are not looking for the solution in NoWifiSDK: https://github.com/pvvx/SDKnoWiFi/blob/master/Makefile#L91-L100?

One thing that comes to my mind is that when the name and the signature of a function are known then you can do the following:
- describe all functions that need to be replaced with their unique name and function signature (params + return value). Put them in a yaml file (manual work)
- group the functions by signature. Let's call this function group (FG). (manual work - in the same yaml file)
- weaken the functions from the list based on their names  (can be automated )
- create new file with one noop function per FGs.  (can be automated)
- in the same file alias all functions to their FG noop functions (can be automated)
```
    int wifi_do_something(int x, int y) __attribute__((alias("noop_int_int_int")));
```
- during link time if the noWifiSDK was selected then include the noop file.

mikee47, 2019-12-14:
Made a start on this, https://github.com/mikee47/Sming/tree/feature/no-wifi.

@slaff I haven't tried weakening functions, may give that a go if this doesn't pan out.

slaff, 2019-12-15:
> I haven't tried weakening functions,

GCC does that for you. For example rBoot is weakening Cache_Read_Enable_New in libmain:

```
xtensa-lx106-elf-objcopy -W Cache_Read_Enable_New libmain.a libmain2.a
```

mikee47, 2019-12-15:
Done some further work on this and weakening helps in some situations so looks like a combination approach is working best. I'll open a PR shortly with progress as it's working for multiple samples now, including debugger :-)

mikee47, 2019-12-15:
Getting the low-level functions into code is probably useful anyway general considering that the [Non-OS SDK](https://github.com/espressif/ESP8266_NONOS_SDK/README.md) is now considered 'end of life'. I'm also continuing to migrate towards the RTOS headers. Arduino has imported all the SDK header files and only uses the pre-built libraries from the various SDK versions. This is a sensible move and we could dispense with the ESP8266_NONOS_SDK as a submodule.

closed 2020-12-04 by mikee47