SmingHub/Sming #2002: Build system issues with libraries depending on other libraries

slaff, 2019-12-15:
A sample project has dependency on library `vl53l1x-st-api-arduino`. Its component.mk file has the following line
```
COMPONENT_DEPENDS := vl53l1x-st-api-arduino
```

In addition library `vl53l1x-st-api-arduino`  depends on library `Proximity_Gesture`. Its component.mk file has

```
COMPONENT_DEPENDS := Proximity_Gesture
```

Running make in the sample directory should have resolved the dependencies and the include folders correctly. Instead I have to run `make` multiple times to finish the compilation.

Steps to reproduce:

```
cd  $SMING_HOME
git remote add slaff https://github.com/slaff/Sming.git
git fetch slaff feature/proximity-gesture
git checkout feature/proximity-gesture
cd ../samples/Gesture_Vl53l1x
make dist-clean
make # the first make fails?
make # running make again, without doing anything brings us further
make # running make again, without doing anything helps us compile the sample
```

Is that a bug or I am doing something wrong?

mikee47, 2019-12-15:
You need to use [whole-file patching](https://sming.readthedocs.io/en/latest/_inc/Sming/building.html#git-submodules) so the build system can resolve all dependencies before fetching submodules. (This is something I intend to improve with #1983.)

In this case, create sub-directories `Sming/Libraries/.patches/Proximity_Gesture/` and `Sming/Libraries/.patches/vl53l1x-st-api-arduino/` and put the `component.mk` files in there. You can then remove the .patch files as there doesn't seem to be anything else of substance in there.

slaff, 2019-12-15:
> You need to use whole-file patching so the build system

Thanks, that made it work. 

> something I intend to improve with

This would be great :+1:

closed 2019-12-15 by slaff

mikee47, 2019-12-15:
The main reason your libraries need a `component.mk` file at all is because of the dependency of `Proximity_Gesture` on `VL53L0X`. I've had a re-read of the [Arduino Library Specification](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification) and there is a `depends` field, but that's missing from `Proximity_Gesture`.

As an aside, the specification also says that `src` is used as a default include folder. I deliberately decided not to do that - it's illogical - so we need the COMPONENT_INCDIRS. (Don't need COMPONENT_SRCDIRS though as it defaults to `src`).

Something we may consider for #1983 is marking a Component specifically as `arduino compliant` so the defaults can be changed appropriately. Depending on how things pan out that may or may not require a component.mk file. If libraries are compliant then we can decode the `library.properties` file, but obviously only if that's provided and compliant.

Note that in this situation you have two third-party libraries being pulled in directly, so this whole-file patching approach is going to be the only way to deal with it - the information just isn't available anywhere else. I'm not sure what improvements could be made to this, but it needs to be considered (which is why I'm adding this note).