SmingHub/Sming #2005: Strict-aliasing error during compiling

teicors, 2019-12-15:
Hi,
I'm trying to compile old libs from Adafruit. I remember in the previous version of sming (I don't remember winch version) I can compile and use them. But now when i try to compile I catch the error

```
angelo@angelo-thinkpad-x1-carbon-4th:~/Software/esp8266/project/LCDTest_nonos$ make

LCDTest_nonos: Invoking 'all' for Esp8266 (debug) architecture

Building /opt/sming/Sming/out/Esp8266/debug/lib/clib-Adafruit_GFX_AS.a
C+ /opt/sming/Components/Adafruit_GFX_AS/Adafruit_GFX_AS.cpp
In file included from /opt/sming/Sming/Wiring/FakePgmSpace.h:14:0,
                 from /opt/sming/Sming/Arch/Esp8266/Components/libc/include/assert.h:10,
                 from /opt/sming/Sming/Arch/Esp8266/Components/esp8266/include/esp_systemapi.h:26,
                 from /opt/sming/Sming/Arch/Esp8266/System/include/user_config.h:7,
                 from /opt/sming/Sming/Wiring/Arduino.h:5,
                 from /opt/sming/Components/Adafruit_GFX_AS/Adafruit_GFX_AS.h:7,
                 from /opt/sming/Components/Adafruit_GFX_AS/Adafruit_GFX_AS.cpp:34:
/opt/sming/Components/Adafruit_GFX_AS/Adafruit_GFX_AS.cpp: In member function 'int Adafruit_GFX_AS::drawUnicode(unsigned int, int, int, int)':
/opt/sming/Sming/Arch/Esp8266/Components/libc/include/sys/pgmspace.h:136:59: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
 #define pgm_read_dword(addr) (*(const unsigned long*)(addr))
                                                           ^
/opt/sming/Components/Adafruit_GFX_AS/Adafruit_GFX_AS.cpp:527:22: note: in expansion of macro 'pgm_read_dword'
      flash_address = pgm_read_dword(&chrtbl_f16[uniCode]);
                      ^
/opt/sming/Sming/Arch/Esp8266/Components/libc/include/sys/pgmspace.h:136:59: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
 #define pgm_read_dword(addr) (*(const unsigned long*)(addr))
```

mikee47, 2019-12-15:
The line reads as if it's attempting a 32-bit read on a 16-bit value. Highly suspicious:

```
flash_address = pgm_read_dword(&chrtbl_f16[uniCode]);
```

Can you direct me to the source for this library and I'll have a closer look?

teicors, 2019-12-16:
I sent an email to you.

mikee47, 2019-12-16:
Got the email, but no definition for `chrtbl_f16`. However, I found `https://github.com/MasterSch/Adafruit_GFX_AS/Font16.h` which has it defined as:

```
extern const unsigned char* const chrtbl_f16[96];
```

So the compiler is warning you about the conversion from a `const unsigned char**` to a `const unsigned long *`. The macro itself hasn't changed since it was added to the framework, so it's just the warning being thrown up.

In this particular instance you can just de-reference the pointer directly:

```
flash_address = chrtbl_f16[uniCode]
```

There may be a better implementation for the macro so I'll take a look at that. (e.g. in C++ we can use reinterpret_cast<>)

teicors, 2020-01-05:
I did some deep search in my hard disks and I found the very old version of SmingRTOS was able to compile and running the code. I can play also with SmingRTOS, but to manage a single framework :)

teicors, 2020-01-16:
I can compile and the apps is running with sming 3.8.1