SmingHub/Sming #2018: valgrind warning on uninitialized bytes for SPIFFS in HttpServer_Bootstrap

kpishere, 2020-01-07:
It looks like there may be a paging mis-calculation here.  Understanding that SPI flash must be written in full blocks, on writing or clearing flash before write, it appears that the padding to a full block is off.

It leaves data vulnerable to corruption if coupled with another file/stream error?

Error is :

root@debian:/opt/Sming/samples/HttpServer_Bootstrap# bash /home/kpeck/vfdCalendar/memorydebug.sh 
==15917== Memcheck, a memory error detector
==15917== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==15917== Using Valgrind-3.14.0 and LibVEX; rerun with -h for copyright info
==15917== Command: out/Host/debug/firmware/app --flashfile=out/Host/debug/firmware/flash.bin --uart=0 --uart=1 --pause=5 --flashsize=4M
==15917== 

Welcome to the Sming Host emulator

host_flashmem_init: Opened "out/Host/debug/firmware/flash.bin", size = 0x00400000
host_init_bootloader: MAGIC mismatch
host_init_bootloader: Write default config: OK
host_init_bootloader: ROM count = 2, current = 0, #0: 0x2000, #1: 0x202000
host_lwip_init: Initialising LWIP
CUartServer::thread_routine: UART0 server listening on port 10000
CUartServer::thread_routine: UART1 server listening on port 10001
host_lwip_init: Using interface 'tap0', gateway = 192.168.13.1, netmask = 255.255.255.0; using ip = 192.168.13.10
host_lwip_init: MAC: 9a:1a:a7:16:c5:81
main: If required, you may start terminal application(s) now
pause: Waiting for 5 seconds...
main: >> Starting Sming <<

==15917== Syscall param write(buf) points to uninitialised byte(s)
==15917==    at 0x485B447: __libc_write (write.c:26)
==15917==    by 0x485B447: write (write.c:24)
==15917==    by 0x1265DE: writeFlashFile (flashmem.cpp:113)
==15917==    by 0x1265DE: writeFlashFile(unsigned int, void const*, unsigned int) (flashmem.cpp:107)
==15917==    by 0x126835: flashmem_write (flashmem.cpp:176)
==15917==    by 0x13A139: api_spiffs_write (spiffs_sming.c:34)
==15917==    by 0x13DAC9: spiffs_object_create (spiffs_nucleus.c:960)
==15917==    by 0x13A8B6: SPIFFS_open (spiffs_hydrogen.c:255)
==15917==    by 0x13A3DC: spiffs_mount_internal (spiffs_sming.c:139)
==15917==    by 0x13A3DC: spiffs_mount_internal (spiffs_sming.c:101)
==15917==    by 0x13A433: spiffs_mount (spiffs_sming.c:153)
==15917==    by 0x1276DE: init() (application.cpp:88)
==15917==    by 0x10C94F: main (startup.cpp:239)
==15917==  Address 0xbe96b2c7 is on thread 1's stack
==15917==  in frame #4, created by spiffs_object_create (spiffs_nucleus.c:918)
==15917==  Uninitialised value was created by a stack allocation
==15917==    at 0x13D9C3: spiffs_object_create (spiffs_nucleus.c:918)
==15917== 
pinMode: 0, 1
IF_UP, AP: Host WiFi
IP_CHANGE, ip: 192.168.13.10, netmask: 255.255.255.0, gateway: 192.168.13.1
digitalWrite: 0, 0
digitalWrite: 0, 1
digitalWrite: 0, 0
digitalWrite: 0, 1
digitalWrite: 0, 0
digitalWrite: 0, 0
digitalWrite: 0, 0
digitalWrite: 0, 0
^C
signal_handler: SIGINT - Ctrl+C pressed
host_exit: returning 0
main: >> Normal Exit <<

host_flashmem_cleanup: Closed "out/Host/debug/firmware/flash.bin"
CSocket::close: 0.0.0.0:10000
CUartServer::terminate: UART0 server destroyed
CSocket::close: 0.0.0.0:10001
CUartServer::terminate: UART1 server destroyed
cleanup: Goodbye!
Assertion "invalid socket state for poll" failed at line 2117 in /opt/Sming/Sming/Arch/Host/Components/lwip/lwip/src/core/tcp.c

signal_handler: SIGABRT - usually caused by an abort() or assert()

==15917== 
==15917== HEAP SUMMARY:
==15917==     in use at exit: 652 bytes in 7 blocks
==15917==   total heap usage: 2,028 allocs, 2,021 frees, 103,600 bytes allocated
==15917== 
==15917== LEAK SUMMARY:
==15917==    definitely lost: 0 bytes in 0 blocks
==15917==    indirectly lost: 0 bytes in 0 blocks
==15917==      possibly lost: 0 bytes in 0 blocks
==15917==    still reachable: 652 bytes in 7 blocks
==15917==         suppressed: 0 bytes in 0 blocks
==15917== Reachable blocks (those to which a pointer was found) are not shown.
==15917== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==15917== 
==15917== For counts of detected and suppressed errors, rerun with: -v
==15917== ERROR SUMMARY: 4 errors from 1 contexts (suppressed: 0 from 0)

slaff, 2020-01-07:
@kpishere Did you create a flash image before running the sample?
```
make flash # run at least ones
make valgrind
```

kpishere, 2020-01-08:
Did make a flash file, it shows loading in the log above.  However, didn't realize you had make valgrind scripted ... I'll use that going forward.  Amazing framework by the way!

kpishere, 2020-01-09:
Ah, without the network up, there is that error.  By having tap0 interface up, no error.  OK ticket can be closed.

root@debian:/opt/Sming/samples/HttpServer_Bootstrap# /opt/Sming/Sming/Arch/Host/Tools/setup-network-linux.sh enp0s3
ioctl(TUNSETIFF): Device or resource busy
RTNETLINK answers: File exists
net.ipv4.ip_forward = 1
net.ipv6.conf.default.forwarding = 1
net.ipv6.conf.all.forwarding = 1

closed 2020-01-09 by kpishere

slaff, 2020-01-09:
>  Amazing framework by the way! 

Glad to hear that :) Make sure to star the project  and/or recommend us to your IoT friends :)

slaff, 2020-01-10:
> However, didn't realize you had make valgrind scripted ..

Run `make help` in a sample or in the library to see the list of available high-level tasts.