SmingHub/Sming #2020: cannot set date

frankdownunder, 2020-01-12:
In the excerpt below is the warning "cannot set date" of concern?
. . . .
Signature ok
subject=O = Sming Project, CN = 127.0.0.1
Getting CA Private Key
**date: cannot set date: Operation not permitted**
Wed 01 Jan 2025 12:00:00 AM AEDT
Signature ok
subject=O = Sming Project, CN = 127.0.0.1
Getting CA Private Key
rm: cannot remove 'srl': No such file or directory
wiringCenter: Linking out/Host/debug/firmware/app
   text    data     bss     dec     hex filename
 839968    9100   36380  885448   d82c8 out/Host/debug/firmware/app

slaff, 2020-01-13:
Which sample, which OS? @frankdownunder Please provide more information.

mikee47, 2020-01-13:
I believe the message is from `make-certs.sh`, line 139:

```
# note: must be root to do this
DATE_NOW=$(date)
if date -s "Jan 1 2025"; then
   ^^^^^^^
openssl x509 -req -in x509_512.req -out x509_bad_before.pem \
            -sha1 -CAcreateserial -days 365 \
            -CA ca_x509.pem -CAkey ca_key.pem
```

frankdownunder, 2020-01-13:
This was my own project, with Ubuntu.  Im not using ssl anyway,so doesnt worry me.
It doesnt happen the second time ! 
When I type date I get
xxxx@ubu:~/protr$ date
Tue 14 Jan 2020 08:41:47 AM AEDT
Note the order day month !

Should I have run with sudo ? How to make it happen again I wonder?

mikee47, 2020-01-13:
The `make-certs.sh` script is invoked only when ENABLE_SSL is set. You can make it happen again by deleting or renaming the `include\ssl` directory in the project.

mikee47, 2020-01-13:
NB. In current develop branch you can find `make-cert.sh` in `Sming/Components/ssl/Tools`  - previously `Components/axtls-8266/axtls-9266/tools` but behaviour hasn't changed. I think the file is only really for demonstration/testing purposes, in practice the certificates would be generated more carefully then copied into a project. The message in question appears to relate to creation of a 'bad' certificate with invalid dates.

@slaff I'm thinking we should probably make this script run only on user request, replacing it instead with a prompt to run it or add own certificates. Perhaps something for a new PR.

mikee47, 2020-01-13:
Short version: You can safely ignore the error.

slaff, 2020-01-14:
>  replacing it instead with a prompt to run it or add own certificates. 

If there will be prompt I would like to have also non-interactive mode. Don't want to break automated testing.

closed 2020-01-19 by slaff