SmingHub/Sming #2021: basic_Blink compilation error

icodk, 2020-01-14:
Fresh installation on Windows 10
Followed instructions on :
https://sming.readthedocs.io/en/latest/arch/esp8266/getting-started/windows-manual.html

Running `make -j ` results:
```
C:\tools\sming\Sming\samples\Basic_Blink>make -j
Makefile:9: C:\tools\sming\Sming/project.mk: No such file or directory
make: *** No rule to make target `C:\tools\sming\Sming/project.mk'.  Stop.

C:\tools\sming\Sming\samples\Basic_Blink>
```
The project.mk file is in C:\tools\sming\Sming\Sming 
However if I change the 
include $(SMING_HOME)/project.mk
to
include $(SMING_HOME)/Sming/project.mk
I still get (other) errors
Also in section 1 the istruction is` cd /d C:\tools\sming` 
At section 3 the instruction is `cd samples\Basic_Blink`
This doesn't work either (still one Sming too short:-)
Is there a good reason to have such path: `sming/Sming/Sming` ?
Thanks for a great framework (BTW)

mikee47, 2020-01-14:
This should work:

```
set SMING_HOME=C:\tools\sming\Sming
cd /d C:\tools\sming\samples\Basic_Blink
make -j
```

mikee47, 2020-01-14:
I see your point. Instructions need correcting. Wipe it and try this:

```
mkdir C:\tools
cd /d C:\tools
git clone https://github.com/SmingHub/Sming sming
set SMING_HOME=C:\tools\sming\Sming
cd sming\samples\Basic_Blink
make -j
```

icodk, 2020-01-14:
@mikee47 Thanks it works

icodk, 2020-01-14:
Almost
last 4 lines of the compilation:
```
LDGEN out/Esp8266/debug/build/rom0.ld
Basic_Blink: Linking out/Esp8266/debug/build/app_0.out
/bin/sh: python: command not found
make: *** [out/Esp8266/debug/build/app_0.out] Error 127

C:\tools\Sming\samples\Basic_Blink>
```

mikee47, 2020-01-14:
Do you have python installed? https://www.python.org/downloads/

icodk, 2020-01-14:
No. Can't find it mentioned in the Windows manual installation.
Which version Should I install?

mikee47, 2020-01-14:
Current version.

kwis2, 2020-01-15:
@icodk  Windows install on fresh system is often problem, and often with new version there are new problems.
The most important with Sming it is folder name. It should be `/tools/sming/Sming` and every automatic or manual method install to /tools/Sming/Sming. It was written "you could change dir" but in windows it not work. I mean System should be in `c:/tools/sming/Sming/System`, not `c:/tools/sming/Sming/Sming/System`
In my case the most important was where you are in `C:\tools\` to use `git clone https://github.com/SmingHub/Sming sming` instead of  `git clone https://github.com/SmingHub/Sming` to install into `c:/tools/sming/Sming` 

In fresh windows system there is a lot of problems with system settings - which are not known in linux. For instance there is no python at all ... In my opinion the easiest way it is (it was a few Sming version ago) to first install with `windows manual` way and then run `install Chocolatey` way which probably fail, but is often able to setup windows environment.

One of the most important problem with windows it is length of system PATH string. I am sure you will have to increase length to set full paths of Sming.
And finally you have to setup ESP8266 module for proper parameters to it could work. There is no manual (I don't know one) how to set up different modules, even to check if Sming work, especially small memory versions like  ESP8266-01 is hard to setup, and in my opinion the best part of manual is `Note also that samples using SPIFFS for Sming may not work` and there should be same note about OTA (rBoot) ...

icodk, 2020-01-15:
@kwis2 Thanks for your write up.
Following @mikee47 instructions  seems to work. Needs only to update the docs and to include python installation (if not already installed)

icodk, 2020-01-16:
Now it is not working again
I run 
`c:\Temp\Basic_Blink>make -j`
The first line from the compilation
`Basic_Blink: Invoking 'all' for esp8266 (debug) architecture`

Here is the error ( after many output lines)
```
Building /c/Temp/Basic_Blink/out/esp8266/debug/lib/clib-App-83f1f006d199462a6fb5fb6198612f89.a
AS /C/tools/sming/Sming/Arch/esp8266/Components/gdbstub/appcode/gdbstub-entry.S
C+ /c/Temp/Basic_Blink/app/application.cppCC /C/tools/sming/Sming/Components/rboot/appcode/rboot-overrides.c

CC /C/tools/sming/Sming/Components/rboot/rboot/appcode/rboot-api.c
CC /C/tools/sming/Sming/Components/rboot/rboot/appcode/rboot-bigflash.c
C+ /C/tools/sming/Sming/Components/rboot/host/rboot.cpp
C+ /C/tools/sming/Sming/Arch/esp8266/Components/gdbstub/appcode/gdb_hooks.cpp
c:/tools/sming/Sming/Components/rboot/host/rboot.cpp:13:29: fatal error: hostlib/hostmsg.h: No such file or directory
compilation terminated.
make[1]: *** [Components/rboot/host/rboot.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [App-build] Error 2

c:\Temp\Basic_Blink>
```

kwis2, 2020-01-17:
@icodk I don't know your dir configuration, but there is a lot of references to `/tools/sming/Sming` in Sming and some time ago I gave up and decided to install Sming in dir `/tools/sming/Sming` (windows of course)
Btw. there is a lot of configurations to set and I found that running Basic_Blink it is not enough to be sure that Sming works. You'll be able to say Sming works when you run the Basic_WebSkeletonApp and Basic_rBoot examples

icodk, 2020-01-17:
@kwis2 

>  decided to install Sming in dir /tools/sming/Sming 

So you run the git pull in this directory to 'install' sming ?
`c:\ tools\sming\Sming>git clone https://github.com/SmingHub/Sming`
or
`c:\ tools\sming>git clone https://github.com/SmingHub/Sming`

Thanks

mikee47, 2020-01-17:
@icodk See #2000, may be related to your issues

Specifically, your working directory looks like `c:\Temp\Basic_Blink` (lower-case drive letter 'c', as indicated by the `c:\Temp\Basic_Blink>` prompt) but SMING_HOME is `C:\tools\sming\Sming` (upper-case 'C').

To fix this:

```
make components-clean clean
set SMING_HOME=c:\tools\sming\Sming
```

Then rebuild.

kwis2, 2020-01-17:
in my case it is `c:\ tools>git clone git clone https://github.com/SmingHub/Sming sming` where last `sming` it is dir name
and then check your windows PATH  if SMING_HOME and ESP_HOME are proper set

mikee47, 2020-01-19:
You may also need a `make config-clean` to wipe the incorrect cached values.

icodk, 2020-02-16:
Fresh installation on Windows 10(today)
try to compile Basic_Blink got the following error (showing the last few lines of the build ):
```

CC rboot-stage2a.c
LD /C/tools/Sming/Sming/out/Esp8266/debug/build/rboot/rboot-stage2a.elf
E2 /C/tools/Sming/Sming/out/Esp8266/debug/build/rboot/rboot-hex2a.h
CC rboot.c
LD /C/tools/Sming/Sming/out/Esp8266/debug/build/rboot/rboot.elf
E2 /c/tools/Sming/samples/Basic_Blink/out/Esp8266/debug/firmware/rboot.bin
make[1]: Leaving directory `/C/tools/Sming/Sming/Components/rboot/rboot'
Prepare libmain
LDGEN out/Esp8266/debug/build/rom0.ld
Basic_Blink: Linking out/Esp8266/debug/build/app_0.out
 - Cannot openmake: *** [out/Esp8266/debug/build/app_0.out] Error 1

C:\tools\Sming\samples\Basic_Blink>
```

icodk, 2020-02-16:
Development branch

kmihaylov, 2020-06-01:
@icodk Some time has passed since your last comment? Some progress?

icodk, 2020-06-01:
@kmihaylov Nice hearing from you. No I didn't try recently. I hooked up on another project,while my esp8266 still runs on Sming 2.0. Wanted to upgrade at that time but decided to wait until 
you smoose it out:-)

kmihaylov, 2020-06-02:
Have you tried to clean the whole toolchain and project and to run it again with the latest version? Also I notice that you use ```make -j```. In the past I had problems with the -j flags (despite it was on other projects and long time ago).

I'm curious about the message ```cannot openmake```. Is everything with the paths OK?

So I suggest:
- git clone latest version
- recheck paths
- make w/o -j

icodk, 2020-06-02:
Thanks for the suggestion. I will give it a try. Hopefully sometimes next week

VooDooSi, 2020-06-03:
I was creating new virtual machine with windows 10 x64 for sming today. Followed manual install  (https://sming.readthedocs.io/en/latest/arch/esp8266/getting-started/windows-manual.html)

This happens for all sample projects: 
`make -j12 all 

Basic_rBoot: Invoking 'all' for Esp8266 (debug) architecture

Building /c/tools/Basic_rBoot/out/Esp8266/debug/lib/clib-App-6903af82e680ef61e341721f1de501b4.a
AS /C/tools/sming/Sming/Arch/Esp8266/Components/gdbstub/appcode/gdbstub-entry.S
CC /C/tools/sming/Sming/Components/rboot/appcode/rboot-overrides.c
CC /C/tools/sming/Sming/Components/rboot/rboot/appcode/rboot-api.c
CC /C/tools/sming/Sming/Components/rboot/rboot/appcode/rboot-bigflash.c
C+ /C/tools/sming/Sming/Arch/Esp8266/Components/gdbstub/appcode/gdb_hooks.cpp
C+ /c/tools/Basic_rBoot/app/application.cpp
AR /c/tools/Basic_rBoot/out/Esp8266/debug/lib/clib-App-6903af82e680ef61e341721f1de501b4.a
make[1]: Entering directory `/C/tools/sming/Sming/Components/rboot/rboot'
E2 /c/tools/Basic_rBoot/out/Esp8266/debug/firmware/rboot.bin
make[1]: Leaving directory `/C/tools/sming/Sming/Components/rboot/rboot'
Prepare libmain
LDGEN out/Esp8266/debug/build/rom0.ld
Basic_rBoot: Linking out/Esp8266/debug/build/app_0.out
 - Cannot openmake: *** [out/Esp8266/debug/build/app_0.out] Error 1
"make -j12 all" terminated with exit code 2. Build might be incomplete.`

I have tried building without "j", cloned latest version (develop), rechecked paths. Everything looks ok, except not building the project :D

mikee47, 2020-06-04:
>     * Cannot openmake: *** [out/Esp8266/debug/build/app_0.out] Error 1
>       "make -j12 all" terminated with exit code 2. Build might be incomplete.`

Python needs to be in the path. Try typing `python` and see what happens...

When I tried this on a fresh VM python got installed here:

```
C:\Users\xxx\AppData\Local\Programs\Python\Python38\python.exe
```

Wherever it ends up will either need to be in the path:

```
setx PATH "C:\Users\xxx\AppData\Local\Programs\Python\Python38;%PATH%"

```

or for Sming-only, set the PYTHON environment variable:

```
setx PYTHON "/c/Users/Mike/AppData/Local/Programs/Python/Python38/python.exe"
```

(Currently this needs to be in MinGW format - regular DOS path doesn't work.)

icodk, 2020-06-04:
@mikee47 Thanks for the tip

closed 2021-11-16 by slaff