SmingHub/Sming #2023: Make error on Blink sample on Windows 10

HanSolo72, 2020-01-14:
Hi,

I tried my first steps with Sming with the Blink sample, but I can't build it. Here is the content of my cmd windows. I use Windows 10 1909 x64.

In file included from c:\espressif\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\5.2.0\random:42:0,
                 from c:\espressif\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\5.2.0\bits\stl_algo.h:66,
                 from c:\espressif\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\5.2.0\algorithm:62,
                 from system/m_printf.cpp:13:
c:\espressif\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\5.2.0\limits:1607:7: internal compiler error: Segmentation fault
       max() _GLIBCXX_USE_NOEXCEPT { return __FLT_MAX__; }
       ^
libbacktrace could not find executable to open
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
make[2]: *** [out/build/system/m_printf.o] Error 1
make[2]: Leaving directory `/C/tools/sming/Sming'
make[1]: *** [reload] Error 2
make[1]: Leaving directory `/C/tools/sming/Sming'
make: *** [/C/tools/sming/Sming/compiler/lib//libsming.a] Error 2

aemseemann, 2020-01-15:
> c:\espressif\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\5.2.0\limits:1607:7: internal compiler error: Segmentation fault

Seems the compiler (g++) crashed. That should not happen under any circumstances, no matter how  faulty the source file may be.
Are you sure the compiler is installed correctly? Does it work for other projects?
Can you try reinstalling it and try a clean build afterwards?

HanSolo72, 2020-01-15:
I installed everything described here:
https://sming.readthedocs.io/en/latest/arch/esp8266/getting-started/windows.html

I just reinstalled choco  with this command from a administrative cmd window.
@powershell -NoProfile -ExecutionPolicy unrestricted -Command "iex ((new-object net.webclient).DownloadString('https://chocolatey.org/install.ps1'))" && SET PATH=%PATH%;%ALLUSERSPROFILE%\chocolatey\bin

Then I installed Sming (what includes the compiler) with this command (Powershell):
choco install -y sming --force

But during the installation I saw many errors like this one:
mingw-get.exe: *** ERROR *** C:\tools\mingw64\/msys/1.0/share/man/man1/find.1: extraction failed
mingw-get.exe: *** ERROR *** C:\tools\mingw64\/msys/1.0/share/man/man1/find.1: probable package conflict; existing file not overwritten

Slashes and backslashes are mixed, so I downloaded the online installer for mingw64 from this source: https://sourceforge.net/projects/mingw-w64/files/Toolchains%20targetting%20Win32/Personal%20Builds/mingw-builds/installer/mingw-w64-install.exe

I tried another build and got the same error:

In file included from c:\espressif\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\5.2.0\random:42:0,
                 from c:\espressif\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\5.2.0\bits\stl_algo.h:66,
                 from c:\espressif\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\5.2.0\algorithm:62,
                 from system/m_printf.cpp:13:
c:\espressif\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\5.2.0\limits:1607:7: internal compiler error: Segmentation fault
       max() _GLIBCXX_USE_NOEXCEPT { return __FLT_MAX__; }
       ^
libbacktrace could not find executable to open
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
make[2]: *** [out/build/system/m_printf.o] Error 1
make[2]: Leaving directory `/C/tools/sming/Sming'
make[1]: *** [reload] Error 2
make[1]: Leaving directory `/C/tools/sming/Sming'
make: *** [/C/tools/sming/Sming/compiler/lib//libsming.a] Error 2

mikee47, 2020-01-16:
What's the output of your `PATH` variable? (`echo %PATH%`)

HanSolo72, 2020-01-17:
My %PATH% variable:
C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\Sophos\Sophos SSL VPN Client\bin;C:\Program Files\Eclipse Foundation\4.4.2\eclipse;C:\Program Files\Git\cmd;C:\Program Files\Microsoft SQL Server\110\Tools\Binn\;C:\Program Files\PuTTY\;C:\ProgramData\chocolatey\bin;C:\ProgramData\chocolatey\lib\mingw\tools\install\mingw64\bin;C:\Python38\;C:\Python38\Scripts\;C:\Users\Christof\AppData\Local\Microsoft\WindowsApps;C:\Users\Christof\AppData\Local\Programs\Microsoft VS Code\bin;C:\WINDOWS;C:\WINDOWS\System32\OpenSSH\;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\system32;C:\mingw\bin;C:\mingw\msys\1.0\bin;C:\tools\mingw64;C:\tools\mingw64\bin;C:\tools\mingw64\msys\1.0\bin

mikee47, 2020-01-18:
Firstly, the error is occuring with the Espressif compiler - not MinGW - so it's definitely worth re-installing that. The easiest way is to remove or rename the `C:\Espressif` directory then follow the instructions at https://sming.readthedocs.io/en/latest/arch/esp8266/getting-started/windows-manual.html#install-esp8266-toolchain.

Secondly, your path contains references to two different MinGW installations and there are probably conflicts between them. This should fix that issue:

```
set PATH=C:\mingw\bin;C:\mingw\msys\1.0\bin;%PATH%
```

I would suggest ditching Choco and going manual. See https://sming.readthedocs.io/en/latest/arch/esp8266/getting-started/windows-manual.html.

icodk, 2020-01-18:
@mikee47 May be to add the "Manual Windows Installation" chapter header to the menu in the documentation?
IMHO This is an  important chapter and should be accessible directly from the menu.

HanSolo72, 2020-01-18:
Ok, I renamed Espressif and MingGW and installed it manually like described. I deleted Choco

HanSolo72, 2020-01-18:
Ok, I renamed Espressif and MingGW and installed it manually like described. I also deleted Choco.

Now this is my Path:
```
C:\Python38\Scripts\;C:\Python38\;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Microsoft SQL Server\110\Tools\Binn\;C:\Program Files\PuTTY\;C:\Program Files\Git\cmd;C:\Program Files\Eclipse Foundation\4.4.2\eclipse;C:\mingw\bin;C:\mingw\msys\1.0\bin;C:\Users\Christof\AppData\Local\Microsoft\WindowsApps;C:\Program Files (x86)\Sophos\Sophos SSL VPN Client\bin;C:\Users\Christof\AppData\Local\Programs\Microsoft VS Code\bin;
```

```
C:\>where gcc
C:\MinGW\bin\gcc.exe

C:\>gcc --version
gcc (MinGW.org GCC-8.2.0-5) 8.2.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

And this happens when I build Basic_Blink:
```
make[2]: Leaving directory `/C/tools/sming/Sming/third-party/esp-open-lwip'
make[1]: Leaving directory `/C/tools/sming/Sming'
make[1]: Entering directory `/C/tools/sming/Sming'
make[2]: Entering directory `/C/tools/sming/Sming'
make[3]: Entering directory `/C/tools/sming/Sming/third-party/esp-open-lwip'
/C/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -Os -Iinclude -I/C/Espressif/ESP8266_SDK/include -Wl,-EL -mlongcalls -mtext-section-literals -I../../system/include -I../../Wiring -Wl,-EL -nostdlib -mlongcalls -mtext-section-literals -finline-functions -fdata-sections -ffunction-sections   -c -o lwip/core/def.o lwip/core/def.c
In file included from include/lwip/arch.h:43:0,
                 from include/lwip/debug.h:35,
                 from include/lwip/opt.h:46,
                 from lwip/core/def.c:39:
include/arch/cc.h:38:21: fatal error: c_types.h: No such file or directory
compilation terminated.
make[3]: *** [lwip/core/def.o] Error 1
make[3]: Leaving directory `/C/tools/sming/Sming/third-party/esp-open-lwip'
make[2]: *** [compiler/lib/liblwip_open.a] Error 2
make[2]: Leaving directory `/C/tools/sming/Sming'
make[1]: *** [reload] Error 2
make[1]: Leaving directory `/C/tools/sming/Sming'
make: *** [/C/tools/sming/Sming/compiler/lib//libsming.a] Error 2
```

mikee47, 2020-01-18:
You've updated over the top of a previous version of Sming; delete this directory:

```
/C/tools/sming/Sming/third-party/esp-open-lwip
```

See https://sming.readthedocs.io/en/latest/_inc/Sming/building.html#known-issues.
You can check for this using `git status`, or probably simpler is delete `C:\tools\sming` and pull it in again.

HanSolo72, 2020-01-18:
Thank you very much, now it works!

closed 2020-01-18 by HanSolo72

mikee47, 2020-01-18:
Most welcome. I'll try to improve the documentation so it covers these sorts of issues. Thanks for using Sming!