SmingHub/Sming #2025: RBOOT_TWO_ROMS calculation wrong

etmmahi, 2020-01-15:
In the file Sming/Components/rboot/component.mk the awk calculation for RBOOT_TWO_ROMS is wrong.
The % should be an / (division) 

```
 # determine number of roms to generate
 ifneq ($(RBOOT_ROM1_ADDR),)
-RBOOT_TWO_ROMS := $(shell $(AWK) 'BEGIN { print (ARGV[1] % (1024*1024)) != (ARGV[2] % (1024*1024))}' $(RBOOT_ROM0_ADDR) $(RBOOT_ROM1_ADDR))
+RBOOT_TWO_ROMS := $(shell $(AWK) 'BEGIN { print (ARGV[1] / (1024*1024)) != (ARGV[2] / (1024*1024))}' $(RBOOT_ROM0_ADDR) $(RBOOT_ROM1_ADDR))
 else
 RBOOT_TWO_ROMS := 0
 endif

```

The issue can be seen in the Basic_rBoot example

Martijn

aemseemann, 2020-01-15:
Two ROMs have to be linked if the offset within their 1MB flash region differs for `RBOOT_ROM0_ADDR` and `RBOOT_ROM1_ADDR`. These offset is calculated as the remainder of dividing by 1MB=1024*1024. Hence, I think '%' is correct here.

etmmahi, 2020-01-16:
You are right, I misinterpreted the TWO_ROMS define.

Martijn

closed 2020-01-16 by etmmahi