SmingHub/Sming #2026: DISABLE_SPIFFS flag not working

etmmahi, 2020-01-15:
When in the example Basic_rBoot in the component.mk

```
## option to completely disable spiffs
DISABLE_SPIFFS		?= 1
```

the DISABLE_SPIFFS is set, this is not reflected in the compiler flags during a build.

```
make config-clean
make clean
make VERBOSE=1
```

pse add something like:
-DDISABLE_SPIFFS=$(DISABLE_SPIFFS)
to the compiler flags

Martijn

mikee47, 2020-01-15:
https://sming.readthedocs.io/en/latest/_inc/Sming/Components/spiffs/index.html

etmmahi, 2020-01-15:
This is a snippet from the rboot sample. It can expect DISABLE_SPIFFS to be set 

```
#ifndef DISABLE_SPIFFS
			Serial.println("  ls - list files in spiffs");
			Serial.println("  cat - show first file in spiffs");
#endif
```
But the flag is not given to the GCC compiler.

Furthermore the 'readthedocs' says  

 DISABLE_SPIFFS

    0
        Disable filesystem generation
    1 (default)
        Enable filesystem generation 

Where setting  DISABLE_SPIFFS=1 means **enable** ?? A bit confusing.

Martijn

mikee47, 2020-01-15:
Ah, yes the docs are wrong. The point is is only determines if a filesystem image is generated for upload to the device. It doesn't disable SPIFFS. (Yes, I agree the name is confusing - historical.)

etmmahi, 2020-01-15:
The basic_rboot sample expects the flag to be set or not (conditional code with #ifdef / #endif) 
Can you add an automatic -DDISABLE_SPIFFS=$(DISABLE_SPIFFS) to the compiler flags when it is set?
The the basic-rboot sample is correct

mikee47, 2020-01-15:
Yes, that would be logical. I did that for `Basic_WebSkeletonApp`, can't think why I didn't apply it universallly.

closed 2020-01-15 by slaff