SmingHub/Sming #2029: Sming crash after flash

HanSolo72, 2020-01-19:
After flashing Basic_Blink and also Basic_SmartConfig, I get the following on my serial monitor. I flashed with theses settings:
make SMING_ARCH=Esp8266 SPI_MODE=dio SPI_SIZE=4M COM_SPEED_ESPTOOL=74880
make flash

```
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 1100, room 16
tail 12
chksum 0x23
ho 0 tail 12 room 4
load 0x3ffe8000, len 576, room 12
tail 4
chksum 0x32
csum 0x32

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz

Booting rom 0 at 2000, load addr 3b780.
ota1 not set
ota2 not set
rf_cal[0] !=0x05,is 0xFF

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 1100, room 16
tail 12
chksum 0x23
ho 0 tail 12 room 4
load 0x3ffe8000, len 576, room 12
tail 4
chksum 0x32
csum 0x32

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz

Booting rom 0 at 2000, load addr 3b780.
ota1 not set
ota2 not set
rf_cal[0] !=0x05,is 0xFF
rf cal sector: 1019
freq trace enable 0
rf[112] : ff
rf[113]mode : sta(ec:fa:bc:63:2d:49)
add if0

***** Fatal exception 28 (LOAD_PROHIBITED)
pc=0x4020d616 sp=0x3ffffa60 excvaddr=0x00000000
ps=0x00000030 sar=0x00000020 vpri=0x00000000
r00: 0x4020d613=1075893779 r01: 0x3ffffa60=1073740384 r02: 0x3ffed7d4=1073666004
r03: 0x0000010f=       271 r04: 0x00000001=         1 r05: 0x00000010=        16
r06: 0x3ff1fe00=1072823808 r07: 0x00000010=        16 r08: 0x00000000=         0
r09: 0x00000000=         0 r10: 0x00000100=       256 r11: 0x00000100=       256
r12: 0x3ffed654=1073665620 r13: 0x00000000=         0 r14: 0x40202528=1075848488
r15: 0x3ffedd5c=1073667420

Stack dump:
To decode the stack dump call from command line:
   make decode-stacktrace
and copy & paste the text enclosed in '===='.

================================================================
3ffffa60:  4020fa6d 3ffffab0 0000010f 3ffed7d0
3ffffa70:  3ffed7d4 0000010f 3ffef500 4021c27c
3ffffa80:  40202528 40219ded 00000001 3ffe89cc
3ffffa90:  3ffe89a0 3ffef500 3ffef500 00000000
3ffffaa0:  3ffffab0 40202528 3ffef500 402027cf
3ffffab0:  00000000 00000000 00000100 00000000
3ffffac0:  00000000 00000000 402175d9 4021755b
3ffffad0:  3ffef198 40100946 000003fd 0000001f
3ffffae0:  ffffffff 40001100 00000001 3ffef248
3ffffaf0:  00000001 401007af 00000003 560000c0
3ffffb00:  ffffffff ffffffff ffff0002 00000000
3ffffb10:  00000000 00000000 00000000 00000000
3ffffb20:  00000000 00000000 00000000 00000000
3ffffb30:  ffffffff 00ffffff 00000000 00000000
3ffffb40:  00000000 00000000 00000000 00000000
3ffffb50:  00000000 00000000 00000000 00000000
3ffffb60:  00000000 00000000 00000000 00000000
3ffffb70:  00000000 ff000000 ffffffff ffffffff
3ffffb80:  ffffffff ffffffff ffffffff ffffffff
3ffffb90:  ffffffff ffffffff ffffffff ffffffff
3ffffba0:  ffffffff ffffffff ffff00ff ffffffff
3ffffbb0:  0000000a 2d505345 44323336 00003934
3ffffbc0:  00000000 00000000 00000000 00000000
3ffffbd0:  00000000 00000000 00000000 00000000
3ffffbe0:  00000000 00000000 00000000 00000000
3ffffbf0:  00000000 00000000 00000000 00000000
3ffffc00:  00000000 00000000 00000000 00000000
3ffffc10:  00000000 ffffffff ffffffff ffffffff
3ffffc20:  ffffffff ffffffff ffffffff ffffffff
3ffffc30:  ffffffff 000001ff ffffff04 ffffff01
3ffffc40:  ffffffff ffffffff ffffffff ffffffff
3ffffc50:  ffffffff ffffffff ffffffff ffffffff
3ffffc60:  ffffffff ffffffff ffffffff ffffffff
3ffffc70:  ffffffff ffffffff ffffffff ffffffff
3ffffc80:  ffffffff ffffffff ffffffff ffffffff
3ffffc90:  ffffffff ffffffff ffffffff ffffffff
3ffffca0:  ffffffff ffffffff ffffffff ffffffff
3ffffcb0:  ffffffff ffffffff ffffffff ffffffff
3ffffcc0:  ffffffff ffffffff ffffffff ffffffff
3ffffcd0:  ffffffff ffffffff ffffffff ffffffff
3ffffce0:  ffffffff ffffffff ffffffff ffffffff
3ffffcf0:  ffffffff ffffffff ffffffff ffffffff
3ffffd00:  ffffffff ffffffff ffffffff ffffffff
3ffffd10:  ffffffff ffffffff ffffffff ffffffff
3ffffd20:  ffffffff ffffffff ffffffff ffffffff
3ffffd30:  ffffffff ffffffff ffffffff ffffffff
3ffffd40:  ffffffff ffffffff ffffffff ffffffff
3ffffd50:  ffffffff ffffffff ffffffff ffffffff
3ffffd60:  ffffffff ffffffff ffffffff ffffffff
3ffffd70:  ffffffff ffffffff ffffffff ffffffff
3ffffd80:  ffffffff ffffffff ffffffff ffffffff
3ffffd90:  ffffffff ffffffff ffffffff ffffffff
3ffffda0:  ffffffff ffffffff ffffffff ffffffff
3ffffdb0:  ffffffff ffffffff ffffffff ffffffff
3ffffdc0:  ffffffff ffffffff ffffffff ffffffff
3ffffdd0:  ffffffff ffffffff ffffffff ffffffff
3ffffde0:  ffffffff ffffffff ffffffff ffffffff
3ffffdf0:  ffffffff ffffffff ffffffff ffffffff
3ffffe00:  ffffffff ffffffff ffffffff ffffffff
3ffffe10:  ffffffff ffffffff ffffffff ffffffff
3ffffe20:  ffffffff ffffffff ffffffff ffffffff
3ffffe30:  ffffffff ffffffff ffffffff ffffffff
3ffffe40:  ffffffff 00000003 ffffffff ffffffff
3ffffe50:  ffffffff ffffffff ffffffff ffffffff
3ffffe60:  ffffffff ffffffff ffffffff ffff0064
3ffffe70:  00000000 00000000 ffffffff ffffffff
3ffffe80:  ffffffff ffffffff ffffffff ffffffff
3ffffe90:  ffffffff ffffffff ffffffff ffffffff
3ffffea0:  ffffffff ffffffff ffffffff ffffffff
3ffffeb0:  ffffffff ffffffff ffffffff ffffffff
3ffffec0:  ffffffff ffffffff ffffffff ffffffff
3ffffed0:  ffffffff ffffffff ffffffff ffffffff
3ffffee0:  ffffffff ffffffff ffffffff ffffffff
3ffffef0:  ffffffff ffffffff ffffffff ffffffff
3fffff00:  ffffffff ffffffff ffffffff ffffffff
3fffff10:  ffffffff ffffffff ffffffff ffffffff
3fffff20:  ffffffff ffffffff ffffffff ffffffff
3fffff30:  ffffffff ffffffff ffffffff ffffffff
3fffff40:  ffffffff ffffffff ffffffff ffffffff
3fffff50:  ffffffff ffffffff ffffffff ffffffff
3fffff60:  ffffffff ffffffff ffffffff ffffffff
3fffff70:  ffffffff ffffffff ffffffff ffffffff
3fffff80:  ffffffff ffffffff ffffffff ffffffff
3fffff90:  faecffff 492d63bc ffff0081 0000043c
3fffffa0:  400202e9 0c0f6856 ffffff01 ffffffff

================================================================
To decode the stack dump call from command line:
   make decode-stacktrace
and copy & paste the text enclosed in '===='.
```

mikee47, 2020-01-19:
Offending error is `rf_cal[0] !=0x05,is 0xFF`. You need to run `make flashinit` to write the device configuration, then re-flash. (Something else which I think is missing from the docs. Sorry.)

mikee47, 2020-01-19:
https://sming.readthedocs.io/en/latest/troubleshooting/random-restart.html

HanSolo72, 2020-01-19:
Thank you, now it works.

closed 2020-01-19 by HanSolo72