SmingHub/Sming #2036: How to start my Project and problem with compiling Basic_WebSkeletonApp

HanSolo72, 2020-01-30:
After I set up the Sming environment and I'm able to build some of the samples, I need some help from experienced developers for my first NodeMCU project.

In my house I have a bunch of windows and doors with reed contacts that I want to monitor (MQTT to ioBroker). I will also connect some temperature sensors (D18B20). My project should include:

- Wifi Config
- PIN expander (MCP23017)
- NTP
- MQTT
- temperature sensors (D18B20)
- Show temperatures and window states (open/closed with names) on internal website

I think the Basic_WebSkeletonApp is a good start and all I have to do is to expand it with MCP23017, NTP, MQTT and D18B20.

My questions are:
1. Is this a good way to start for my project?
2. I tried to compile the Basic_WebSkeletonApp without any changes. But it don't work. Other samples work, but not this one.

```
C:\tools\Sming\samples\Basic_WebSkeletonApp>make flash

Basic_WebSkeletonApp: Invoking 'flash' for Esp8266 (debug) architecture

Building /c/tools/Sming/samples/Basic_WebSkeletonApp/out/Esp8266/debug/lib/clib-App-7372ec6ba11e0cfd4693f13ff14eafbf.a
C+ /c/tools/Sming/samples/Basic_WebSkeletonApp/app/webserver.cpp
In file included from c:/tools/Sming/samples/Basic_WebSkeletonApp/app/webserver.cpp:3:0:
c:/tools/Sming/Sming/Components/FlashString/src/include/FlashString/Map.hpp:108:79: error: the value of 'CONTENT_bootstrap' is not usable in a constant expression
  } FSTR_PACKED FSTR_ALIGNED name PROGMEM = {{sizeof(name.data)}, {__VA_ARGS__}};                                    \
                                                                               ^
c:/tools/Sming/Sming/Components/FlashString/src/include/FlashString/Map.hpp:91:2: note: in expansion of macro 'DEFINE_FSTR_MAP_DATA_SIZED'
  DEFINE_FSTR_MAP_DATA_SIZED(name, KeyType, ContentType,                                                             \
  ^
c:/tools/Sming/Sming/Components/FlashString/src/include/FlashString/Map.hpp:60:9: note: in expansion of macro 'DEFINE_FSTR_MAP_DATA'
  static DEFINE_FSTR_MAP_DATA(FSTR_DATA_NAME(name), KeyType, ContentType, __VA_ARGS__);                              \
         ^
c:/tools/Sming/samples/Basic_WebSkeletonApp/app/webserver.cpp:32:1: note: in expansion of macro 'DEFINE_FSTR_MAP_LOCAL'
 DEFINE_FSTR_MAP_LOCAL(fileMap, FlashString, FlashString, FILE_LIST(XX));
 ^
In file included from c:/tools/Sming/Sming/Components/FlashString/src/include/FlashString/String.hpp:24:0,
                 from c:/tools/Sming/Sming/Wiring/WString.h:64,
                 from c:/tools/Sming/Sming/Wiring/Print.h:24,
                 from c:/tools/Sming/Sming/Wiring/WiringFrameworkIncludes.h:18,
                 from c:/tools/Sming/Sming/Core/SmingCore.h:16,
                 from c:/tools/Sming/samples/Basic_WebSkeletonApp/include/configuration.h:3,
                 from c:/tools/Sming/samples/Basic_WebSkeletonApp/include/tytherm.h:3,
                 from c:/tools/Sming/samples/Basic_WebSkeletonApp/app/webserver.cpp:1:
c:/tools/Sming/samples/Basic_WebSkeletonApp/app/webserver.cpp:26:36: note: 'CONTENT_bootstrap' was not declared 'constexpr'
 #define XX(name, file) IMPORT_FSTR(CONTENT_##name, PROJECT_DIR "/files/" file);
                                    ^
c:/tools/Sming/Sming/Components/FlashString/src/include/FlashString/Object.hpp:49:69: note: in definition of macro 'DEFINE_FSTR_REF'
 #define DEFINE_FSTR_REF(name, ObjectType, object) const ObjectType& name PROGMEM = object.template as<ObjectType>();
                                                                     ^
c:/tools/Sming/Sming/Components/FlashString/src/include/FlashString/String.hpp:133:33: note: in expansion of macro 'IMPORT_FSTR_OBJECT'
 #define IMPORT_FSTR(name, file) IMPORT_FSTR_OBJECT(name, FSTR::String, file)
                                 ^
c:/tools/Sming/samples/Basic_WebSkeletonApp/app/webserver.cpp:26:24: note: in expansion of macro 'IMPORT_FSTR'
 #define XX(name, file) IMPORT_FSTR(CONTENT_##name, PROJECT_DIR "/files/" file);
                        ^
c:/tools/Sming/samples/Basic_WebSkeletonApp/app/webserver.cpp:13:2: note: in expansion of macro 'XX'
  XX(bootstrap, "bootstrap.min.css.gz")                                                                              \
  ^
c:/tools/Sming/samples/Basic_WebSkeletonApp/app/webserver.cpp:27:1: note: in expansion of macro 'FILE_LIST'
 FILE_LIST(XX)
 ^
In file included from c:/tools/Sming/samples/Basic_WebSkeletonApp/app/webserver.cpp:3:0:
c:/tools/Sming/Sming/Components/FlashString/src/include/FlashString/Map.hpp:108:79: error: the value of 'CONTENT_config_html' is not usable in a constant expression
  } FSTR_PACKED FSTR_ALIGNED name PROGMEM = {{sizeof(name.data)}, {__VA_ARGS__}};                                    \
                                                                               ^
c:/tools/Sming/Sming/Components/FlashString/src/include/FlashString/Map.hpp:91:2: note: in expansion of macro 'DEFINE_FSTR_MAP_DATA_SIZED'
  DEFINE_FSTR_MAP_DATA_SIZED(name, KeyType, ContentType,                                                             \
  ^
[... 166 lines cut ...]
                 from c:/tools/Sming/Sming/Core/SmingCore.h:16,
                 from c:/tools/Sming/samples/Basic_WebSkeletonApp/include/configuration.h:3,
                 from c:/tools/Sming/samples/Basic_WebSkeletonApp/include/tytherm.h:3,
                 from c:/tools/Sming/samples/Basic_WebSkeletonApp/app/webserver.cpp:1:
c:/tools/Sming/samples/Basic_WebSkeletonApp/app/webserver.cpp:26:36: note: 'CONTENT_jquery' was not declared 'constexpr'
 #define XX(name, file) IMPORT_FSTR(CONTENT_##name, PROJECT_DIR "/files/" file);
                                    ^
c:/tools/Sming/Sming/Components/FlashString/src/include/FlashString/Object.hpp:49:69: note: in definition of macro 'DEFINE_FSTR_REF'
 #define DEFINE_FSTR_REF(name, ObjectType, object) const ObjectType& name PROGMEM = object.template as<ObjectType>();
                                                                     ^
c:/tools/Sming/Sming/Components/FlashString/src/include/FlashString/String.hpp:133:33: note: in expansion of macro 'IMPORT_FSTR_OBJECT'
 #define IMPORT_FSTR(name, file) IMPORT_FSTR_OBJECT(name, FSTR::String, file)
                                 ^
c:/tools/Sming/samples/Basic_WebSkeletonApp/app/webserver.cpp:26:24: note: in expansion of macro 'IMPORT_FSTR'
 #define XX(name, file) IMPORT_FSTR(CONTENT_##name, PROJECT_DIR "/files/" file);
                        ^
c:/tools/Sming/samples/Basic_WebSkeletonApp/app/webserver.cpp:18:2: note: in expansion of macro 'XX'
  XX(jquery, "jquery-2.1.4.min.js.gz")
  ^
c:/tools/Sming/samples/Basic_WebSkeletonApp/app/webserver.cpp:27:1: note: in expansion of macro 'FILE_LIST'
 FILE_LIST(XX)
 ^
make[1]: *** [app/webserver.o] Error 1
make: *** [App-build] Error 2
```

mikee47, 2020-01-30:
The problem is due to #2013, you need to update line 26 in `webserver.cpp` from `IMPORT_FSTR` to `IMPORT_FSTR_LOCAL`. I'll raise a PR to fix this.

The sample will build if you `make DISABLE_SPIFFS=0` which is the default.

mikee47, 2020-01-31:
Something else to note is that the above will compile without issue if you use the [latest toolchain](https://sming.readthedocs.io/en/latest/arch/esp8266/getting-started/eqt.html).

closed 2020-06-24 by slaff