SmingHub/Sming #2042: Allow to specify maximum ROM size

fooker, 2020-02-04:
The new mechanism to generate linker scripts does not allow to specify a maximum ROM size.

I used to limit the `irom0_0_seg` length to `(1M / 2 - 0x2010)` which allows to ensure that the first rom will now overwrite the second one. For me, both ROMs should have the same `0x2010` offset from the base addresses (0x00000 for ROM 0 and 0x80000 for ROM 1). The remaining space between the two ROMs is used to store some persistent data.

It would be nice to have a way to specify this limit again. Or as an alternative, overwrite the generative linker scripts with provided ones.

fooker, 2020-02-05:
See #2044 for a possible solution.

slaff, 2021-03-26:
With the arrival of (partition tables)[https://sming.readthedocs.io/en/latest/_inc/Sming/Components/Storage/index.html] this should be possible. Feel free to re-open this ticket if you have problems achieving what you want with the latest develop version.

closed 2021-03-26 by slaff