SmingHub/Sming #2047: axtls-8266 Sming v 4.1 make bug

kwis2, 2020-02-06:
Sming 4.1, Windows 10, fresh Sming instalation. 
While make v 4.0 Sming working program there is problem with axtls-8266 component

> Fetching submodule '/C/tools/sming/Sming/Components/axtls-8266/axtls-8266' ...
> Submodule path 'axtls-8266': checked out '6c3e87b307ffb67e670909a3c5ac613032eed0ba'
> Submodule path 'axtls-8266/compat': checked out 'bc2bf7b61b9295b07d2fd9b5a9bb7824b1677c85'
> error: corrupt patch at line 411
> make: *** [/C/tools/sming/Sming/Components/axtls-8266/axtls-8266/.submodule] Error 128

slaff, 2020-02-07:
The CI system hasn't detected such a problem. Excerpt from the appveyor log.
```
Fetching submodule '/c/projects/sming-sb483/Sming/Components/axtls-8266/axtls-8266' ...
Submodule 'ESP8266.axtls' (https://github.com/igrr/axtls-8266.git) registered for path 'axtls-8266'
Cloning into 'C:/projects/sming-sb483/Sming/Components/axtls-8266/axtls-8266'...
Submodule path 'axtls-8266': checked out '6c3e87b307ffb67e670909a3c5ac613032eed0ba'
Submodule 'compat' (https://github.com/attachix/lwirax.git) registered for path 'axtls-8266/compat'
Cloning into 'C:/projects/sming-sb483/Sming/Components/axtls-8266/axtls-8266/compat'...
Submodule path 'axtls-8266/compat': checked out 'bc2bf7b61b9295b07d2fd9b5a9bb7824b1677c85'
Checking patch crypto/crypto_misc.c...
Checking patch replacements/time.c...
Checking patch ssl/tls1.c...
Checking patch crypto/bigint.c...
Checking patch ssl/crypto_misc.h...
Checking patch ssl/tls1.h...
Checking patch ssl/loader.c...
Checking patch ssl/x509.c...
Checking patch ssl/gen_cert.c...
Checking patch ssl/p12.c...
Checking patch crypto/crypto.h...
Applied patch crypto/crypto_misc.c cleanly.
Applied patch replacements/time.c cleanly.
Applied patch ssl/tls1.c cleanly.
Applied patch crypto/bigint.c cleanly.
Applied patch ssl/crypto_misc.h cleanly.
Applied patch ssl/tls1.h cleanly.
Applied patch ssl/loader.c cleanly.
Applied patch ssl/x509.c cleanly.
Applied patch ssl/gen_cert.c cleanly.
Applied patch ssl/p12.c cleanly.
Applied patch crypto/crypto.h cleanly.
Building /c/projects/sming-sb483/Sming/out/Esp8266/debug/lib/clib-libc.a
make[2]: Nothing to be done for `build'.
```

> Submodule path 'axtls-8266': checked out '6c3e87b307ffb67e670909a3c5ac613032eed0ba'

`6c3e87b307ffb67e670909a3c5ac613032eed0ba` is the commit that we use. So far so good.

> error: corrupt patch at line 411

That seems to be the problem. Can you paste here the rejected lines? In addition can you tell us your Windows version, your git version and also run the following command: 
```
cd %SMING_HOME%
git diff develop master
```

kwis2, 2020-02-07:
I have really fresh Sming installation ... Windows 10 64 bit 

> C:\tools\sming\Sming>git update-git-for-windows
> Git for Windows 2.25.0.windows.1 (64bit)
> Up to date

> C:\tools\sming\Sming>git diff develop master
> fatal: ambiguous argument 'master': unknown revision or path not in the working tree.
> Use '--' to separate paths from revisions, like this:
> 'git <command> [<revision>...] -- [<file>...]'

and

> C:\tools\sming\Sming>git status
> On branch develop
> Your branch is up to date with 'origin/develop'.
> 
> Changes not staged for commit:
>   (use "git add <file>..." to update what will be committed)
>   (use "git checkout -- <file>..." to discard changes in working directory)
> 
>         modified:   ../samples/Basic_WebSkeletonApp/app/configuration.cpp
> 
> no changes added to commit (use "git add" and/or "git commit -a")

In my oppinion it is not problem with windows, git or so.

after `make dist-clean` I got like below, what mean another components are ok. @slaff I dont exactly know what do you mean "Can you paste here the rejected lines?"

> CO2_sensor: Invoking 'all' for Esp8266 (debug) architecture
> 
> Fetching submodule '/C/tools/sming/Sming/Components/rboot/rboot' ...
> Submodule path 'rboot': checked out '24f6021da82b303262a939c955f6a984bce6208f'
> Checking patch Makefile...
> Checking patch rboot.c...
> Checking patch appcode/rboot-api.c...
> Applied patch Makefile cleanly.
> Applied patch rboot.c cleanly.
> Applied patch appcode/rboot-api.c cleanly.
> 
> Fetching submodule '/C/tools/sming/Sming/Components/rboot/esptool2' ...
> Submodule path 'esptool2': checked out 'a550572b9de8c25f26b7a7675b47e58861227f64'
> Checking patch elf.h...
> Checking patch esptool2.c...
> Checking patch esptool2.h...
> Checking patch Makefile...
> Applied patch elf.h cleanly.
> Applied patch esptool2.c cleanly.
> Applied patch esptool2.h cleanly.
> Applied patch Makefile cleanly.
> 
> Fetching submodule '/C/tools/sming/Sming/Arch/Esp8266/Components/esp8266/ESP8266_NONOS_SDK' ...
> rm: cannot remove directory `ESP8266_NONOS_SDK/documents/CN': Directory not empty
> Submodule path 'ESP8266_NONOS_SDK': checked out '214129bf542c832006bd26ad0ff21a3da7465ce6'
> Checking patch lib/libc.a => lib/libc.orig.a...
> Checking patch lib/libgcc.a => lib/libgcc.orig.a...
> Applied patch lib/libc.a => lib/libc.orig.a cleanly.
> Applied patch lib/libgcc.a => lib/libgcc.orig.a cleanly.
> 
> Fetching submodule '/C/tools/sming/Sming/Arch/Esp8266/Components/driver/new-pwm' ...
> Submodule path 'new-pwm': checked out '19cb69e9ce5071686d0e2a28962fd06e11d03a61'
> Checking patch pwm.c...
> Applied patch pwm.c cleanly.
> 
> Fetching submodule '/C/tools/sming/Sming/Arch/Esp8266/Components/esptool/esptool' ...
> Submodule path 'esptool': checked out 'f2560887541414de54dbfb4368cdfc3fdd62e019'
> 
> Fetching submodule '/C/tools/sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip' ...
> Submodule path 'esp-open-lwip': checked out 'b9a23a6e9ef88d995ae74014e71a6a4227f61d62'
> Checking patch include/user_config.h...
[... 44 lines cut ...]
> Applied patch src/spiffs_nucleus.c cleanly.
> 
> Fetching submodule '/C/tools/sming/Sming/Components/http-parser' ...
> Submodule path 'http-parser': checked out 'edeedb1b4d2f34e4c7d8045ac8b92adbc35e7ed7'
> Checking patch http_parser.c...
> Applied patch http_parser.c cleanly.
> 
> Fetching submodule '/C/tools/sming/Sming/Components/ws_parser' ...
> Submodule path 'ws_parser': checked out '5d3fa49e2e291f38340249e3150c40184244b7da'
> Checking patch ws_parser.c...
> Applied patch ws_parser.c cleanly.
> 
> Fetching submodule '/C/tools/sming/Sming/Components/mqtt-codec' ...
> Submodule path 'mqtt-codec': checked out 'b9fed830a33651dc9fd873d589ee26f85e74077b'
> Checking patch src/platform.h...
> Applied patch src/platform.h cleanly.
> 
> Fetching submodule '/C/tools/sming/Sming/Components/libyuarel' ...
> Submodule path 'libyuarel': checked out '463760fb101861d760eae9b46dbf7f8dfaa30179'
> 
> Fetching submodule '/C/tools/sming/Sming/Components/axtls-8266/axtls-8266' ...
> Submodule path 'axtls-8266': checked out '6c3e87b307ffb67e670909a3c5ac613032eed0ba'
> Submodule path 'axtls-8266/compat': checked out 'bc2bf7b61b9295b07d2fd9b5a9bb7824b1677c85'
> error: corrupt patch at line 411
> make: *** [/C/tools/sming/Sming/Components/axtls-8266/axtls-8266/.submodule] Error 128

mikee47, 2020-02-07:
Can you try opening `C:\tools\sming\Sming\Components\axtls-8266\axtls-8266.patch` in a text editor, then on line 411 (which is blank) type a single space character. Save it then try again; does that fix the problem?

kwis2, 2020-02-07:
space fixed problem. I commit `axtls-8266.patch` file.

kwis2, 2020-02-11:
It looks like there is more windows bugs or so.
I make HostTests and got 

```
C:\tools\sming\tests\HostTests>make
..................

..................
C+ /c/tools/sming/tests/HostTests/app/test-serial.cpp
C+ /c/tools/sming/tests/HostTests/app/test-spiffs.cpp
C+ /c/tools/sming/tests/HostTests/app/test-string.cpp
In file included from c:/tools/sming/Sming/Wiring/WString.h:64:0,
                 from c:/tools/sming/Sming/Wiring/Print.h:24,
                 from c:/tools/sming/Sming/Wiring/WiringFrameworkIncludes.h:18,
                 from c:/tools/sming/Sming/Core/SmingCore.h:16,
                 from c:/tools/sming/Sming/Libraries/SmingTest/include/TestGroup.h:24,
                 from c:/tools/sming/Sming/Libraries/SmingTest/include/SmingTest.h:24,
                 from c:/tools/sming/tests/HostTests/app/test-string.cpp:1:
c:/tools/sming/tests/HostTests/app/test-string.cpp: In member function 'void StringTest::templateTest(T)':
c:/tools/sming/Sming/Components/FlashString/src/include/FlashString/String.hpp:47:4: sorry, unimplemented: unexpected AST of kind static_assert
  }))
    ^
c:/tools/sming/tests/HostTests/app/test-string.cpp:27:16: note: in expansion of macro 'FS_PTR'
   auto fstr1 = FS_PTR("This FSTR should get moved out of RAM, filtered by __fstr__ in symbol name.");
                ^
c:/tools/sming/tests/HostTests/app/test-string.cpp:27: confused by earlier errors, bailing out
make[1]: *** [app/test-string.o] Error 1
make: *** [App-build] Error 2

C:\tools\sming\tests\HostTests>
```

mikee47, 2020-02-11:
As this is a different problem, and this issue is closed, please open a new issue. I suspect your compiler is old, but the output from `make config` plus `set` should help to pin things down.

PR #2048 (2020-02-10): Fix axtls-8266 patch
Fails on some systems but not others. Reason unclear.
Fixes #2047
closed 2020-02-11 by mikee47