SmingHub/Sming #2058: Installation on Mac issues

IanAber, 2020-04-08:
Installing on a Mac with Catalina and trying to follow the Mac quick start guide. Downloaded the esp-open-sdk and sming. The instructions say to change to the smaing/sming directory and run make. Make comes back with...

> _Please run `make` from your project directory!
> make: Nothing to be done for `all'._
> 

Change to ../samples/basic_blink and run make...

> _Building /Users/ianaber/documents/projects/esp8266/Sming/Sming/out/Esp8266/debug/lib/clib-driver-a782326d7ee2f03e8ea67fc3ee886339.a
> C+ /Users/ianaber/documents/projects/esp8266/Sming/Sming/Arch/Esp8266/Components/driver/SerialBuffer.cpp
> C+ /Users/ianaber/documents/projects/esp8266/Sming/Sming/Arch/Esp8266/Components/driver/hw_timer.cpp
> C+ /Users/ianaber/documents/projects/esp8266/Sming/Sming/Arch/Esp8266/Components/driver/i2s.cpp
> In file included from /Users/ianaber/documents/projects/esp8266/Sming/Sming/Core/Rational.h:21:0,
>                  from /Users/ianaber/documents/projects/esp8266/Sming/Sming/Core/NanoTime.h:26,
>                  from /Users/ianaber/documents/projects/esp8266/Sming/Sming/Platform/Clocks.h:15,
>                  from /Users/ianaber/documents/projects/esp8266/Sming/Sming/Core/PolledTimer.h:28,
>                  from /Users/ianaber/documents/projects/esp8266/Sming/Sming/Platform/Timers.h:23,
>                  from /Users/ianaber/documents/projects/esp8266/Sming/Sming/Arch/Esp8266/Components/driver/i2s.cpp:51:
> /Users/ianaber/documents/projects/esp8266/Sming/Sming/Wiring/WString.h:64:34: fatal error: FlashString/String.hpp: No such file or directory
>  #include <FlashString/String.hpp>
>                                   ^
> compilation terminated.
> make[1]: *** [i2s.o] Error 1
> make: *** [driver-build] Error 2_
>

mikee47, 2020-04-08:
Hi, are you following the latest instructions at https://sming.readthedocs.io/en/latest/arch/esp8266/getting-started/macos.html ?

IanAber, 2020-04-08:
I am trying to. The first problem here Is that installing SED fails because the homebrew option —with-default-names has been removed from the current version of homebrew. Skipping this and allowing SED to install without the option gets past that but then that’s where I run into the problems listed. i don’t believe I got as far as running SED yet so I don’t think that is the cause.

slaff, 2020-04-09:
Reported Error(s)
============

> /Users/ianaber/documents/projects/esp8266/Sming/Sming/Wiring/WString.h:64:34: fatal error: FlashString/String.hpp: No such file or directory
#include <FlashString/String.hpp>

@IanAber  can you run the following commands and paste the output here:

```
env
git status
git submodule
make V=1 
```

Document Update
==============
We will have to update the MacOSX document. Thanks for pointing out the following issues:

Issue 1. `the homebrew option —with-default-names has been removed from the current version of homebrew.` 

That is correct. And this can be replaced with the following command to be executed or added to your bash profile:

```
export PATH="/usr/local/opt/gnu-sed/libexec/gnubin:$PATH"
```

Issue 2: `The instructions say to change to the smaing/sming directory and run make. Make comes back with...` 

That should be removed. We now recommend directly compiling the samples.

IanAber, 2020-04-09:
Output from env:-
TMPDIR=/var/folders/t3/cmly3ksd2ml3g2248lrh9qnm0000gn/T/
XPC_FLAGS=0x0
LaunchInstanceID=E3E71F1C-91DD-4F52-900F-8BF9E5845880
TERM=xterm-256color
LANG=en_US.UTF-8
SSH_AUTH_SOCK=/private/tmp/com.apple.launchd.FlHa97yJFt/Listeners
SECURITYSESSIONID=186ad
XPC_SERVICE_NAME=0
TERM_PROGRAM=Apple_Terminal
TERM_PROGRAM_VERSION=433
TERM_SESSION_ID=90097CF4-DB92-478C-8E3D-5F03533348F7
SHELL=/bin/zsh
HOME=/Users/ianaber
LOGNAME=ianaber
USER=ianaber
PATH=/usr/local/opt/gnu-sed/libexec/gnubin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Volumes/ESP8266/esp-open-sdk/xtensa-lx106-elf/bin:/Users/ianaber/Library/Python/3.7/bin:/Applications/VMware Fusion.app/Contents/Public:/usr/local/go/bin:/Applications/Wireshark.app/Contents/MacOS:/usr/local/sbin
SHLVL=1
PWD=/Users/ianaber/documents/projects/esp8266/sming/samples/basic_blink
OLDPWD=/Users/ianaber/documents/projects/esp8266/sming/samples
GOROOT=/usr/local/go
GOPATH=/Users/ianaber/Documents/Projects/Go_Home
ESP_HOME=/opt/esp-open-sdk
SMING_HOME=/Users/ianaber/Documents/Projects/ESP8266/Sming/Sming
_=/usr/bin/env

Output from git status:-
On branch develop
Your branch is up to date with 'origin/develop'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)

	../../Sming/Components/README.rst
	../../Sming/Components/component.mk
	../../Sming/Libraries/README.md
	../../Sming/Libraries/README.rst
	../../Sming/Libraries/component.mk

nothing added to commit but untracked files present (use "git add" to track)

Output from git submodule
 19cb69e9ce5071686d0e2a28962fd06e11d03a61 ../../Sming/Arch/Esp8266/Components/driver/new-pwm (19cb69e)
 b9a23a6e9ef88d995ae74014e71a6a4227f61d62 ../../Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip (heads/sdk-1.5.0)
 3fe474e040450e0ab72ea75f8f00c3d9c6279025 ../../Sming/Arch/Esp8266/Components/esp8266/ESP8266_NONOS_SDK (v3.0.3)
 f2560887541414de54dbfb4368cdfc3fdd62e019 ../../Sming/Arch/Esp8266/Components/esptool/esptool (v2.6-18-gf256088)
-5473e7749ed00ac6eab71dc5d77934231f4c08e6 ../../Sming/Arch/Esp8266/Components/heap/umm_malloc
-5b73bb11d59285d49a6efeb20726186cc5cc2b74 ../../Sming/Arch/Esp8266/Components/lwip2/lwip2
-ec11b289cb18490fae94fda199fbb54cc35226f8 ../../Sming/Arch/Host/Components/lwip/lwip
 804f22b4ea5ea7782a265ec30850a564da25ed71 ../../Sming/Components/FlashString (2.0.0-18-g804f22b)
-6c3e87b307ffb67e670909a3c5ac613032eed0ba ../../Sming/Components/axtls-8266/axtls-8266
-5c771bed8b11b968e2f75f26b9b00de51f2e9333 ../../Sming/Components/bearssl-esp8266/bearssl
 2343fd6b5214b2ded2cdcf76de2bf60903bb90cd ../../Sming/Components/http-parser (v2.9.0-11-g2343fd6)
 463760fb101861d760eae9b46dbf7f8dfaa30179 ../../Sming/Components/libyuarel (1.0-20-g463760f)
 b9fed830a33651dc9fd873d589ee26f85e74077b ../../Sming/Components/mqtt-codec (1.0.0-10-gb9fed83)
 a550572b9de8c25f26b7a7675b47e58861227f64 ../../Sming/Components/rboot/esptool2 (a550572)
 24f6021da82b303262a939c955f6a984bce6208f ../../Sming/Components/rboot/rboot (remotes/origin/rtc-52-g24f6021)
 ec68ba8208d7550860e4e78299d58a529b88bf85 ../../Sming/Components/spiffs/spiffs (0.2-234-gec68ba8)
 e94a4b9d8eab5dface51c666db546c60ca9f8377 ../../Sming/Components/ws_parser (heads/master)
-321186220e1e04080bb70fa45aae6e6a5820899f ../../Sming/Libraries/Adafruit_BME280_Library
-ddfec78fa15f0ff8dfc8a76524077ba6bb5fc6f3 ../../Sming/Libraries/Adafruit_SSD1306
-591fa4165d95c33ba13bfad2f09758852fe1282a ../../Sming/Libraries/Adafruit_ST7735
-e985f2253a687ef377cde3dcfb1f788830d1bc09 ../../Sming/Libraries/Adafruit_Sensor
-cc4ca55e938891a2903a9fedd5e10c811da20579 ../../Sming/Libraries/Adafruit_VL53L0X
-ad4b13c8f044e67f1610fba96e8dc108ebc90cd5 ../../Sming/Libraries/ArduinoJson5/ArduinoJson
-1b8107094f00d89b8ef781a4ac71c5c2528ff63e ../../Sming/Libraries/ArduinoJson6/ArduinoJson
-c295afa686ce3bc69050e977b7cd66066c737656 ../../Sming/Libraries/DFRobotDFPlayerMini
-c25875ae1ab726644ebbd49c06abe30c3e91681f ../../Sming/Libraries/DHTesp
-24c589f6e6a4dafd7dfe201ad8cd72d6efdd7ed0 ../../Sming/Libraries/IR
-5a5cf0abbf7f12e9bd98bb4540bbe8350bb5c9a3 ../../Sming/Libraries/ITEADLIB_Arduino_Nextion
-280d6bec2f424f39955e62f2bc45efc01577f783 ../../Sming/Libraries/MCP_CAN_lib
-c298f454faba6119758c12fee29a6a4e3dd124f9 ../../Sming/Libraries/ModbusMaster/ModbusMaster
-772639cf10db6d9f5a655ee9b7eb20b815fab396 ../../Sming/Libraries/MultipartParser/multipart-parser
-9e091e9632bc3eba513f422ee7cb17895c565a1a ../../Sming/Libraries/RingBufCPP
-8ea2c65533ef2037f32958f96b893e33374a2422 ../../Sming/Libraries/SmingTest
-1337247015fa3a15e7ce0aa10e2d1010b89cc3ab ../../Sming/Libraries/TM1637
-940ef42797baa0278df6b7fd9e67c7590f87744b ../../Sming/Libraries/libsodium/libsodium

Output from make V=1

The instruction created a massive amount of output and ultimately ended in a successful compilation of the Basic_Blink application. Rather than append the output I decided to try running a simple ‘make’ again and this time it came back saying nothing was done. I altered the application changing the time to 1500 from 1000 so it would force a recompilation of the application and ran make with no options. This time it recompiled the application without errors.

I then ran make clean followed by make which again ran without error.

It would seem that running make V=1 on a sample application was the key to getting the Sming library compiled so perhaps this should be included in the instructions.

Many thanks for the help. Now I will try and get Eclipse up and running so I have an IDE to use. I think it should be possible to use Visual Studio Code which I would much prefer but I have not seen anything about that for working with Sming.

Regards,
Ian Abercrombie

Ian.Abercrombie@CedarTechnology.com <mailto:Ian.Abercrombie@CedarTechnology.com>
Phone: +1 (352) 281-6286
www.CedarTechnology.com <http://www.cedartechnology.com/>

IanAber, 2020-04-14:
Problem is not solved after all.If I run make clean the build then fails again. Here is the V=1 output...
`Checking patch Makefile...
Checking patch rboot.c...
Checking patch appcode/rboot-api.c...
Applied patch Makefile cleanly.
Applied patch rboot.c cleanly.
Applied patch appcode/rboot-api.c cleanly.
Checking patch elf.h...
Checking patch esptool2.c...
Checking patch esptool2.h...
Checking patch Makefile...
Applied patch elf.h cleanly.
Applied patch esptool2.c cleanly.
Applied patch esptool2.h cleanly.
Applied patch Makefile cleanly.
Checking patch lib/libc.a => lib/libc.orig.a...
Checking patch lib/libgcc.a => lib/libgcc.orig.a...
Applied patch lib/libc.a => lib/libc.orig.a cleanly.
Applied patch lib/libgcc.a => lib/libgcc.orig.a cleanly.
Checking patch pwm.c...
Applied patch pwm.c cleanly.
Checking patch include/user_config.h...
Checking patch Makefile.open...
Checking patch include/arch/cc.h...
Checking patch include/lwipopts.h...
Checking patch include/lwip/ip_addr.h...
Checking patch lwip/app/dhcpserver.c...
Checking patch include/lwip/tcp_impl.h...
Checking patch include/lwip/igmp.h...
Checking patch include/lwip/mem.h...
Checking patch include/lwip/app/espconn.h...
Checking patch include/lwip/mdns.h...
Checking patch lwip/app/espconn_udp.c...
Checking patch lwip/core/dhcp.c...
Checking patch lwip/core/ipv4/ip_addr.c...
Checking patch lwip/core/tcp.c...
Checking patch lwip/app/espconn.c...
Checking patch lwip/app/espconn_tcp.c...
Applied patch include/user_config.h cleanly.
Applied patch Makefile.open cleanly.
Applied patch include/arch/cc.h cleanly.
Applied patch include/lwipopts.h cleanly.
Applied patch include/lwip/ip_addr.h cleanly.
Applied patch lwip/app/dhcpserver.c cleanly.
Applied patch include/lwip/tcp_impl.h cleanly.
Applied patch include/lwip/igmp.h cleanly.
Applied patch include/lwip/mem.h cleanly.
Applied patch include/lwip/app/espconn.h cleanly.
Applied patch include/lwip/mdns.h cleanly.
Applied patch lwip/app/espconn_udp.c cleanly.
Applied patch lwip/core/dhcp.c cleanly.
Applied patch lwip/core/ipv4/ip_addr.c cleanly.
Applied patch lwip/core/tcp.c cleanly.
Applied patch lwip/app/espconn.c cleanly.
Applied patch lwip/app/espconn_tcp.c cleanly.
Checking patch src/spiffs.h...
Checking patch src/spiffs_hydrogen.c...
Checking patch src/spiffs_nucleus.c...
Applied patch src/spiffs.h cleanly.
Applied patch src/spiffs_hydrogen.c cleanly.
Applied patch src/spiffs_nucleus.c cleanly.
Checking patch http_parser.c...
Applied patch http_parser.c cleanly.
Checking patch ws_parser.c...
Applied patch ws_parser.c cleanly.
Checking patch src/platform.h...
Applied patch src/platform.h cleanly.
In file included from /Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Core/Rational.h:21:0,
                 from /Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Core/NanoTime.h:26,
                 from /Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Platform/Clocks.h:15,
                 from /Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Core/PolledTimer.h:28,
                 from /Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Platform/Timers.h:23,
                 from /Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Arch/Esp8266/Components/driver/i2s.cpp:51:
/Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Wiring/WString.h:64:34: fatal error: FlashString/String.hpp: No such file or directory
 #include <FlashString/String.hpp>
                                  ^
compilation terminated.
make[1]: *** [i2s.o] Error 1
make: *** [driver-build] Error 2
`
[output.txt](https://github.com/SmingHub/Sming/files/4472928/output.txt)

slaff, 2020-04-14:
> It would seem that running make V=1

`V=1` is making the output more verbose. It does not change the actual tasks that are executed. It helps us see what is your include path for libraries and header files.

> Rather than append the output I decided to try running a simple ‘make’ again and this time it came back saying nothing was done.

Once the application is compiled there is no need for recompilation. 

> Here is the V=1 output...

That does not look like a verbose output.

IanAber, 2020-04-14:
Here is the complete output from make V=1.

First I ran:
make spiffs-image-clean
make clean
make dist-clean
make component-clean
make submodules-clean

This ensures that everything gets recompiled.

Next I ran...
make V=1

`basic_blink % make V=1

Basic_Blink: Invoking 'all' for Esp8266 (debug) architecture
-- Parsing App
-- Parsing Sming
-- Parsing sming-arch
-- Parsing libc
-- Parsing heap
-- Parsing rboot
-- Parsing esp8266
-- Parsing driver
-- Parsing esptool
-- Parsing fatfs
-- Parsing gdbstub
-- Parsing spi_flash
-- Parsing esp_wifi
-- Parsing esp-open-lwip
-- Parsing FlashString
-- Parsing spiffs
-- Parsing http-parser
-- Parsing libb64
-- Parsing ws_parser
-- Parsing mqtt-codec
-- Parsing libyuarel
-- Parsing ssl
-- Parsing crypto
-- Parsing terminal
mkdir -p out/Esp8266/debug/firmware
mkdir -p /Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/out/Esp8266/debug/tools
mkdir -p out/Esp8266/debug/lib
mkdir -p /Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/out/Esp8266/debug/lib

Fetching submodule '/Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Components/rboot/rboot' ...
cd /Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Components/rboot && (rm -rf rboot; git submodule update --init --force --recursive rboot)
Submodule path 'rboot': checked out '24f6021da82b303262a939c955f6a984bce6208f'
cd /Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Components/rboot/rboot && if [ -f ../rboot.patch ]; then 	git apply -v ../rboot.patch --ignore-whitespace --whitespace=nowarn; elif [ -f ../.patches/rboot.patch ]; then 	git apply -v ../.patches/rboot.patch --ignore-whitespace --whitespace=nowarn; fi && if [ -d ../.patches/rboot/ ]; then cp -rf ../.patches/rboot/* . ; fi
Checking patch Makefile...
Checking patch rboot.c...
Checking patch appcode/rboot-api.c...
Applied patch Makefile cleanly.
Applied patch rboot.c cleanly.
Applied patch appcode/rboot-api.c cleanly.
touch /Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Components/rboot/rboot/.submodule

Fetching submodule '/Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Components/rboot/esptool2' ...
cd /Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Components/rboot && (rm -rf esptool2; git submodule update --init --force --recursive esptool2)
Submodule path 'esptool2': checked out 'a550572b9de8c25f26b7a7675b47e58861227f64'
cd /Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Components/rboot/esptool2 && if [ -f ../esptool2.patch ]; then 	git apply -v ../esptool2.patch --ignore-whitespace --whitespace=nowarn; elif [ -f ../.patches/esptool2.patch ]; then 	git apply -v ../.patches/esptool2.patch --ignore-whitespace --whitespace=nowarn; fi && if [ -d ../.patches/esptool2/ ]; then cp -rf ../.patches/esptool2/* . ; fi
Checking patch elf.h...
Checking patch esptool2.c...
Checking patch esptool2.h...
Checking patch Makefile...
Applied patch elf.h cleanly.
Applied patch esptool2.c cleanly.
Applied patch esptool2.h cleanly.
Applied patch Makefile cleanly.
[... 162 lines cut ...]
gcc -o /Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/out/Esp8266/debug/tools/esptool2 /Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/out/Esp8266/debug/build/rboot/esptool2/esptool2.o /Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/out/Esp8266/debug/build/rboot/esptool2/esptool2_elf.o
mkdir -p /Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/out/Esp8266/debug/build/driver/driver-8b6facc9103f6bd2196eb58c28dd7b16

Building /Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/out/Esp8266/debug/lib/clib-driver-8b6facc9103f6bd2196eb58c28dd7b16.a
/Library/Developer/CommandLineTools/usr/bin/make -r -R --no-print-directory -C /Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/out/Esp8266/debug/build/driver/driver-8b6facc9103f6bd2196eb58c28dd7b16 -f /Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/component-wrapper.mk COMPONENT_NAME=driver COMPONENT_PATH=/Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Arch/Esp8266/Components/driver COMPONENT_LIBDIR=/Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/out/Esp8266/debug/lib COMPONENT_LIBNAME=driver COMPONENT_LIBHASH=8b6facc9103f6bd2196eb58c28dd7b16  build
WriteConfig /Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/out/Esp8266/debug/build/driver/driver-8b6facc9103f6bd2196eb58c28dd7b16/build.cfg
mkdir -p new-pwm
/opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -I/Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Components -I/Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/System/include -I/Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Wiring -I/Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Core -I/Users/ianaber/Documents/Projects/ESP8266/Sming/Sming -I/Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Arch/Esp8266 -I/Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Arch/Esp8266/Core -I/Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Arch/Esp8266/System/include -I/Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Arch/Esp8266/Components -I/Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Arch/Esp8266/Components/libc/include -I/Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Components/rboot/rboot -I/Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Components/rboot/appcode -I/Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Components/rboot/rboot/appcode -I/Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Arch/Esp8266/Components/esp8266/include -I/Users/ianaber/Documents/Projects/ESP8266/Sming/ [... cut ...]
                 from /Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Platform/Timers.h:23,
                 from /Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Arch/Esp8266/Components/driver/i2s.cpp:51:
/Users/ianaber/Documents/Projects/ESP8266/Sming/Sming/Wiring/WString.h:64:34: fatal error: FlashString/String.hpp: No such file or directory
 #include <FlashString/String.hpp>
                                  ^
compilation terminated.
make[1]: *** [i2s.o] Error 1
make: *** [driver-build] Error 2`

IanAber, 2020-04-14:
After running this make and getting the error 'FlashString/String.hpp: No such file or directory' I immediately executed the same command again (clicked the up arrow to get the last command executed and hit Enter so no mistakes in typing). This second time it ran to completion with no apparent errors.

Next I tried running make dist-clean. After this the make command again failed with the same FlashString/String.hpp error. Running make again after getting the error ran to completion with no apparent errors.

Again I tried make dist-clean then make and got the same FlashString/String.hpp error result.

It appears that the error is reliably reproduced by running make immediately after a make dist-clean.

I am doing this in a the Basic_Blink sample project of an unmodified fresh installation of Sming following the installation instructions in the Getting Started page for Mac OS. I have not attempted to install Eclipse but am running purely in a terminal window from the commend line. The iMac is running the latest version of Catalina.

Hope this helps.

closed 2024-04-04 by mikee47