SmingHub/Sming #2064: Basic_rBoot sample OTA fails

VooDooSi, 2020-05-02:
When i start OTA update from rom0 (first boot after flashing) everything seems ok, it downloads file and flashes rom1. When you run OTA from rom1 update starts but resets and rom0 gets corrupted, on restart it tries to boot rom1 but that does not happen either. 
I have attached logs, rom0.log is right after flashing from command line and starting "ota" from terminal, rom1.log is when esp restartes from flashing and ota is started again. 

This has been tried on quite a few modules i have at home and the same happens on all of them. So i would say its not related to undervoltage or something like that. OTA worked on them with Sming 3.8. 

Sming version 4.1.0
SDK 3.1

[rom0.log](https://github.com/SmingHub/Sming/files/4567970/rom0.log)
[rom1.log](https://github.com/SmingHub/Sming/files/4567971/rom1.log)

dpechman, 2020-05-20:
same problem happening here ... is it already fixed in 4.1.1? I'm currently using the 3.8.1 bootloader to work around the problem

VooDooSi, 2020-05-20:
@dpechman, same thing happens with 4.1.1.

slaff, 2020-05-22:
@VooDooSi How do you compile and how do you test? Let's try to make it easier for us to reproduce the problem. First - paste here your config. Go to your application and run:
```
make list-config
```

Second: what is the size of your ESP8266 flash memory?  If it is below 1MB then things can work but not by default.

Third: what is the rboot configuration that you have? Do you use the default one? Can you paste here the output from the flashing command? 

```
make flash V=1 
```

Fourth: Do you have sample code that is demonstrating the problem? Are you using the Basic_rBoot sample (as the log suggest) or you have your own modifications? 

Fifth++: The SDK is SDK: v3.1.0-dev(bd176e4) but what is the compiler that you use and the actual host OS where you compile the stuff?

All that can help us understand better and be able to reproduce the problem.

slaff, 2020-05-22:
One more thing here: There used to be a problem with ROM2 but it should have been fixes with the merging of this PR #1900

VooDooSi, 2020-06-03:
Hi, 
sorry been a bit busy lately.
Here is list-config:
[config.txt](https://github.com/SmingHub/Sming/files/4724220/config.txt)

I am using esp12e, so it is 4MB. I Tried 3 or 4 different versions and same problem on all of them.
The only thing that was changed in Basic_rBoot sample was that it connects to wifi on start. No other configuration other than com port and flash related was changed. 

make flash V=1:
[flash.txt](https://github.com/SmingHub/Sming/files/4724263/flash.txt)

Host OS is Windows 10 , gcc (MinGW.org GCC-8.2.0-5) 8.2.0.

dpechman, 2020-06-03:
it seems to work using: make dist-clean

VooDooSi, 2020-06-03:
Tried "make dist-clean" and something was different. I got stack dump : 
[stacktrace.txt](https://github.com/SmingHub/Sming/files/4724467/stacktrace.txt)

But after reset of the module no joy.

VooDooSi, 2020-06-04:
I created Linux virtual machine (Debian 10, gcc 6.3.0) and ota works as it should. Same example (copied) works when compiled on linux..

kmihaylov, 2020-08-09:
So I'm playing now with the OTA updater. The sample works pretty fine.

My app sucessfully flashes the rom for the first time (e.g. rom 1 slot), but if I try OTA again (it should flash rom 0 slot), then it fails.

```
void updateCallBack(RbootHttpUpdater& client, bool result)
{
	Serial.printf("In callback...");
	if(result == true) {
		uint8 slot;
		slot = rboot_get_current_rom();
		if(slot == 0)
			slot = 1;
		else
			slot = 0;
		Serial.printf("Firmware updated, rebooting to rom %d...\r\n", slot);
		rboot_set_current_rom(slot);
		System.restart();
	} else {
		Serial.printf("Firmware update failed!");
	}
}

void update(  )
{
	uint8 slot;
	rboot_config bootconf;

	Serial.printf("Updating...");

	if(otaUpdater)
		delete otaUpdater;
	otaUpdater = new RbootHttpUpdater();

	// select rom slot to flash
	bootconf = rboot_get_config();
	slot = bootconf.current_rom;
	if(slot == 0)
		slot = 1;
	else
		slot = 0;

	otaUpdater->addItem(bootconf.roms[slot], "http://some_url/some_name.bin");

	otaUpdater->setCallback(updateCallBack);
	Serial.printf("rBoot_config, romSlot: %d , addr1: %#010x , addr2: %#010x\r\n", bootconf.current_rom,
				  bootconf.roms[0], bootconf.roms[1]);
	otaUpdater->start();
}
```

Here is the output of the second updae:
```
updateFWUpdating...rBoot_config, romSlot: 1 , addr1: 0000002000 , addr2: 0000102000
Sming>

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 1072, room 16 
tail 0
chksum 0x50
load 0x3ffe8000, len 576, room 8 
tail 8
chksum 0x41
csum 0x41

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz

Rom 1 at 102000 is bad.
Booting rom 0 at 2000, load addr 5a4b0.
--- RTS active ---
--- RTS inactive ---
```
And won't boot after that...

Any ideas how to debug it?

First I thought that maybe my app has bigger bin file and maybe something happens with the server (I had problems with the server in the past), but then it would fail on the first update.

I think something happens with the flash since I have few timers that execute different tasks on the MCU and maybe they brake the flash writing process somehow.

Is there some command to disable all processes or I have to do it manually?

closed 2022-02-09 by slaff