SmingHub/Sming #2068: Problems with MqttClient_Hello + SSL sample

Vinimuller, 2020-05-11:
Hello! I'm running Sming 4.1.0 on MacOS with ESP8266. I am only trying to use the MqttClient_Hello sample with ENABLE_SSL but I am getting this output:

```
mode : sta(68:c6:3a:9f:ad:ef) + softAP(6a:c6:3a:9f:ad:ef)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
146941 mode: 0 -> 3

scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with SalameAP, channel 1
dhcp client start...
262326 connect to ssid SalameAP, channel 1

ip:192.168.0.4,mask:255.255.255.0,gw:192.168.0.1
1183063 ip:192.168.0.4,mask:255.255.255.0,gw:192.168.0.1

Connecting to mqtts://attachix.com:8883/
please start sntp first !
please start sntp first !
2865009 SSL Validator: list empty, allow connection
MQTT Broker Disconnected!!
```

It keeps asking me to start the sntp server. How do I fix it in this sample?

slaff, 2020-05-11:
> It keeps asking me to start the sntp server

Basically the sntp server syncs the current time of your ESP device. If that is not done the ESP8266 device starts from 0. A current time is required when you want to validate also the expiration dates in a certificate. In the `MqttClient_Hello` that is not needed because we don't make a full validation of the remote certificate. 

> How do I fix it in this sample?

Well it works for me... So either try it again or change the MQTT server address to point to one of the test servers here: https://test.mosquitto.org/.  For example `mqtts://attachix.com:8883/` can be replaced with `mqtts://test.mosquitto.org:8883/`

Vinimuller, 2020-05-11:
> Well it works for me... So either try it again or change the MQTT server address to point to one of the test servers here: https://test.mosquitto.org/. For example mqtts://attachix.com:8883/ can be replaced with mqtts://test.mosquitto.org:8883/

Changing the server worked for me! Thanks.

I would also like to make a full validation of the certificate for another application. How would I start the sntp?

slaff, 2020-05-14:
> How would I start the sntp?

You can take a look at the time.c file in axtls-8266 replacements. Basically if sntp is not started yet you should:
```
// set a server from where to get the exact time
// Better use a server that is close to you: https://support.ntp.org/bin/view/Servers/NTPPoolServers
sntp_setservername(1, "0.pool.ntp.org");

// then set the timezone. The timezone is integer. See sntp.c for details
sntp_set_timezone(timezone/3600);

// and finally send the sntp request to the remote server.
sntp_init();
```

closed 2020-05-24 by slaff