SmingHub/Sming #2070: IR lib

teicors, 2020-05-16:
Hi,
I'm looking for integrate an inexpensive IR remote into my project.
The first test with Arduino works and if I hold continuously the button, I receive 0xFFFFFFFF.
Compiling the sample/IR_lib, I see the code of the button, but if I keep the button pressed, I receive nothing.
Thanks!

kmihaylov, 2020-06-01:
I observed some delays when I played with the IR lib. It was long ago, so you're welcome to post more information.

teicors, 2020-06-02:
@kmihaylov please let me understand how I can be more detailed.

I simply put an ir receiver on the D2 (Lolin module) and then
```
(base) angelo@angelo-thinkpad-x1-carbon-4th:/opt/sming/samples/IR_lib$ make flash

IR_lib: Invoking 'flash' for Esp8266 (debug) architecture

Building /opt/sming/samples/IR_lib/out/Esp8266/debug/lib/clib-App-42490a945869de8bdcf187d7754669f4.a
make[1]: Nessuna operazione da eseguire per "build".
IR_lib: Linking out/Esp8266/debug/build/app_0.out
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE853D|    1341
    rodata|           ReadOnly Data (RAM)|    3FFE8540|    3FFE8BFC|    1724
       bss|      Uninitialized Data (RAM)|    3FFE8C00|    3FFEF240|   26176
      text|            Cached Code (IRAM)|    40100000|    401066FB|   26363
irom0_text|           Uncached Code (SPI)|    40202010|    40238217|  221703
Total Used RAM : 29241
Free RAM : 52679
Free IRam : 6423
ESPTOOL2 out/Esp8266/debug/firmware/rom0.bin
(!) Spiffs support disabled. Remove 'DISABLE_SPIFFS' make argument to enable spiffs.
SSL support disabled
Killing Terminal to free /dev/ttyUSB3
make: [/opt/sming/Sming/Components/terminal/component.mk:28: kill_term] Ucciso (ignorato)
WriteFlash 0x00000=out/Esp8266/debug/firmware/rboot.bin 0x002000=out/Esp8266/debug/firmware/rom0.bin 
python /opt/sming/Sming/Arch/Esp8266/Components/esptool/esptool/esptool.py -p /dev/ttyUSB3 -b 1000000 write_flash -ff 40m -fm dio -fs 4MB 0x00000 out/Esp8266/debug/firmware/rboot.bin 0x002000 out/Esp8266/debug/firmware/rom0.bin 
esptool.py v2.7-dev
Serial port /dev/ttyUSB3
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 18:fe:34:d7:5e:e8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 1000000
Changed.
Configuring flash size...
Compressed 1728 bytes to 1282...
Wrote 1728 bytes (1282 compressed) at 0x00000000 in 0.0 seconds (effective 725.6 kbit/s)...
Hash of data verified.
Compressed 251216 bytes to 185480...
Wrote 251216 bytes (185480 compressed) at 0x00002000 in 3.2 seconds (effective 629.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
python -m serial.tools.miniterm --raw --encoding ascii /dev/ttyUSB3 115200
--- Miniterm on /dev/ttyUSB3  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
�gn��ۃ�`~�c
           �og��Ó`�cl`s�'dph����g�s��'|�
                                        d
                                         ldl c�|;�l�o��o�l ��{�d�l
                                                                  �j
Setting up...
Ready...
mode : softAP(1a:fe:34:d7:5e:e8)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
Encoding  : NEC
Code      : FD609F (32 bits)

Encoding  : NEC
Code      : FDB04F (32 bits)

Encoding  : NEC
Code      : FDB04F (32 bits)

```
The last key is "vol up" in my remote, but keeping the key pressed, I receive nothing.
I did some test using different gpio, but always the same.
As I told, under Arduino I can receive the 0xFFFFFFFF that means key still pressed and I have to put a short delay in order to read a single key.

I agree with you about the delay: in my opinion the lib goes worng when have to read quickly from the pin.

Forcing the cpu at 160mhz, I can't see any difference.

kmihaylov, 2020-06-02:
@teicors now I remember that in the very first moments there was (almost) no delay. In a while the ESP started to respond slower and slower. Don't remember, however, whether it was in correlation with the commands sent by the remote or not. @mikee47 knows well the internals of the framework, especially the interrupts and event handling.

Also I see that the IR lib version in Sming is almost 2 years old, while the last commit in the official repo is about 2 weeks old. If it doesn't require patching, then we could try updating it...

teicors, 2020-06-03:
it's empiric test, but I can't notice any delay after two minutes of "keys pressed". I can notice a small delay between the keypressed and the decode.

kmihaylov, 2020-06-03:
It is interesting that for me it introduces some big delays. Just tied with my KEYES remote and immediately after the first received signal it starts to delay, better said, its almost stops.

This was with the current Sming release, therefore I decided to update the IR lib, and I think this is the way to go. Now there is new problem (constant repeat of the received command, as seen by the ESP), but it might be caused by the IR version change.

kmihaylov, 2020-06-03:
I'm attaching my quick and dirty patch (remove the txt extension).
[ir.patch.txt](https://github.com/SmingHub/Sming/files/4722063/ir.patch.txt)

The lib can be updated by issuing these commands (the sample should be ```make``` once for the IR lib to be downloaded into Libraries/IR):
```
cd $SMING_HOME/Libraries/IR
git checkout -- .
git checkout master
git pull
git apply /path/to/ir.patch
(lib/googletest warning is OK here)
cd $SMING_HOME/../samples/IR_lib
make IR-rebuild
make flash
```

The if(irrecv.decode(&dresults)) should be modified:
```
	if(irrecv.decode(&dresults)) {
		Serial.println(resultToHumanReadableBasic(&dresults)); // Output the results as source code
		Serial.println(resultToTimingInfo(&dresults));
		irrecv.resume();
	}
```
(```irrecv.resume();``` is added).

For me the updated IR lib works and recognizes properly long-pressed button (0xfff...).
I hardly can say that the current Sming's IR lib works for me.

Later I'll prepare PR, but meanwhile you may try to update it manually.

teicors, 2020-06-03:
@kmihaylov GREAT !!
Now it works as expected !!

```

Protocol  : NEC
Code      : 0xFDB04F (32 Bits)

Protocol  : NEC (Repeat)
Code      : 0xFFFFFFFFFFFFFFFF (0 Bits)

Protocol  : NEC (Repeat)
Code      : 0xFFFFFFFFFFFFFFFF (0 Bits)

```

slaff, 2020-06-03:
> Later I'll prepare PR, but meanwhile you may try to update it manually.

@kmihaylov Thanks a lot in advance.

closed 2020-06-04 by teicors

slaff, 2020-06-07:
@teicors the latest develop has PR #2083 . Can you check if it is working for you as expected?