SmingHub/Sming #2079: Sming on FreeBSD

kmihaylov, 2020-06-02:
Hello. I'm a bit behind the latest build environment requirements. SMING_HOME is clear. ES_HOME in the past was esp-open-sdk build. But it changed few months ago and now uses the toolchains in SmingTools. However there are Linux and Windows binaries, and the source code archive is related to MinGW.

I remembered that Sming changed from native esp-open-sdk to ESP Quick Toolchain. Do you prepare Sming's toolchain binaries (in SmingTools) following the procedure as noted in https://github.com/earlephilhower/esp-quick-toolchain ?

Thanks!

slaff, 2020-06-03:
> Do you prepare Sming's toolchain binaries ?

We just use the already pre-compiled binaries from the `earlephilhower/esp-quick-toolchain` project.  @mikee47 can you confirm this?

mikee47, 2020-06-03:
That's right. See https://sming.readthedocs.io/en/latest/arch/esp8266/getting-started/eqt.html for details.

kmihaylov, 2020-06-23:
https://github.com/earlephilhower/esp-quick-toolchain/issues/13

closed 2020-06-23 by kmihaylov