SmingHub/Sming #2084: httpClient fail to download the large content

arikfe, 2020-06-11:
When using httpClient.downloadString on request with return body more then 1k the return result is incomplete.
Using downloadString on address such as  http://samples.openweathermap.org/data/2.5/forecast?id=524901&appid=439d4b804bc8187953eb36d2a8c26a02
Will result a 200 success code but the stream available bytes is only ~500
-  release mode did not help.
-  using getBody 
-  tried to create an application that load connect and then try to download with no other action  did not help.

- Trying requests with small return body works.

Release Version 4.1.1

slaff, 2020-06-12:
The microcontrollers are devices with limited resources. ESP8266 has between 20K and max 50K free RAM. Therefore we have to be careful how that RAM is used. 

By default Sming keeps up to 1K in the memory, as you have correctly noticed. Otherwise a device with Sming can be easily restarted when it runs out of memory or just made incapable of doing anything by sending large response body and causing Denial-Of-Service. 

But Sming allows you to download as much data as wanted. It is up to the developer - you  how to handle it.  Take a look at our HttpClient sample and at line 119 to instruct the httpclient to store more data in RAM.

https://github.com/SmingHub/Sming/blob/b17e6c9dcebcf828fb9e6c36923b6422ca7a2adf/samples/HttpClient/app/application.cpp#L119

Meanwhile we should update the documentation of the downloadString method and put a note explaining that it stores up to 1K in memory.

arikfe, 2020-06-12:
Thanks for the quick reply.
I'll check this solution ASAP.
I have managed to download that data using sming version 3.6

closed 2020-06-13 by arikfe