SmingHub/Sming #2089: SDCard sample not working with DISABLE_WIFI=1

mopac, 2020-06-26:
SDCard sample works fine with DISABLE_WIFI=0.
This results in about 200k of code.
But set DISABLE_WIFI=1, compiles with no errors and now the code size in only 60K but the code appears to crash before it has printed out any SDCard debug info.
I can't find any code in sdcard.cpp or the sample app.cpp that references anything wifi.
I assume it must be something in the esp_no_wifi component that is missing or broken.
Any pointers?
Ray

slaff, 2020-06-26:
@mikee47 You are the esp_no_wifi expert. When you have time can you help us with this issue?

mikee47, 2020-06-26:
@mopac Problem tracked down to one line in `SPIClass.begin()`, fix is below.

diff --git a/Sming/Arch/Esp8266/Core/SPI.cpp b/Sming/Arch/Esp8266/Core/SPI.cpp
index 02fb8bcba..a1d8a421f 100644
--- a/Sming/Arch/Esp8266/Core/SPI.cpp
+++ b/Sming/Arch/Esp8266/Core/SPI.cpp
@@ -32,7 +32,7 @@ SPIClass SPI;
  */
 void SPIClass::begin()
 {
-	WRITE_PERI_REG(PERIPHS_IO_MUX, 0x105); //clear bit9
+	CLEAR_PERI_REG_MASK(PERIPHS_IO_MUX, BIT9);
 
 	PIN_FUNC_SELECT(PERIPHS_IO_MUX_MTDI_U, 2); // HSPIQ MISO == GPIO12
 	PIN_FUNC_SELECT(PERIPHS_IO_MUX_MTCK_U, 2); // HSPID MOSI == GPIO13

closed 2020-06-26 by slaff

slaff, 2020-06-26:
@mopac The fix should be merged in the latest develop version. Try the following commands, if you are using Linux:

```
# make sure that you are using the develop branch
git checkout develop 

# get the latest changes
git pull 

cd $SMING_HOME/../samples/SDCard/
make DISABLE_WIFI=1 sming-arch-build all 

# now flash the new application and test.
make flashapp
```

mopac, 2020-06-26:
Fix tested and working.
Fabulous response.
Thanks for your support.
Ray

PR #2090 (2020-06-26): Fix bug in SPI class
Fixes #2089.
Don't overwrite IO_MUX reg, just clear required bit.