SmingHub/Sming #2095: GDB corrupts hardware SPI used for SDCard

mopac, 2020-07-25:
SDCard Sample fails to access SD Card when built with GDB.

Switching to softwareSPI makes it work agains.

Tracing the problem in hardware SPI.......................
The application should send 6 byte messages to the SD Card.
1 byte command. 4 bytes attributes. 1 byte CRC.

With GDB enabled, it still sends 6 bytes but the 5th byte is a copy of the 6th byte.
The init command 0 should be 0x40_00_00_00_00_95
What it actually sends is           0x40_00_00_00_95_95

The SD card sends back an error saying CRC error in the command.

So GDB is in some way interfering with the queue structure of hardware SPI

slaff, 2020-07-26:
@mikee47 Can you take a look at this issue?

mikee47, 2020-07-27:
I can confirm this behaviour, feels familiar... It's likely a race condition somewhere in the code, which I haven't pinned down yet. It isn't related to the GDB build per-se; try building with -Og instead of -Os and the same thing happens.

mikee47, 2020-07-27:
This patch fixes the problem:

```
diff --git a/Sming/Arch/Esp8266/Core/SPI.cpp b/Sming/Arch/Esp8266/Core/SPI.cpp
index a1d8a421f..d320980ce 100644
--- a/Sming/Arch/Esp8266/Core/SPI.cpp
+++ b/Sming/Arch/Esp8266/Core/SPI.cpp
@@ -183,7 +183,14 @@ void SPIClass::transfer(uint8_t* buffer, size_t numberBytes)
 											  (((num_bits - 1) & SPI_USR_MISO_BITLEN) << SPI_USR_MISO_BITLEN_S));
 
 		// copy the registers starting from last index position
-		memcpy((void*)SPI_W0(SPI_NO), &buffer[bufIndx], bufLength);
+//		memcpy((void*)SPI_W0(SPI_NO), &buffer[bufIndx], bufLength);
+		auto spiBuffer = reinterpret_cast<volatile uint32_t*>(SPI_W0(SPI_NO));
+		uint32_t words[BLOCKSIZE /4];
+		memcpy(words, buffer, bufLength);
+		unsigned wordCount = (bufLength + 3) / 4;
+		for(unsigned i = 0; i < wordCount; ++i) {
+			spiBuffer[i] = words[i];
+		}
 
 		// Begin SPI Transaction
 		SET_PERI_REG_MASK(SPI_CMD(SPI_NO), SPI_USR);
```

mikee47, 2020-07-27:
Edit: Line should be `memcpy(words, &buffer[bufIndx], bufLength);` but you get the idea.

mikee47, 2020-07-28:
Simplest fix:

`memcpy((void*)SPI_W0(SPI_NO), &buffer[bufIndx], ALIGNUP4(bufLength));`

mopac, 2020-07-28:
Hi Mike,

The first fix is all working fine.
Thanks for your support

Ray

PR #2096 (2020-07-28): Ensure SPI buffer writes are strictly 32-bit aligned
Fixes #2095.

Compiling with debugger, or optimisation at, say, -Og, will cause issues with buffered SPI data. The SPI FIFO should be written in 32-bit words to avoid this.
closed 2020-07-28 by slaff