SmingHub/Sming #2097: [semifix] esp32 Basic_Blink

kmihaylov, 2020-08-06:
Hello,

Few weeks ago I had some unluck with esp32:
```
undefined reference to `g_wifi_feature_caps'
```
So today we spent some time and as expected the esp-idf v.4 that Sming is patched for differs a lot from the current esp-idf v.4 (release/v4.0 that is stable and should be used).

Also even if you check out earlier esp-edf (v4) there is no warranty that the submodules would be the correct ones.

So @slaff gave me his esp-idf working dir and the samples compile and run successfully (yay!)!

I'm playing with the revisions in order to specify what combination works and will post my findings in a few minutes.

kmihaylov, 2020-08-06:
Okay I think these are the steps I did:

```
rm -rf ~/.espressif
rm -rf /opt/esp-idf
cd /opt
git clone -b release/v4.0 https://github.com/espressif/esp-idf.git
cd /opt/esp-idf
git checkout fe67bedee29e40d4a46e19f4d275b2d821f1eb62
./install.sh
. ./export.sh
. ./export.sh
cd $SMING_HOME/../samples/Basic_Blink
make dist-clean
make SMING_ARCH=Esp32 sdk-clean
make SMING_ARCH=Esp32 sdk-menuconfig
```
I noticed that if I do only once ```menucondig``` later I got a lot of linker errors, undefined references to spi_flash_read, spi_flash_write, esp_partition_find_first, esp_partition_erase_range, esp_log_write and etc...

Therefore I make it once, then I do it once again. Absolutely no idea how this influences the build process.
```
make -j4 SMING_ARCH=Esp32 sdk-build
```

Don't worry if the exit shows this error:
```
make[1]: Circular /opt/Sming/Sming/out/Esp32/debug/build/sdk/Sming <- /opt/Sming/Sming/out/Esp32/debug/build/sdk/Sming dependency dropped.
Done!
LD /opt/Sming/Sming/out/Esp32/debug/build/sdk/Sming
xtensa-esp32-elf-gcc: error: Application.map: No such file or directory
/home/opt/esp-idf/make/project.mk:528: recipe for target '/opt/Sming/Sming/out/Esp32/debug/build/sdk/Sming' failed
make[1]: *** [/opt/Sming/Sming/out/Esp32/debug/build/sdk/Sming] Error 1
make[1]: Leaving directory '/home/opt/Sming/Sming/Arch/Esp32/Components/esp_idf/project'
/opt/Sming/Sming/Arch/Esp32/Components/esp_idf/component.mk:249: recipe for target 'sdk-build' failed
make: [sdk-build] Error 2 (ignored)
```

Now try
```
make SMING_ARCH=Esp32 flash
```

kmihaylov, 2020-08-06:
Excuse me for writing 3rd comment in a row, but while playing with the current esp-idf I one more time got the too many undefined references.

I fixed it again using make SMING_ARCH=Esp32 sdk-menuconfig (once or twice, with using the Save button), and after make sdk-build (w/o sdk-clean) and dist-clean they disappear.

[undefined-references.txt](https://github.com/SmingHub/Sming/files/5037349/undefined-references.txt)

kmihaylov, 2020-08-06:
I'm still playing with the current esp-idf. I suspect that something with the Makefiles may mix the Sming provided esp-idf and the custom compiled one.

```g_wifi_feature_caps``` is related to ```$IDF_PATH/components/esp_wifi/``` and is introduced in https://github.com/espressif/esp-idf/commit/d36663b79884d864e80ddd57d90c6e0c18468362

In ```Basic_Blink``` I did:
```
make dist-clean
make clean
make SMING_ARCH=Esp32 sdk-menuconfig
make SMING_ARCH=Esp32 sdk-build
```
In ```$SMING_HOME/out/Esp32/debug/build/sdk/esp_wifi``` I do ```xtensa-esp32-elf-nm libesp_wifi.a | grep feature``` and the symbol is there:
```
00000000 B g_wifi_feature_caps
```

So I suspect that some mixing happens between the $IDF_PATH includes and the Sming provided (older) esp-idf libs.

------

Now I'm struggling with undefined references to these (maybe other) symbols:
```
g_flash_guard_default_ops
spi_flash_init
spi_flash_guard_set
esp_flash_app_init
esp_flash_init_default_chip
esp_log_write
esp_log_writev
```

I did the above trick with ```nm``` and I'm sure they exist in the Sming's custom esp-idf build, so I'm curious if they're present, then where the compiler looks for them?
```
/opt/Sming/Sming/out/Esp32/debug/build/sdk/log$ xtensa-esp32-elf-nm liblog.a | grep esp_log_write
00000000 T esp_log_write
00000000 T esp_log_writev
```

The result is the same for both files:
```
Sming/Arch/Esp32/Components/esp_idf/sdk/lib/liblog.a
Sming/out/Esp32/debug/build/sdk/log/liblog.a
```

However the result for ```Sming/out/Esp32/debug/build/sdk/lib/liblog.a``` is U:
```
 xtensa-esp32-elf-nm $SMING_HOME/out/Esp32/debug/build/sdk/lib/liblog.a | grep esp_log_write
         U esp_log_write
```

I expected that this script copies these files inside ```lib```. Am I wrong?
```
.PHONY: sdk-build
sdk-build: $(SDK_PROJECT_PATH)/sdkconfig $(SDK_BUILD_BASE) ##Build ESP-IDF SDK
	$(Q) rm -f $(FLASH_BOOT_LOADER)
	-$(MAKE) -C $(SDK_PROJECT_PATH) BUILD_DIR_BASE=$(SDK_BUILD_BASE)
	$(Q) mkdir -p $(SDK_BUILD_BASE)/lib
	$(Q) rm -rf $(SDK_BUILD_BASE)/lib/*
	$(Q) for i in $$(find $(SDK_BUILD_BASE) -name '*.a'); do \
		cp $$i $(SDK_BUILD_BASE)/lib/.; \
	done
```

slaff, 2020-08-07:
Here is a short how-to to use the latest release/v4.0 ESP-IDF SDK. 

First make sure to download the latest `release/v4.0` version. This can be done using the command below:
```
git clone -b release/v4.0 --recursive https://github.com/espressif/esp-idf.git
```

Once the download finishes it is time setup ESP-IDF. On a Linux OS this can be done with the commands below. If you are using other OS see the official [ESP-IDF documentation](https://docs.espressif.com/projects/esp-idf/en/v4.0/get-started/index.html#step-1-install-prerequisites):
```
cd esp-idf
./install.sh
```

After this you will need the ESP-IDF environmental variables. They can be set using the commands below:
```
source <path-to>/esp-idf/export.sh
export IDF_PATH=<path-to>/esp-idf/
```

You can add these commands to your `.bash_profile` file so that you have them set always.

Now it is time to compile new static libraries from the latest SDK. This can be done using the commands below:
```
cd  $SMING_HOME/../Basic_Blink
make dist-clean
make SMING_ARCH=Esp32 sdk-menuconfig
make sdk-build
```

The newly compiled libraries will be kept until you run again `make dist-clean`. In general this means that you have to compile ESP-IDF libraries only once.

Once you have your ESP-IDF libraries it is finally time to compile the samples and flash them on the device. This can be done with the commands below:
```
cd  $SMING_HOME/../Basic_Blink
make  SMING_ARCH=Esp32
make flash
````

And that should be it.

> So I suspect that some mixing happens between the $IDF_PATH includes and the Sming provided (older) esp-idf libs.

Exactly. From our side we should probably do some changes to the build process to make it a bit easier to integrate with new/old ESP-IDF v4.0.x versions.
- [ ] Remove the pre-compiled libraries from the git repository
- [ ] And make sure to run a new ESP-IDF compilation if there are no precompiled libraries

closed 2020-08-13 by slaff