SmingHub/Sming #2103: App crash on SPIFFS when booting rom slot 1

kmihaylov, 2020-08-15:
When application is using only one SPIFFS address space for both roms (in my case 0x1....), the second (or newly flashed) rom fails to boot (probably because spiffs_mount problem).

**If separate addressing is used for SPIFFS partition for rom 0 and rom 1 (0x1xxxx and 0x3xxxx), no problem occurs.
Also if the app uses only one SPIFFS partition located at 0x3xxxx, again no problem occurs (multiple OTA updates and successfull boots from both slot 0 and slot 1).**

The decoded stacktraces are below!
_Attached is the Basic_rBoot example with minor changes, that shows similar, but not the same behaviour (it successfully updates and switches to slot 1, mounting it from 0x8200 for some reason, but breaks when OTA-ing slot 0)._
[rboot_spiffs.tar.gz](https://github.com/SmingHub/Sming/files/5079279/rboot_spiffs.tar.gz)

Here I'm pasting my serial output, including the bootloader messages and other information:
```
rBoot_config, romSlot: 0 , addr1: 0000002000 , addr2: 0000102000
Sming>
In callback...Firmware updated, rebooting to rom 1...

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 1072, room 16 
tail 0
chksum 0x50
load 0x3ffe8000, len 576, room 8 
tail 8
chksum 0x41
csum 0x41

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz

Booting rom 1 at 102000, load addr 15a3c0.
~�2
    �Nn�䒃� ����
                 �no��ۂ�l�bl`r�olph���p�o��{��N|�
                                                    l$ll c�
                                                                ��2r�l�N�
                                                                          �o�l`��s�p��l2�

                                                                                            l ��{�$���

                                                                                                         $ �.��110438 Commandhandlercommand status registered
110869 Commandhandlercommand echo registered
111129 Commandhandlercommand help registered
111876 Commandhandlercommand debugon registered
116130 Commandhandlercommand debugoff registered
120489 Commandhandlercommand command registered
Welcome to the Stream Command executor
128056 fs.start: size:64 Kb, offset:0x100000

133726 mount res: 0
260380 FS check res: 0
261198 Commandhandlercommand debug registered
261381 
Currently running rom 1.

261730 Commandhandlercommand restart registered
263323 Commandhandlercommand heap registered
267280 Commandhandlercommand hostname registered
271667 Commandhandlercommand ls registered
275540 Commandhandlercommand cat registered
279480 Commandhandlercommand fscheck registered
283742 Commandhandlercommand fsformat registered
288035 Commandhandlercommand ver registered
291819 SPIClass::setClock(prediv 20, cntdiv 2) for target 2000000
297803 Commandhandlercommand updateFW registered
302084 Commandhandlercommand switchFW registered
306423 Commandhandlercommand deviceInfo registered
sleep disable
332482 SPIClass::setClock(prediv 10, cntdiv 2) for target 4000000
332951 Commandhandlercommand mqtt registered
[... 118 lines cut ...]

Decoded stack-trace after the WDT reset:
```
0x40209930: scan_start at ??:?
0x4020baa1: chm_end_op at ??:?
0x40209930: scan_start at ??:?
0x402484f4: StationImpl::smartConfigStart(SmartConfigType, Delegate<bool (SmartConfigEvent, SmartConfigEventInfo const&)>)::{lambda(sc_status, void*)#1}::_FUN(sc_status, void*) at StationImpl.cpp:?
0x4020ba93: chm_end_op at ??:?
0x4020bae4: chm_cancel_op at ??:?
0x40209930: scan_start at ??:?
0x40209aac: scan_cancel at ??:?
0x40209868: scan_start at ??:?
0x4020bd95: cnx_sta_scan_cmd at ??:?
0x4022d1bb: pp_enable_noise_timer at /home/xcg/workspace/debug/esp8266_nonos_sdk_core_20180510/app/pp/pp.c:323
0x40215d9e: wifi_get_opmode at ??:?
0x402484f4: StationImpl::smartConfigStart(SmartConfigType, Delegate<bool (SmartConfigEvent, SmartConfigEventInfo const&)>)::{lambda(sc_status, void*)#1}::_FUN(sc_status, void*) at StationImpl.cpp:?
0x40216cad: wifi_station_scan at ??:?
0x402484f4: StationImpl::smartConfigStart(SmartConfigType, Delegate<bool (SmartConfigEvent, SmartConfigEventInfo const&)>)::{lambda(sc_status, void*)#1}::_FUN(sc_status, void*) at StationImpl.cpp:?
0x40248071: StationImpl::onSystemReady() at /opt/Sming/Sming/Arch/Esp8266/Platform/StationImpl.cpp:305
 (inlined by) StationImpl::onSystemReady() at /opt/Sming/Sming/Arch/Esp8266/Platform/StationImpl.cpp:302
0x4024808c: non-virtual thunk to StationImpl::onSystemReady() at /opt/Sming/Sming/Arch/Esp8266/Platform/StationImpl.h:62
0x40257c88: SystemClass::onReady(ISystemReadyHandler*)::{lambda(void*)#1}::_FUN(void*) at /opt/Sming/Sming/Platform/System.h:165
0x402185f6: wifi_set_status_led_output_level at ??:?
0x402579be: SystemClass::taskHandler(ETSEventTag*) at /opt/Sming/Sming/Platform/System.cpp:46
```

slaff, 2020-08-17:
@kmihaylov Try to flash the same APP on rom 0 and then on rom 1 over OTA. The stack trace points to a problem in the smartconfig code / WIFI and not in the flash reading/writing.

kmihaylov, 2020-08-17:
@slaff would you please try the attached project (1st post)? Try OTA several times with the same firmware.

slaff, 2021-05-06:
@kmihaylov Are you still having this issue with the latest `develop` version?

kmihaylov, 2021-05-30:
@slaff with the latest develop f2bcf7960b692f2f69796397da49d333a40237bc and the new Ota updated everything works fine.

With few versions earlier (with the bit older now rBoot updater) everything workef fine too, but under the condition that I use rom0 and rom1 images for flashing the appropriated slots.

closed 2021-05-30 by kmihaylov