SmingHub/Sming #2128: Does Sming only detects no internet when disconnected from wifi router?

Vinimuller, 2020-10-28:
Hello!

I'm using Sming 3.8 (sorry for that) and MqttClient. Seems that Mqtt disconnect event is only fired when ESP is disconnected from the wifi router. If the wifi router keeps on and the only the connection to internet goes off, MqttClient does not fire the disconnect event. Is it a known issue? Is it solved at Sming 4?

slaff, 2020-10-29:
>  Is it a known issue? Is it solved at Sming 4?

Can you try to compile Sming AND your application using DEBUG_VERBOSE_LEVEL=3 ? Once this is done flash the application on your device and let it connect to internet. After that stop the internet connection (unplug the fiber cable for example) but leave the Wifi router running. Copy the log information and paste it here.

Vinimuller, 2020-10-29:
Sure, here it is
```
[MqttClient:221] > MQTT status: MQTT_MSG_PINGRESP (len: 2)
[TcpConnection:104] TCP received: 2 bytes
[TcpConnection:165] onReadyToSendData: 1 // DISCONNECTED FIBER CABLE HERE
[TcpClient:87] Storing 2 bytes in stream
[TcpConnection:269] Written: 2, Available: 2, isFinished: 1, PushCount: 1 [TcpBuf: 2778]
[TcpClient:173] TcpClient stream finished
[TcpClient:87] Storing 2 bytes in stream
[TcpConnection:269] Written: 2, Available: 2, isFinished: 1, PushCount: 1 [TcpBuf: 2776]
[TcpClient:173] TcpClient stream finished
[TcpClient:87] Storing 2 bytes in stream
[TcpConnection:269] Written: 2, Available: 2, isFinished: 1, PushCount: 1 [TcpBuf: 2774]
[TcpClient:173] TcpClient stream finished
[TcpClient:87] Storing 2 bytes in stream
[TcpConnection:269] Written: 2, Available: 2, isFinished: 1, PushCount: 1 [TcpBuf: 2772]
[TcpClient:173] TcpClient stream finished
[TcpClient:87] Storing 2 bytes in stream
[TcpConnection:269] Written: 2, Available: 2, isFinished: 1, PushCount: 1 [TcpBuf: 2770]
[TcpClient:173] TcpClient stream finished
[TcpClient:87] Storing 2 bytes in stream
[TcpConnection:269] Written: 2, Available: 2, isFinished: 1, PushCount: 1 [TcpBuf: 2768]
[TcpClient:173] TcpClient stream finished
[TcpClient:87] Storing 2 bytes in stream
[TcpConnection:269] Written: 2, Available: 2, isFinished: 1, PushCount: 1 [TcpBuf: 2766]
[TcpClient:173] TcpClient stream finished
[TcpClient:87] Storing 2 bytes in stream
[TcpConnection:269] Written: 2, Available: 2, isFinished: 1, PushCount: 1 [TcpBuf: 2764]
[TcpClient:173] TcpClient stream finished
[TcpClient:87] Storing 2 bytes in stream
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[DataSourceStream:72] realloc 2 -> 133
[TcpClient:87] Storing 2 bytes in stream
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE // RECONNECTED FIBER CABLE AROUND HERE
[TcpClient:87] Storing 2 bytes in stream
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[TcpConnection:249] WAIT FOR FREE SPACE
[dataHandler:243] === BROKER DISCONNECTED === // AFTER FEW SECONDS OF RECONNECTING FIBER CABLE, SMING FIRE THE DISCONNECT EVENT
[dataHandler:255] MQTT Broker Unreachable!!
[TcpConnection:159] TCP connection error: -9
```

Vinimuller, 2020-11-04:
@slaff Any idea on what is happening?