SmingHub/Sming #2158: http_parser failing with HPE_CLOSED_CONNECTION

mikee47, 2020-11-30:
The problem is occuring when the UPnP scan tool https://github.com/mikee47/Sming-UPnP/tree/develop/tools/scan.

Here is a log extract:

```
14479905 Fetching '(service) http://192.168.1.254:1900/gateconnSCPD.xml' -> 'out/upnp/devices/192.168.1.254/1900/urn:schemas-upnp-org:service:WANIPConnection:1.xml''
14483388 Fetching description from URL: 'http://192.168.1.254:1900/gateconnSCPD.xml'
14489705 TCP 3e53968 +connection
14497519 TCP 3e53968 connect to "192.168.1.254:1900"
14499531 TCP connect result: 0
14500941 TCP 3e53968 connected: useSSL: 0, Error: 0
14505351 TCP 3e53968 connected
14506302 MemoryDataStream::realloc 0 -> 32
14511211 Storing 32 bytes in stream
14515110 MemoryDataStream::realloc 32 -> 186
14516623 Storing 26 bytes in stream
14519555 Storing 19 bytes in stream
14520969 Storing 2 bytes in stream
14525029 TCP 3e53968 onReadyToSendData: 0
14527349 TCP 3e53968 Written: 79, Available: 2780, isFinished: 0, PushCount: 1
14531706 TcpClient stream finished
14534160 TCP 3e53968 sent: 79
14537567 TCP 3e53968 onReadyToSendData: 2
14543533 HTTP parser error: HPE_CLOSED_CONNECTION
14545742 TcpClient::onReceive: Aborted from receive callback
```

The received header looks like this:

``
HTTP/1.1 200 OK
CONTENT-LENGTH: 12924
CONTENT-TYPE: text/xml
DATE: Mon, 30 Nov 2020 09:29:43 GMT
LAST-MODIFIED: Thu, 01 Jan 1970 00:00:19 GMT
SERVER: Linux/2.6.36.4brcmarm, UPnP/1.0, Portable SDK for UPnP devices/1.6.19
X-User-Agent: redsonic
CONNECTION: close
```

http_parser appears to be declaring the connection dead before the content has been processed.

slaff, 2020-11-30:
> http_parser appears to be declaring the connection dead before the content has been processed.

Yes, that can happen when the response contains Connect header with content "close". But we should be already handling this case: https://github.com/SmingHub/Sming/blob/develop/Sming/Core/Network/Http/HttpClientConnection.cpp#L139. 

Can I simulate the device somehow? My TV is closing the connections and things work as expected, but your device may things in a way that is not handled correctly. Or can you run Wireshark, sniff the communication and send me a capture so that I can analyze it locally?

mikee47, 2020-11-30:
One of the devices is my network router. Does the scanner run to completion without any failures on your system?

mikee47, 2020-11-30:
I can post a full log of the scan at DBG verbosity. Would that help?

slaff, 2020-11-30:
> One of the devices is my network router. Does the scanner run to completion without any failures on your system?

It was working just fine compiled under the Host architecture. Let me test with the absolute latest version from some minutes ago.

> I can post a full log of the scan at DBG verbosity. Would that help?

Yes, please.

mikee47, 2020-11-30:
[upnp.zip](https://github.com/SmingHub/Sming/files/5617049/upnp.zip)

mikee47, 2020-11-30:
That was taken with https://github.com/mikee47/Sming-UPnP/tree/fix/scanner-updates

slaff, 2020-11-30:
The scanner works for me but I don't have a device like your one. Having a [Wireshark](https://www.wireshark.org/download.html) capture would help a lot more. This way I can see what is actually happening on the network level.

Looking at the log I can see the following:

Short version
------------------
1st request to your redsonic device is successful followed by expected close.
2nd request to your redsonic device seems to be sent but it never gets completed. ... I will need also to see Wireshark capture to following that particular TCP stream. 

Long version
------------------
Annotated excerpt from the log below:

```
SSDP has received and answer from a device. The device location is http://192.168.1.64:54380/SRS-X88_DDD.xml      
            
            13284715 [SSDP] RX 192.168.1.64:53462 response: 8 headers
            13288019 Found match for SSDP {upnp:rootdevice}
            13289995   location: http://192.168.1.64:54380/SRS-X88_DDD.xml
            13297845   usn: uuid:00000000-0000-1010-8000-ac9b0a411929::upnp:rootdevice
            13300214 Writing out/upnp/devices/192.168.1.64/54380/ssdp-upnp-rootdevice.txt
            13301001 Opened file 'out/upnp/devices/192.168.1.64/54380/ssdp-upnp-rootdevice.txt', 0 bytes

A new HttpClientConnection is created to host:port{ 192.168.1.254:1900 }. The HTTP request URL path is: /gatedesc.xml

            15303292 Fetching '(device) http://192.168.1.254:1900/gatedesc.xml' -> 'out/upnp/devices/192.168.1.254/1900/gatedesc.xml'', attempt #1
            15303648 Fetching description from URL: 'http://192.168.1.254:1900/gatedesc.xml'
            15304801 Creating new HttpClientConnection
            15309688 HCC::connect: TCP state: -1, isStarted: 0, isActive: 0
            15311614 HCC::connect: connecting ...
            15314645 TCP 3f039d8 +connection
            15315877 TCP 3f039d8 connect to "192.168.1.254:1900"
            15318893 TCP connect result: 0
            15322335 TCP 3f039d8 connected: useSSL: 0, Error: 0
            15327378 TCP 3f039d8 connected
            15330523 HCC::onReadyToSendData: executionQueue: 0, waitingQueue.count: 1
            15335576 MemoryDataStream::realloc 0 -> 28
            15336619 Storing 28 bytes in stream
            15337257 MemoryDataStream::realloc 28 -> 182
            15339656 Storing 26 bytes in stream
            15345836 Storing 19 bytes in stream
            15349708 Storing 2 bytes in stream
            15351141 TCP 3f039d8 onReadyToSendData: 0
            15352166 TCP 3f039d8 Written: 75, Available: 2780, isFinished: 0, PushCount: 1
            15352992 TcpClient stream finished
            15357137 HCC::onReadyToSendData: executionQueue: 1, waitingQueue.count: 0
            15360591 HCC::onReadyToSendData: Nothing in the waiting queue
            15362434 TCP 3f039d8 sent: 75
            15366036 HCC::onReadyToSendData: executionQueue: 1, waitingQueue.count: 0
            15368895 HCC::onReadyToSendData: Nothing in the waiting queue
            15369740 TCP 3f039d8 onReadyToSendData: 2
            15370373 The headers are complete
            15372194 MemoryDataStream::realloc 0 -> 3455
            15372765 TCP 3f039d8 received: 270 bytes
            15379017 HCC::onReadyToSendData: executionQueue: 0, waitingQueue.count: 0
            15380564 HCC::onReadyToSendData: Nothing in the waiting queue
            15385495 TCP 3f039d8 onReadyToSendData: 1
            15393755 TCP 3f039d8 received: 1390 bytes
            15395602 HCC::onReadyToSendData: executionQueue: 0, waitingQueue.count: 0
            15402231 HCC::onReadyToSendData: Nothing in the waiting queue
            15407358 TCP 3f039d8 onReadyToSendData: 1
            15414206 TCP 3f039d8 received: 1390 bytes
            15416110 HCC::onReadyToSendData: executionQueue: 0, waitingQueue.count: 0
            15418886 HCC::onReadyToSendData: Nothing in the waiting queue
            15423937 TCP 3f039d8 onReadyToSendData: 1
            15428798 HCC::onMessageComplete: Execution queue: 0, http://192.168.1.254:1900/gatedesc.xml
            15431750 Received description
            15434663 MemoryDataStream::realloc 3455 -> 3520
            
The Http request was sent successfully and there was complete response         
[... 142 lines cut ...]
            
            17979285 TCP 3f039d8 receive: pbuf is NULL
            17985056 TCP 3f039d8 received: (null)
            17987974 TCP 3f039d8 connection closing
            
!!! We are reconecting.... No request is resend because there is nothing in the waitingQueue... But still we have one sent request 
in the execution queue that has not received yet its response !!!!            
            
            17990924 HCC::onClosed: Trying to reconnect and send pending requests
            17993876 HCC::connect: TCP state: -1, isStarted: 1, isActive: 0
            17995359 HCC::connect: connecting ...
            17995896 TCP 3f039d8 +connection
            18004111 TCP 3f039d8 connect to "192.168.1.254:1900"
            18004762 TCP connect result: 0
            18009449 TCP 3f039d8 connected: useSSL: 0, Error: 0
            18013378 TCP 3f039d8 connected
            18016303 HCC::onReadyToSendData: executionQueue: 1, waitingQueue.count: 0
            18019236 HCC::onReadyToSendData: Nothing in the waiting queue
            18021697 TCP 3f039d8 onReadyToSendData: 0
           
The connection gets closed (probably due to timeout from our side or the remote side)            
            
            18289712 -TCP connection

```

mikee47, 2020-11-30:
OK, this test fetches two URLs to begin with, the second fails with bad request. Have included wireshark dump!

[upnp1.zip](https://github.com/SmingHub/Sming/files/5619058/upnp1.zip)

slaff, 2020-12-01:
From the capture above I can see that the first and the second requests complete successfully. But then comes an explained, at least for me, response with bad request. From the capture it looks like the HTTP server pushed the response ( see line with yellow )?!

[image]

# Tcp Stream 20

Following the TCP stream I can see the following:

[image]

1. The communication is initiated from the client (which is good). You can see the initial SYN and ACK TCP packets. 
2. But then it looks like the HTTP server sends immediately to the http client a response!? Plus forces the client TCP stack not buffer the data using the TCP Push flag. The HTTP response itself is using HTTP version 0.0 ?! and the response says that there was a bad request. 
3. The response headers contain Connection:close so the HTTP parser thinks that the TCP connection will be closed/should be closed.
4. Right after that the Http client sends the GET request. Which will not work as the HTTP server is in unexpected/undefined state.

That behaviour is not exactly typical.

@mikee47 What is the exact model of this router - maybe I can find a cheap second-hand one on ebay and test locally? My wild guess is that either the router misbehaves (as the network capture hints at) or the http client does not close the connection as expected from the TCP stack of the router.

# Tcp Stream 19

If the TCP stream caused a problem the we should analyze TCP stream 19. 

[image]

As shown below I cannot see a place where the Http client misbehaved. It sends the request without any unexpected trailing bytes. The HTTP response comes to the client, after a bit of pushing. It seems also that the Http client closes the connection (The FIN packet is initiated from the client IP). So it seems to me that the traffic in stream 19 was as expected. Which leads to me to a second suspicion - the LWIP TCP stack on the host or our TcpConnection looses packets somewhere and we are unaware about it.... ...

Is there a network GURU that can spot a problem in Tcp Stream 19?

mikee47, 2020-12-01:
I've pushed my test code to https://github.com/mikee47/Sming-UPnP/tree/fix/failed-http-requests.
It queues two description fetches, and the second one always fails. This happens with other devices - my Sony speaker, for example.
I've increased the 'no response' timeout from 2 seconds to 10 seconds and as you can see in the attached log although it succeeds every fetch after the first requires two attempts.

On your system, do all fetches succeed at the first attempt? If not, maybe reduce the retry timeout and see if that forces the fault. It feels like there's a race condition going on somewhere, and in Linux things just run that bit quicker so it never occurs.

[log.zip](https://github.com/SmingHub/Sming/files/5624013/log.zip)

mikee47, 2020-12-01:
@slaff Thanks for your help with this. I'm pursing my own leads but nothing to report yet...

mikee47, 2020-12-01:
Scanner now runs like a champ. It almost works now without any fetch delays at all, but I've reduced them and a full scan takes < 40seconds without any issues whatsoever.

[upnp.zip](https://github.com/SmingHub/Sming/files/5626109/upnp.zip)

mikee47, 2020-12-03:
I noted an issue in the ControlPoint response handling which now seems to be almost certainly related to the same issue.
See https://github.com/mikee47/Sming-UPnP/blob/3e66a17cbd9c9287485a4a2b177a79a0744d2f64/src/ControlPoint.cpp#L211 onwards.

There are two log files attached. In (2) the response is wrong, and it seems like that server is responding to the previous request rather than the intended one.

[debug-logs.zip](https://github.com/SmingHub/Sming/files/5637985/debug-logs.zip)

**controlpoint-task-queue.txt** (1)

The user callback is invoked via task queue. I wanted this because the stack is getting a bit busy by this point, and I couldn't think of any good reasons why this should **not** work!

Line 241: Send a POST request
Line 514: Response to that POST request

Compare this with the same request/response in (2)...

**controlpoint-no-task-queue.txt** (2)

The user callback is invoked directly.

Line 257: The POST request (same one as in (1)
Line 362: The response, as expected.

mikee47, 2020-12-03:
Note: The actual request/response handling is in `ActionRequest::send()` https://github.com/mikee47/Sming-UPnP/blob/3e66a17cbd9c9287485a4a2b177a79a0744d2f64/src/ActionRequest.cpp#L47 so actually unrelated...

mikee47, 2020-12-03:
@slaff The reason for the request failure is that the request body stream has been eaten, so re-attempting the request fails:

```
POST /ctl/ContentDir HTTP/1.1
Content-Type: text/xml; charset="utf-8"
SOAPACTION: "urn:schemas-upnp-org:service:ContentDirectory:1#Browse"
Host: 192.168.1.1:8200
Content-Length: 0
Transfer-Encoding: 

HTTP/1.1 500 Internal Server Error
Content-Type: text/xml; charset="utf-8"
Connection: close
Content-Length: 396
Server: 3.13.0-32-generic DLNADOC/1.50 UPnP/1.0 MiniDLNA/1.1.0
Date: Thu, 03 Dec 2020 21:23:14 GMT
EXT:

<s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/" s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body><s:Fault><faultcode>s:Client</faultcode><faultstring>UPnPError</faultstring><detail><UPnPError xmlns="urn:schemas-upnp-org:control-1-0"><errorCode>402</errorCode><errorDescription>Invalid Args</errorDescription></UPnPError></detail></s:Fault></s:Body></s:Envelope>

```

slaff, 2020-12-04:
> The reason for the request failure is that the request body stream has been eaten:

This is of course something to be expected. We don't keep a copy of the body.

slaff, 2020-12-06:
@mikee47 I was able to reproduce the issue using an apache server that serves the description and the service xml files and sends Connection:close:
```vhost
	<Location "/ssdp">
		Header set Connection close
	</Location>
```

The content of the ssdp folder is attached as zip to this comment.

[ssdp.zip](https://github.com/SmingHub/Sming/files/5648686/ssdp.zip)

PR #2175 addresses the issue but I have to test it also with other devices and web servers. IF PR #2175 works for you I would recommend you to decrease or completely remove the delays in the UPnP scanner app.

PR #2175 (2020-12-08): Improve connection:close handling for HTTP clients.
Attempt to fix #2158.

- [x] Test against real UPnP devices that cannot handle keep-alives/pipelining.
closed 2020-12-08 by slaff