SmingHub/Sming #2161: Cannot compile with latest Sming develop

teicors, 2020-12-01:
Hi,
I recently upgraded the Sming env. (git pull) I cannot compile anymore: the file 
/opt/sming/Sming/Arch/Esp8266/Components/libc/include/c_types.h
is missed. I got it from the prev version of Sming in github and I can compile now.

mikee47, 2020-12-01:
That file has moved into Sming/System/include as it is required for all architectures. I suspect you may have files left over from the previous version. Can you check with `git status` from your root Sming directory and remove any extraneous files.

teicors, 2020-12-01:
I did before a 
`make dist-clean config-clean components-clean submodules-clean`
but no luck. After your direction and checking with 
`git status` 
I found some spurious files
```
	Sming/Arch/Esp8266/Components/gdbstub/appcode/gdb_hooks.cpp.d
	Sming/Arch/Esp8266/Components/gdbstub/appcode/gdb_hooks.o
	Sming/Arch/Esp8266/Components/gdbstub/appcode/gdbstub-entry.o
	Sming/Arch/Esp8266/Components/libc/include/c_types.h
	Sming/Components/rboot/appcode/rboot-overrides.c.d
	Sming/Components/rboot/appcode/rboot-overrides.o
```
I removed them and I can recompile.
Thanks !

closed 2020-12-01 by teicors