SmingHub/Sming #2182: Cannot receive big files on Linux Mint 20

Matzz, 2020-12-12:
Hi,
I have quite weird problem. I cannot fetch huge files using WebServer and spiffs. When I request them `curl` / `netcat` just hangs.
It happens only when I access ESP8266 wifi using my Linux. When I switch to Windows on the same machine everything is working fine. Android phone is also downloading this file correctlly..
I wouldn't create that issue if I had some other networking issues with my laptop. However, all web pages using home wifi works fine on linux.

I made some sample project which I use to reproduce that issue. All logs are attached there.
 
https://github.com/Matzz/SmingBigFileIssue

# Linux
```
hostnamectl
   Static hostname: ---
         Icon name: ---
           Chassis: laptop
        Machine ID: ---
           Boot ID: ---
  Operating System: Linux Mint 20
            Kernel: Linux 5.4.0-56-generic
      Architecture: x86-64

```
Files mounted to spiffs:
```
rw-rw-r-- 1 matzz matzz  84K Dec 12 23:26 jquery.min.js
-rw-rw-r-- 1 matzz matzz   13 Dec 12 21:37 test.txt
```

## Working file on Linux Mint 20 (test.txt):

### Request:
```
echo -e "GET /get?file=test.txt\n\n" | nc 192.168.4.1 80
HTTP/1.1 200 OK
Content-Type: text/plain
ETag: "00f-2-d0-8"
Content-Length: 13
Connection: keep-alive
Server: HttpServer/Sming

a
b
cde
1234
```

### Logs:

```
271542673 onAccept, tcp: 3fff1360, state: 0 K=0, Free heap size=43904
271542976 TCP 3fff14c8 +connection
271543345 TCP 3fff14c8 timeout updating: 70 -> 0
271545457 Opening connection. Total connections: 1
271549902 TcpServer onClient: 192.168.4.2

271555572 The headers are complete
271571410 found test.txt
271571793 attached file: 'test.txt' (13 bytes) #0x3fff1170
271572899 MemoryDataStream::realloc 0 -> 17
271573062 Storing 17 bytes in stream
271575532 MemoryDataStream::realloc 17 -> 171
271579152 Storing 26 bytes in stream
271582472 Storing 20 bytes in stream
271585781 Storing 20 bytes in stream
271589083 Storing 24 bytes in stream
271592359 Storing 26 bytes in stream
271595625 Storing 2 bytes in stream
271598876 TCP 3fff14c8 onReadyToSendData: 1
271602962 TCP 3fff14c8 Written: 135, Available: 2780, isFinished: 0, PushCount: 1
271609967 TcpClient stream finished
[... 132 lines cut ...]
620094714 TCP 3fff17d8 Written: 1024, Available: 1390, isFinished: 0, PushCount: 1
620101605 TCP 3fff17d8 Written: 366, Available: 366, isFinished: 0, PushCount: 2
620109301 TCP 3fff17d8 sent: 1390
620111484 TCP 3fff17d8 onReadyToSendData: 2
620115989 TCP 3fff17d8 Written: 1024, Available: 1390, isFinished: 0, PushCount: 1
620122862 TCP 3fff17d8 Written: 366, Available: 366, isFinished: 0, PushCount: 2
620130797 TCP 3fff17d8 sent: 1390
620132829 TCP 3fff17d8 onReadyToSendData: 2
620137338 TCP 3fff17d8 Written: 1024, Available: 1390, isFinished: 0, PushCount: 1
620144289 TCP 3fff17d8 Written: 366, Available: 366, isFinished: 0, PushCount: 2
620151919 -TCP connection
620154013 TCP 3fff17d8 sent: 1390
620156519 TCP 3fff17d8 onReadyToSendData: 2
620161049 TCP 3fff17d8 Written: 1024, Available: 1390, isFinished: 0, PushCount: 1
620167890 TCP 3fff17d8 Written: 366, Available: 366, isFinished: 0, PushCount: 2
620175863 TCP 3fff17d8 sent: 1390
620177848 TCP 3fff17d8 onReadyToSendData: 2
620182434 TCP 3fff17d8 Written: 1024, Available: 1390, isFinished: 0, PushCount: 1
620189231 TCP 3fff17d8 Written: 366, Available: 366, isFinished: 0, PushCount: 2
621298272 TCP 3fff17d8 connection closed by timeout: 1 (from 0)
621298620 TCP 3fff17d8 connection closing
621299493 TCP 3fff17d8 ~connection
621299874 Destroying connection. Total connections: 0
621548015 -TCP connection
```

slaff, 2020-12-14:
> [Here it hangs...]

Install Wireshark on the Linux machine where the problem happens. Reproduce the problem and during the reproduction capture with Wireshark  the incoming and outgoing data from the device to your computer and save it to file and send us the file (attach it to your next comment). My wild guess is that the [TCP MSS](https://en.wikipedia.org/wiki/Maximum_segment_size) value is problematic for your distribution.

slaff, 2020-12-14:
> 352823017 TCP 3fff14c8 connection closed by timeout: 1 (from 0)

Or maybe the timeout on the server is too low. You can increase the default server timeout to a bigger value like 5 or 10. 
```c++
HttpServer server;

void gotIP(IpAddress ip, IpAddress netmask, IpAddress gateway)
{
   //...
   HttpServerSettings settings;
   settings.keepAliveSeconds = 5;
   server.configure(settings);
}

Matzz, 2020-12-14:
I thought that it might be my laptop / distro related. So I downloaded Fedora live cd and started it on a different computer.
However the results were exactly the same. So it is Linux / Linux kernel related.

I allso tried to set folowing settings but without any success.
```
	settings.keepAliveSeconds = 10;
	server->setTimeOut(5000);
```

I also tested different file which contains many short lines rather than very long line like jquery.min.js. No luck.

Matzz, 2020-12-14:
Wireshark logs:
```
echo -e "GET /get?file=jquery.min.js\n" | netcat 192.168.4.1 80
```
```
1	0.000000000	192.168.4.2	192.168.4.1	TCP	74	42026 → 80 [SYN] Seq=0 Win=64480 Len=0 MSS=1240 SACK_PERM=1 TSval=2857396046 TSecr=0 WS=128
2	0.004989016	192.168.4.1	192.168.4.2	TCP	58	80 → 42026 [SYN, ACK] Seq=0 Ack=1 Win=5560 Len=0 MSS=1390
3	0.005063974	192.168.4.2	192.168.4.1	TCP	54	42026 → 80 [ACK] Seq=1 Ack=1 Win=64480 Len=0
4	0.005234305	192.168.4.2	192.168.4.1	HTTP	83	GET /get?file=jquery.min.js 
5	0.113913378	192.168.4.1	192.168.4.2	TCP	54	80 → 42026 [ACK] Seq=1 Ack=30 Win=5531 Len=0
```
Then netcat session hangs.

slaff, 2020-12-15:
> Wireshark logs:

It would be great to send me the wireshark capture. The logs don't contain all the information.

> echo -e "GET /get?file=jquery.min.js\n"

That is not a valid HTTP request to start with. You are missing the version information in the first line as well as a mandatory HOST header. A proper way to to test will be:

```
curl -vvv http://192.168.4.1 80/get?file=jquery.min.js
```

Matzz, 2020-12-15:
@slaff I just wanted to make it minimal to reduce influence other parts of request. Usually GET just works (and it work here for small file).
I sent you Wireshark log for `curl 192.168.4.1/get?file=jquery.min.js -i` request to email address found in Sming git log.

slaff, 2020-12-15:
Short story: Please, send me a sample so that I can reproduce the problem locally? I am also running Linux machines.

Long story: There is nothing suspicion in the TCP connection in the capture that you have send me.  There is just no HTTP response. 

What I can see is that you are actually using your device as access point and you are connected to it. This is an important detail that was omitted initially.  There might be an issue with ESP8266 being used as access point and, at the same time, trying to deliver a large file. But I can say more for sure only if I can reproduce the problem locally and for that I need a sample application from you.
[image]

Matzz, 2020-12-15:
@slaff The minimal code I use is available here - https://github.com/Matzz/SmingBigFileIssue
It should be enough to flash it to ESP8266 and try to connect.
Thanks!

slaff, 2020-12-15:
Works like charm for me. No hiccups, no delays. No issues in Wireshark. I am sending you the compiled code that I have used together with the application output.txt and the config.txt with the configuration options used to compile this application. 

You can test the same compiled code by flashing the stuff (make flash). The commands that I have used to test were:
```
curl -vvv http://192.168.4.1/get?file=jquery.min.js
curl -vvv http://192.168.4.1/test
curl -vvv http://192.168.4.1/get?file=jquery.min.js
curl -vvv http://192.168.4.1/test
curl -vvv http://192.168.4.1/get?file=jquery.min.js
```

[SmingBigFileIssue.zip](https://github.com/SmingHub/Sming/files/5695514/SmingBigFileIssue.zip)

Matzz, 2020-12-15:
@slaff I flashed your roms that way:
```
python /data/Sming/Sming/Components/esptool/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 --chip esp8266 --before default_reset --after hard_reset write_flash -z -ff 40m -fm dio -fs 4MB 0x00000 out/Esp8266/debug/firmware/rboot.bin 0x002000 out/Esp8266/debug/firmware/rom0.bin 0x100000 out/Esp8266/debug/firmware/spiff_rom.bin
```
and still have the same issue. Maybe its hardware problem of my wemos or some internal persistent config issue? I'll try with another chip and post update here.

Thanks for taking your time.

Matzz, 2020-12-21:
@slaff I ordered brand new wemos chip and have the same effect. Maybe its sth with my ESP/Sming setup wrong?
Do you think such issue could be related to esp_init_data_default.bin (that's is the only firmware piece which I not flashed from your archive SmingBigFileIssue.zip)
 I'll try to download some Sming docker image and try reproduce it there.

anakod, 2020-12-22:
May be issue related to power supply? Can you check it with another USB cable and different power supply?

Matzz, 2020-12-25:
@anakod @slaff Its not a power supply issue.
To be sure that I use clean env I downloaded precompiled esp toolchaing (SmingTools). With the same result.
So the issue is not related to:
- computer (tested on 2 laptops)
- toolchain (downloaded precompiled)
- chip (tested on two)
- command (tested curl, nc, and wget)
- client system (the same on ubuntu and fedora)

The next step was to run ubuntu docker image and try to curl from it. And it... worked. However when I switch docker to `--network=host` issue persist.

Matzz, 2020-12-25:
I finaly resolved it on my laptop. Setting MTU to 1500 fixed that. My defaut value was 1280 (which is somehow related to ip6)
```
3: wlp7s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1280 qdisc noqueue state UP mode DORMANT group default qlen 1000
    link/ether f8:a2:d6:e6:6c:d3 brd ff:ff:ff:ff:ff:ff
```
```
sudo ip link set dev wlp7s0 mtu 1500
```

Matzz, 2020-12-25:
https://github.com/SmingHub/Sming/issues/406

Matzz, 2021-03-10:
Im closing that issue as it seems to not be directly related to Sming.

closed 2021-03-10 by Matzz