SmingHub/Sming #2186: make flash is not flashing the esp8266 modul

vesley, 2020-12-26:
I've installed Sming and the ESP toolcahin as described. I can also successfully build/Compile the Basic_Blink example. But when I try to flash the esp module (it is a Wemos D1 mini), the command starts the pyserial monitor without flashing the module. There are no errors on the console. What I'm doing wrong?
Here is the console output:

```shell
 {develop} /opt/Sming/samples/Basic_Blink$ make flash

Basic_Blink: Invoking 'flash' for Esp8266 (debug) architecture
/opt/Sming/Sming//project.mk:282: Multiple matches found for Component 'esptool':
- /opt/Sming/Sming/Arch/Esp8266/Components/esptool
- /opt/Sming/Sming/Components/esptool
make components application
make[1]: Entering directory '/opt/Sming/samples/Basic_Blink'
/opt/Sming/Sming/project.mk:282: Multiple matches found for Component 'esptool':
- /opt/Sming/Sming/Arch/Esp8266/Components/esptool
- /opt/Sming/Sming/Components/esptool

Building /opt/Sming/samples/Basic_Blink/out/Esp8266/debug/lib/clib-App-ea08091ace175453acca58ebdde21658.a
make[2]: Nothing to be done for 'build'.
Basic_Blink: Linking out/Esp8266/debug/build/app_0.out
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE853D|    1341
    rodata|           ReadOnly Data (RAM)|    3FFE8540|    3FFE88B8|     888
       bss|      Uninitialized Data (RAM)|    3FFE88B8|    3FFEEB68|   25264
      text|            Cached Code (IRAM)|    40100000|    4010639E|   25502
irom0_text|           Uncached Code (SPI)|    40202010|    402323EB|  197595
Total Used RAM : 27493
Free RAM : 54427
Free IRam : 7284
ESPTOOL2 out/Esp8266/debug/firmware/rom0.bin
SSL support disabled
make[1]: Nothing to be done for 'application'.
make[1]: Leaving directory '/opt/Sming/samples/Basic_Blink'
Killing Terminal to free /dev/ttyUSB0
make: [/opt/Sming/Sming/Components/terminal/component.mk:28: kill_term] Killed (ignored)
python3 -m serial.tools.miniterm --raw --encoding ascii /dev/ttyUSB0 115200
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

--- exit ---
```

mikee47, 2020-12-26:
esptool has been moved, indicated by this warning:

```
/opt/Sming/Sming//project.mk:282: Multiple matches found for Component 'esptool':
- /opt/Sming/Sming/Arch/Esp8266/Components/esptool
- /opt/Sming/Sming/Components/esptool
```

Please `rm -f /opt/Sming/Sming/Arch/Esp8266/Components/esptool`.

Explanation: 

When doing a `git pull` please also run `git status` to ensure there aren't any artifacts from previous versions. Specifically, if folders are moved the previous ones may remain must be removed manually.

vesley, 2020-12-26:
I can't sse that the esptool was moved. I removed the folder you mentioned, but still not working. I forgot to say that I'm using Kubuntu.

```shell
 {develop} /opt/Sming/samples/Basic_Blink$ make flash

Basic_Blink: Invoking 'flash' for Esp8266 (debug) architecture
make components application
make[1]: Entering directory '/opt/Sming/samples/Basic_Blink'

Building /opt/Sming/samples/Basic_Blink/out/Esp8266/debug/lib/clib-App-ea08091ace175453acca58ebdde21658.a
make[2]: Nothing to be done for 'build'.
Basic_Blink: Linking out/Esp8266/debug/build/app_0.out
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE853D|    1341
    rodata|           ReadOnly Data (RAM)|    3FFE8540|    3FFE88B8|     888
       bss|      Uninitialized Data (RAM)|    3FFE88B8|    3FFEEB68|   25264
      text|            Cached Code (IRAM)|    40100000|    4010639E|   25502
irom0_text|           Uncached Code (SPI)|    40202010|    402323EB|  197595
Total Used RAM : 27493
Free RAM : 54427
Free IRam : 7284
ESPTOOL2 out/Esp8266/debug/firmware/rom0.bin
SSL support disabled
make[1]: Nothing to be done for 'application'.
make[1]: Leaving directory '/opt/Sming/samples/Basic_Blink'
Killing Terminal to free /dev/ttyUSB0
make: [/opt/Sming/Sming/Components/terminal/component.mk:28: kill_term] Killed (ignored)
python3 -m serial.tools.miniterm --raw --encoding ascii /dev/ttyUSB0 115200
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

--- exit ---
```

After I removed the folder I did git pull, git status:
```shell
 {develop} /opt/Sming$ git pull
Already up to date.
 {develop} /opt/Sming$ git status
On branch develop
Your branch is up to date with 'origin/develop'.

Changes not staged for commit:
  (use "git add/rm <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        deleted:    Sming/Components/esptool/README.rst
        deleted:    Sming/Components/esptool/blank.bin
        deleted:    Sming/Components/esptool/component.mk

Untracked files:
  (use "git add <file>..." to include in what will be committed)
        Sming/Components/spiffs/spiffy/spiffs_cache.o
        Sming/Components/spiffs/spiffy/spiffs_check.o
        Sming/Components/spiffs/spiffy/spiffs_gc.o
        Sming/Components/spiffs/spiffy/spiffs_hydrogen.o
        Sming/Components/spiffs/spiffy/spiffs_nucleus.o
        Sming/Components/spiffs/spiffy/spiffy
        Sming/Components/spiffs/spiffy/spiffy.o

no changes added to commit (use "git add" and/or "git commit -a")
```

closed 2020-12-26 by vesley