SmingHub/Sming #2205: Problems with building Sming

jcarvalhogo, 2021-01-29:
I define the environment variables ESP_HOME=/opt/esp-quick-toolchain
and SMING_HOME = /Volumes/DEV-PART/iot/espressif/Sming/Sming

I follow the following steps in an attempt to build an example

1: I downloaded the project in the repository: https://github.com/SmingHub/Sming.git
2: I run make submodules and waited to download everything
3: I rum the make test

That was my way out:
...
Building /Volumes/DEV-PART/iot/espressif/Sming/samples/Basic_Blink/out/Esp8266/debug/lib/clib-App-.a
C+ /Volumes/DEV-PART/iot/espressif/Sming/samples/Basic_Blink/app/application.cpp
CC /Volumes/DEV-PART/iot/espressif/Sming/Sming/Components/rboot/appcode/rboot-overrides.c
CC /Volumes/DEV-PART/iot/espressif/Sming/Sming/Components/rboot/rboot/appcode/rboot-api.c
CC /Volumes/DEV-PART/iot/espressif/Sming/Sming/Components/rboot/rboot/appcode/rboot-bigflash.c
AS /Volumes/DEV-PART/iot/espressif/Sming/Sming/Arch/Esp8266/Components/gdbstub/appcode/gdbstub-entry.S
C+ /Volumes/DEV-PART/iot/espressif/Sming/Sming/Arch/Esp8266/Components/gdbstub/appcode/gdb_hooks.cpp
AR /Volumes/DEV-PART/iot/espressif/Sming/samples/Basic_Blink/out/Esp8266/debug/lib/clib-App.a
CC rboot-stage2a.c
LD /Volumes/DEV-PART/iot/espressif/Sming/Sming/out/Esp8266/debug/build/rboot/rboot-stage2a.elf
E2 /Volumes/DEV-PART/iot/espressif/Sming/Sming/out/Esp8266/debug/build/rboot/rboot-hex2a.h
CC rboot.c
LD /Volumes/DEV-PART/iot/espressif/Sming/Sming/out/Esp8266/debug/build/rboot/rboot.elf
E2 /Volumes/DEV-PART/iot/espressif/Sming/samples/Basic_Blink/out/Esp8266/debug/firmware/rboot.bin
Prepare libmain
LDGEN out/Esp8266/debug/build/rom0.ld
Basic_Blink: Linking out/Esp8266/debug/build/app_0.out
/opt/esp-quick-toolchain/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/10.2.0/../../../../xtensa-lx106-elf/bin/ld: /Volumes/DEV-PART/iot/espressif/Sming/Sming/out/Esp8266/debug/lib/clib-Sming-.a: file not recognized: file truncated
collect2: error: ld returned 1 exit status
make[2]: *** [out/Esp8266/debug/build/app_0.out] Error 1
make[1]: *** [all] Error 2
make: *** [Basic_Blink] Error 2

I'm using mac osx catalina 10.15.7

I would like to know what I'm doing wrong, could you help me with this?

mikee47, 2021-01-30:
`clib-Sming-.a: file not recognized` is interesting, there should be a hash included in the filename (e.g. `clib-Sming-4f9c9eb6626d9c7d24853beedc58d107.a`). See [COMPONENT_VARS](https://smingdev.readthedocs.io/en/update-getting-started/_inc/Sming/building.html?highlight=variant#envvar-COMPONENT_VARS) for an explanation of this behaviour.

This implies `md5sum` is missing; try `brew install md5sha1sum` and see if that fixes things. Note that this utility is part of [GNU CoreUtils](https://www.gnu.org/software/coreutils/coreutils.html). See also https://wikihub.berkeley.edu/display/drupal/Install+GNU+coreutils+and+other+CLI+utilities+on+MacOS.

mikee47, 2021-01-30:
@jcarvalhogo Could you try something else for me? Edit your `build.mk` file (in $SMING_HOME) and amend line #227 from this:

```
-$(firstword $(shell echo -n $($1) | md5sum -t))
```

to this:

```
+$(firstword $(shell echo -n $($1) | md5 -s))
```

The `md5` utility is standard for MacOS so if this works we can amend the build scripts accordingly.

Thanks.

jcarvalhogo, 2021-01-30:
Hello Mikee47, how are you? I follow your instructions.
first I tried to modify line 227 as you requested, I was not successful, my exit was:

> make tests
md5: option requires an argument - s
usage: md5 [-pqrtx] [-s string] [files ...]
...
after the failure i went back to line 227 as before.

My second alternative was to proceed with the installation of the brew install md5sha1sum as you suggest, I ran the command that you suggest to me and the brew suggests the following: "brew unlink coreutils".

So I follow and everything now works out, below is the brew output

> brew install md5sha1sum
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 2 taps (homebrew/core and homebrew/cask).
==> Updated Formulae
Updated 43 formulae.
==> Renamed Formulae
glibmm@2.64 -> glibmm@2.66                                        pangomm@2.42 -> pangomm@2.46
==> Updated Casks
Updated 4 casks.

==> Downloading https://homebrew.bintray.com/bottles/md5sha1sum-0.9.5_1.catalina.bottle.tar.gz
######################################################################## 100.0%
Error: Cannot install md5sha1sum because conflicting formulae are installed.
  coreutils: because both install `md5sum` and `sha1sum` binaries

Please `brew unlink coreutils` before continuing.

Unlinking removes a formula's symlinks from /usr/local. You can
link the formula again after the install finishes. You can --force this
install, but the build may fail or cause obscure side effects in the
resulting software.
> brew unlink coreutils
Unlinking /usr/local/Cellar/coreutils/8.32... 256 symlinks removed.
> brew install md5sha1sum
==> Downloading https://homebrew.bintray.com/bottles/md5sha1sum-0.9.5_1.catalina.bottle.tar.gz
Already downloaded: /Users/jcarvalho/Library/Caches/Homebrew/downloads/918ca9090cdc150d16b1c126d8d634952bcd687eb2c24f4fdf2c974ebfde31e9--md5sha1sum-0.9.5_1.catalina.bottle.tar.gz
==> Pouring md5sha1sum-0.9.5_1.catalina.bottle.tar.gz
🍺  /usr/local/Cellar/md5sha1sum/0.9.5_1: 8 files, 46.0KB

after "brew install md5sha1sum" was successfully installed to run make samples and everything went well.
Thank you very much for your help, below you have successfully output to most samples

....
Building /Volumes/DEV-PART/iot/espressif/Sming/samples/HttpServer_AJAX/out/Esp8266/debug/lib/clib-App--47e6e3821acdc7abd1efd8b5ebc80fb2.a
C+ /Volumes/DEV-PART/iot/espressif/Sming/samples/HttpServer_AJAX/app/application.cpp
CC /Volumes/DEV-PART/iot/espressif/Sming/Sming/Components/rboot/appcode/rboot-overrides.c
CC /Volumes/DEV-PART/iot/espressif/Sming/Sming/Components/rboot/rboot/appcode/rboot-api.c
CC /Volumes/DEV-PART/iot/espressif/Sming/Sming/Components/rboot/rboot/appcode/rboot-bigflash.c
AS /Volumes/DEV-PART/iot/espressif/Sming/Sming/Arch/Esp8266/Components/gdbstub/appcode/gdbstub-entry.S
C+ /Volumes/DEV-PART/iot/espressif/Sming/Sming/Arch/Esp8266/Components/gdbstub/appcode/gdb_hooks.cpp
AR /Volumes/DEV-PART/iot/espressif/Sming/samples/HttpServer_AJAX/out/Esp8266/debug/lib/clib-App--47e6e3821acdc7abd1efd8b5ebc80fb2.a
E2 /Volumes/DEV-PART/iot/espressif/Sming/samples/HttpServer_AJAX/out/Esp8266/debug/firmware/rboot.bin
Prepare libmain
LDGEN out/Esp8266/debug/build/rom0.ld
HttpServer_AJAX: Linking out/Esp8266/debug/build/app_0.out
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE868C|    1676
    rodata|           ReadOnly Data (RAM)|    3FFE8690|    3FFE90D4|    2628
       bss|      Uninitialized Data (RAM)|    3FFE90D8|    3FFEFE68|   28048
      text|            Cached Code (IRAM)|    40100000|    401065D6|   26070
irom0_text|           Uncached Code (SPI)|    40202010|    4024D160|  307536
Total Used RAM : 32352
Free RAM : 49568
Free IRam : 6716
ESPTOOL2 out/Esp8266/debug/firmware/rom0.bin
[... 18 lines cut ...]
AR /Volumes/DEV-PART/iot/espressif/Sming/samples/HttpServer_Bootstrap/out/Esp8266/debug/lib/clib-App--47e6e3821acdc7abd1efd8b5ebc80fb2.a
E2 /Volumes/DEV-PART/iot/espressif/Sming/samples/HttpServer_Bootstrap/out/Esp8266/debug/firmware/rboot.bin
Prepare libmain
LDGEN out/Esp8266/debug/build/rom0.ld
HttpServer_Bootstrap: Linking out/Esp8266/debug/build/app_0.out
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8640|    1600
    rodata|           ReadOnly Data (RAM)|    3FFE8640|    3FFE9148|    2824
       bss|      Uninitialized Data (RAM)|    3FFE9148|    3FFEFEA8|   28000
      text|            Cached Code (IRAM)|    40100000|    40106732|   26418
irom0_text|           Uncached Code (SPI)|    40202010|    4024C39A|  304010
Total Used RAM : 32424
Free RAM : 49496
Free IRam : 6368
ESPTOOL2 out/Esp8266/debug/firmware/rom0.bin
Checking for spiffs files
No files found in ./files.
Creating empty out/Esp8266/debug/firmware/spiff_rom.bin
Creating rom 'out/Esp8266/debug/firmware/spiff_rom.bin' of size 0x80000 (524288) bytes.
Creating empty filesystem.
SSL support disabled
make[2]: Nothing to be done for `application'.
/Applications/Xcode.app/Contents/Developer/usr/bin/make components application
....

mikee47, 2021-01-30:
I've succeeded in getting an installation of Catalina (in VirtualBox) for testing. After following the instructions [here](https://sming.readthedocs.io/en/latest/arch/esp8266/getting-started/macos.html) I find that `md5sum` is available.

However, I did get an error patching submodules as MacOS `cp` command doesn't have the `-u` option. @jcarvalhogo Did you have this issue?

jcarvalhogo, 2021-01-30:
yes i had the same problem, after i followed his guidance and installed the "brew install md5sha1sum" as things worked

closed 2021-02-25 by slaff