SmingHub/Sming #2206: OtaUpgrade library doesn't work for MacOS

mikee47, 2021-01-30:
On MacOS the `HttpServer_FirmwareUpload` sample fails to build with:

```
C+ /Users/mike/sandboxes/Sming/samples/HttpServer_FirmwareUpload/out/OtaUpgrade/BuildTimestamp.cpp
/Users/mike/sandboxes/Sming/samples/HttpServer_FirmwareUpload/out/OtaUpgrade/BuildTimestamp.cpp:3:48: error: unable to find numeric literal operator 'operator""NLL'
    3 | extern const uint64_t BuildTimestamp PROGMEM = 16120425073NLL - -2208988800000LL;
```

This code is generated from the makefile, `Libraries/OtaUpgrade/component.mk` at line 99:

```
$(Q) echo 'extern const uint64_t BuildTimestamp PROGMEM = $(shell date +%s%3NLL) - $(OTA_DATE_REF);' >> $(OTA_BUILD_TIMESTAMP_SRC)
```

The MacOS bash `date` function doesn't support nanoseconds, therefore the make logic produces `16120425073NLL` with the extraneous 'N' character.

Are nanoseconds actually necessary? If so, could use a python fragment, such as:

```
$(PYTHON) -c "import time; print(int(1000 * time.time()))"
```

mikee47, 2021-01-30:
This line appears to work:

```
$(Q) echo 'extern const uint64_t BuildTimestamp PROGMEM = $(shell $(PYTHON) -c "import time; print('%uLL' % (1000 * time.time()))") - $(OTA_DATE_REF);' >> $(OTA_BUILD_TIMESTAMP_SRC)
```

slaff, 2021-02-01:
> This line appears to work:

Can you put a PR with this fix?

closed 2021-02-04 by mikee47