SmingHub/Sming #2207: Basic_Blink strange communication

adamvm, 2021-01-31:
Hi,

I recently installed Sming on Ubuntu 20.04

Compilation status is ok, I think
```
$ make

Basic_Blink: Invoking 'all' for Esp8266 (debug) architecture
make components application
make[1]: Entering directory '/opt/Sming/samples/Basic_Blink'

Building /opt/Sming/samples/Basic_Blink/out/Esp8266/debug/lib/clib-App-8efa182d93f006624386359166684eb0.a
make[2]: Nothing to be done for 'build'.
Basic_Blink: Linking out/Esp8266/debug/build/app_0.out
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE853D|    1341
    rodata|           ReadOnly Data (RAM)|    3FFE8540|    3FFE88B8|     888
       bss|      Uninitialized Data (RAM)|    3FFE88B8|    3FFEEB68|   25264
      text|            Cached Code (IRAM)|    40100000|    4010639E|   25502
irom0_text|           Uncached Code (SPI)|    40202010|    4023230F|  197375
Total Used RAM : 27493
Free RAM : 54427
Free IRam : 7284
ESPTOOL2 out/Esp8266/debug/firmware/rom0.bin
(!) Spiffs support disabled. Remove 'DISABLE_SPIFFS' make argument to enable spiffs.
SSL support disabled
make[1]: Nothing to be done for 'application'.
make[1]: Leaving directory '/opt/Sming/samples/Basic_Blink'
```

After make flash I see no significant/any errors beside this:
```

seems that command turns on serial terminal, but what I received is:

�bc�cld��cp��#�cxpl'��rn�d�r<�l�{|�d�s|
                                       b���l��#x�
                                                 �c�

continously

mikee47, 2021-01-31:
Hi @adamvm, welcome to Sming!

The garbage you get is the ROM code and bootloader outputting at 74880 baud. Sming uses a default of 115200, but it doesn't set this until later.  You can set different speeds for terminal and flashing like this:

```
make flash COM_SPEED_SERIAL=74480 COM_SPEED_ESPTOOL=921600
```

This should then show you what the problem is. My guess would be that you need to run `make flashinit` to write the ESP8266 configuration data to the device. Also check that you've set `SPI_SIZE` as well; the default for this is 1MB but if your device has 4M then run `make flashinit SPI_SIZE=4M`. Then you can re-run `make flash`.

mikee47, 2021-01-31:
See also https://sming.readthedocs.io/en/latest/arch/esp8266/getting-started/config.html

adamvm, 2021-01-31:
Great thanks, it works.

What was enough is:

```
$make flashinit SPI_SIZE=4M

$make flash
```

closed 2021-01-31 by adamvm