SmingHub/Sming #2208: Switching to GPIO2 as TX on UART0 after initializing does not work.

iliis, 2021-01-31:
I'm using a H801 LED driver board, which uses GPIO2 to transmit UART0 data. The following works:

```C++
Serial.begin(115200, SERIAL_8N1, SERIAL_FULL, 2);
```

While this does not:    

```C++
Serial.begin(115200, SERIAL_8N1, SERIAL_FULL, 1); // Same as Serial.begin(115200);
Serial.setTx(2);
```

In both cases I receive the bootloader output (at 74480 baud), but only the first version produces anything when calling `Serial.write()`.

I've tried both with the current develop (bdf1d31db839385a8e1075f901b9cd95d6fb538d) and master (4.2.0, 7bc3603e22ca91c36d0b6b1bcfa96b5801afa325) branches.

mikee47, 2021-01-31:
@iliis Aha! You have found a bug in the uart driver. Try this patch to `Sming/Arch/Esp8266/Components/driver/uart.cpp`:

```
diff --git a/Sming/Arch/Esp8266/Components/driver/uart.cpp b/Sming/Arch/Esp8266/Components/driver/uart.cpp
index e6c50b94f..22114d381 100644
--- a/Sming/Arch/Esp8266/Components/driver/uart.cpp
+++ b/Sming/Arch/Esp8266/Components/driver/uart.cpp
@@ -910,9 +910,9 @@ void smg_uart_swap(smg_uart_t* uart, int tx_pin)
 bool smg_uart_set_tx(smg_uart_t* uart, int tx_pin)
 {
 	if(uart != nullptr && uart->uart_nr == UART0 && smg_uart_tx_enabled(uart)) {
-		uart1_pin_restore(uart->tx_pin);
+		uart0_pin_restore(uart->tx_pin);
 		uart->tx_pin = (tx_pin == 2) ? 2 : 1;
-		uart1_pin_select(uart->tx_pin);
+		uart0_pin_select(uart->tx_pin);
 		return true;
 	}
```

iliis, 2021-01-31:
That was quick! And yes, it works :)

PR #2210 (2021-02-01): Fix bug in `smg_uart_set_tx()`
Sets pin on wrong uart. Closes #2208.
closed 2021-02-01 by slaff