SmingHub/Sming #2209: SPIFF fails to flash on 1M boards

iliis, 2021-01-31:
I have an H801 LED driver board with 1MB of flash.
Using current Sming (develop, bdf1d31db839385a8e1075f901b9cd95d6fb538d) and the toolchain from https://github.com/SmingHub/SmingTools/releases/download/1.0/x86_64-linux-gnu.xtensa-lx106-elf-e6a192b.201211.tar.gz, the following empty application fails to flash properly:

---

**Makefile:**
```Makefile
include $(SMING_HOME)/project.mk
```

---

**component.mk:**
```Makefile
COMPONENT_SRCFILES := application.cpp
SPI_SIZE=1M
```

---

**application.cpp:**
```C++
#include <SmingCore.h>

void init()
{
}
```

---

When tyring to flash this, I get the following error:

```
$ make flash COM_PORT=/dev/ttyUSB0 SMING_ARCH=Esp8266

test_sming_flash_err: Invoking 'flash' for Esp8266 (debug) architecture
make components application
make[1]: Entering directory '/home/samuel/projects/private/test_sming_flash_err'

Building /home/samuel/projects/private/test_sming_flash_err/out/Esp8266/debug/lib/clib-App-0989b10f3d74cc46049f099e576943fa.a
C+ /home/samuel/projects/private/test_sming_flash_err/application.cpp
CC /home/samuel/programme/esp/Sming/Sming/Components/rboot/appcode/rboot-overrides.c
CC /home/samuel/programme/esp/Sming/Sming/Components/rboot/rboot/appcode/rboot-bigflash.c
CC /home/samuel/programme/esp/Sming/Sming/Components/rboot/rboot/appcode/rboot-api.c
AS /home/samuel/programme/esp/Sming/Sming/Arch/Esp8266/Components/gdbstub/appcode/gdbstub-entry.S
C+ /home/samuel/programme/esp/Sming/Sming/Arch/Esp8266/Components/gdbstub/appcode/gdb_hooks.cpp
AR /home/samuel/projects/private/test_sming_flash_err/out/Esp8266/debug/lib/clib-App-0989b10f3d74cc46049f099e576943fa.a
make[2]: Entering directory '/home/samuel/programme/esp/Sming/Sming/Components/rboot/rboot'
E2 /home/samuel/projects/private/test_sming_flash_err/out/Esp8266/debug/firmware/rboot.bin
make[2]: Leaving directory '/home/samuel/programme/esp/Sming/Sming/Components/rboot/rboot'
Prepare libmain
LDGEN out/Esp8266/debug/build/rom0.ld
test_sming_flash_err: Linking out/Esp8266/debug/build/app_0.out
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE853C|    1340
    rodata|           ReadOnly Data (RAM)|    3FFE8540|    3FFE87E0|     672
       bss|      Uninitialized Data (RAM)|    3FFE87E0|    3FFEEA48|   25192
      text|            Cached Code (IRAM)|    40100000|    40106212|   25106
irom0_text|           Uncached Code (SPI)|    40202010|    40231E07|  196087
Total Used RAM : 27204
Free RAM : 54716
Free IRam : 7680
ESPTOOL2 out/Esp8266/debug/firmware/rom0.bin
Checking for spiffs files
No files found in ./files.
Creating empty out/Esp8266/debug/firmware/spiff_rom.bin
Creating rom 'out/Esp8266/debug/firmware/spiff_rom.bin' of size 0x80000 (524288) bytes.
Creating empty filesystem.
[... 120 lines cut ...]
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 5c:cf:7f:85:66:d0
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 1712 bytes to 1268...
Wrote 1712 bytes (1268 compressed) at 0x00000000 in 0.1 seconds (effective 107.2 kbit/s)...
Hash of data verified.
Compressed 223280 bytes to 165238...
Wrote 223280 bytes (165238 compressed) at 0x00002000 in 14.7 seconds (effective 121.9 kbit/s)...
Hash of data verified.
Compressed 524288 bytes to 674...
Wrote 524288 bytes (674 compressed) at 0x00080000 in 0.1 seconds (effective 65859.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

It would be nice if this worked out of the box (it took me hours to figure this out and I have no idea if this is actually the correct way of configuring the memory layout). At least esptool recognizes if there's an overlap between the code and the SPIFF.

mikee47, 2021-01-31:
This will all change soon. See #2171

iliis, 2021-02-01:
> At least esptool recognizes if there's an overlap between the code and the SPIFF.

After playing around a bit it seems that this is only the case for `make flashinit`. `make flash` will happily write stuff over other stuff so the module spews out some garbage and crashes :frowning_face:

I think this is because `make flashinit` writes some things (e.g. `esp_init_data_default.bin`) that aren't written during a normal `make flash`, so the esptool doesn't know that there is something there already.

Oh well, let's wait for https://github.com/SmingHub/Sming/pull/2171

mikee47, 2021-02-01:
Feel free to test out #2171 if you like, it's fully functional just needs a bit more documentation...

slaff, 2021-02-01:
@iliis Take a look at this old discussion: https://github.com/SmingHub/Sming/issues/1656#issuecomment-479884834

iliis, 2021-02-01:
> @iliis Take a look at this old discussion: [#1656 (comment)](https://github.com/SmingHub/Sming/issues/1656#issuecomment-479884834)

I've seen this, but unfortunately reducing SPIFF_SIZE alone wasn't enough. Even with just a few hundred bytes, it still gets written to 0x100000.

slaff, 2021-02-04:
> Even with just a few hundred bytes, it still gets written to 0x100000.

It is possible to change the starting address of spiffs using the RBOOT_SPIFFS_0 directive. Before that maybe it is a good idea to understand the location of the different pieces. In 1MB Flash you have the data divided into 256 sectors. One sector is 4K or 0x1000 as hex representation.

| Sector | Start Address | Used for |
|----|---------------|------------------------|
| 0  | 0x000000 | rboot.bin - bootloader |
| 1 | 0x001000 |  rboot-configuration |
| 2 | 0x002000 |  Here starts your firmware .... |
| N | N* 0x001000 |  Here ends your firmware |
| N+1 | (N+1)* 0x001000 |  Here can start your spiffs |
.....
|252+| 0xfb000 | The last 4 sectors (16K) are reserved for Espressif|
| 255 | 0xff000 |  End of 1MB |

What you need to find is `N` from the table above. You can do the following:
1. Compile your application and see how big it is. For example in my environment Basic_Blink is 225184 bytes. You can see this by checking the size of `out/Esp8266/debug/firmware/rom0.bin`
2. The start address where the firmware is written is 0x002000. Adding the 225184 bytes and rounding to the next sector is where the earliest start address of the spiffs can be. Finding N can be done using for example python:
```bash
python -c 'print( (0x002000 + 225184) // (0x1000) )'
```
That is giving 56 and that is your N value. 
The earliest address where you can write your spiffs is (N+1) * 0x001000. Which in our case is 57*0x001000 or as hex representation 0x39000:
```bash
python -c 'print(hex((56+1)*0x1000))'
```

Now you can compile your application by setting the spiffs start address using the following make directives:
```bash
cd <your-application>;
make  RBOOT_SPIFFS_0=0x39000
```

You can check this by running `make flash V=1`. The output will be something like:
```bash
esptool.py -p /dev/ttyUSB0 -b 115200 ... 0x00000 out/Esp8266/debug/firmware/rboot.bin 0x002000 out/Esp8266/debug/firmware/rom0.bin 0x39000 out/Esp8266/debug/firmware/spiff_rom.bin
```

With the time an application grows and recalculate of the start address of the spiffs will be required. A sample one liner that does this is given below:
```bash
python -c "print(hex (( (0x002000 + $( ls -l out/Esp8266/debug/firmware/rom0.bin| awk '{print $5}') ) // (0x1000) +1) * 0x1000 ))"
```

And finally make sure that your spiffs is not overlapping the last 4 sectors.

P.S: We should probably do this automatically in our build system if we detect that the SPI_SIZE is 1M or 512K.

slaff, 2021-03-04:
Should be fixed in the latest `develop` with the introduction of partition tables #2171.

closed 2021-03-04 by slaff

iliis, 2021-03-04:
Sweet, thank you! Also for the detailed and thorough answer above. I unfortunately hadn't had time to work on this in the meantime.

iliis, 2021-06-19:
As far as I can tell, I still have to configure the partition table manually, right?
The following seems to work:
```json
{
    "name": "Project H801wifilight config",
    "base_config": "standard",
    "options": ["spiffs"],
    "partitions": {
        "rom0": {
            "size": "512K"
        },
        "spiffs0": {
            "address": "0x00082000",
            "size": "480K"
        }
    }
}
```

slaff, 2021-06-22:
> still have to configure the partition table manually, right?

Exactly. At some point the ROM size may get bigger than 512K. When that happens you can compile Sming and the application in release mode using [SMING_RELEASE=0](https://sming.readthedocs.io/en/latest/_inc/Sming/index.html#envvar-SMING_RELEASE).