SmingHub/Sming #2213: Conflicting `BYTE_ORDER` definitions

mikee47, 2021-02-01:
When building `HttpServer_FirmwareUpload` sample with esp-quick-toolchain, a `"BYTE_ORDER" redefined` error is raised. CI builds samples with `STRICT=1` which converts this to a warning, however this is unsafe as the definitions are incompatible.

First definition is in `esp-quick-toolchain/xtensa-lx106-elf/xtensa-lx106-elf/include/machine/endian.h`
Conflicting definition is in `Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/include/arch/cc.h`

I'm raising this as an issue as the correct solution requires some thought and may cause other issues.

Original error from build logs:

```
/home/appveyor/projects/sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/include/arch/cc.h:65: warning: "BYTE_ORDER" redefined
   65 | #define BYTE_ORDER LITTLE_ENDIAN
      | 
In file included from /home/appveyor/projects/sming/opt/esp-quick-toolchain/xtensa-lx106-elf/xtensa-lx106-elf/include/sys/types.h:49,
                 from /home/appveyor/projects/sming/opt/esp-quick-toolchain/xtensa-lx106-elf/xtensa-lx106-elf/include/stdio.h:61,
                 from /home/appveyor/projects/sming/Sming/Wiring/FakePgmSpace.h:13,
                 from /home/appveyor/projects/sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/include/arch/cc.h:43,
                 from /home/appveyor/projects/sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/include/lwip/arch.h:43,
                 from /home/appveyor/projects/sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/include/lwip/debug.h:35,
                 from /home/appveyor/projects/sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/include/lwip/opt.h:46,
                 from /home/appveyor/projects/sming/Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/include/lwip/init.h:35,
                 from /home/appveyor/projects/sming/Sming/Arch/Esp8266/Components/esp8266/include/espinc/lwip_includes.h:11,
                 from /home/appveyor/projects/sming/Sming/Arch/Esp8266/System/include/user_config.h:13,
                 from /home/appveyor/projects/sming/Sming/Arch/Esp8266/Components/esp8266/ESP8266_NONOS_SDK/include/osapi.h:30,
                 from /home/appveyor/projects/sming/Sming/Arch/Esp8266/Components/esp8266/include/esp_systemapi.h:40,
                 from /home/appveyor/projects/sming/Sming/Libraries/libsodium/libsodium/src/libsodium/randombytes/sysrandom/randombytes_sysrandom.c:1:
/home/appveyor/projects/sming/opt/esp-quick-toolchain/xtensa-lx106-elf/xtensa-lx106-elf/include/machine/endian.h:27: note: this is the location of the previous definition
   27 | #define BYTE_ORDER _BYTE_ORDER
      | 
AR /home/appveyor/projects/sming/Sming/out/Esp8266/debug/lib/clib-libsodium.a
```

slaff, 2021-03-23:
We can replace the code in `Sming/Arch/Esp8266/Components/esp-open-lwip/esp-open-lwip/include/arch/cc.h` line 65

```C++
#if (1)
#define BYTE_ORDER LITTLE_ENDIAN
#else
#define BYTE_ORDER BIG_ENDIAN
#endif
```

with this one 
```c++
#ifndef BYTE_ORDER
#define BYTE_ORDER LITTLE_ENDIAN
#endif
```

Shall I create a PR?

mikee47, 2021-03-23:
Yes please. I thought it would be more complicated than that, thinking that the definitions used different values. I've taken another look and in boths cases `LITTLE_ENDIAN` and `BIG_ENDIAN` are declared to the same value so we should be fine!

PR #2281 (2021-03-24): Use BYTE_ORDER for LWIP if already defined in SDK.
Closes #2213.
closed 2021-03-24 by slaff