SmingHub/Sming #2214: Code completion and debugging not working in Eclipse CDT with Sming 4.2.0

vesley, 2021-02-03:
Hello Sming team,

I installed few years ago Sming 3 on a Macbook with Eclipse CDT. Code completion and debugging worked very well.
Unfortunately I didn't had time to work on my project.
As I have more time now, I want to continue working on my project. But I don't have my Macbook anymore.
Currently I'm working on Ubuntu 20.04.
I downloaded the [quick toolchain](https://github.com/earlephilhower/esp-quick-toolchain) and checked out latest Sming version from github.
I can build, flash and debug in the shell/console, but I can not debugging in Eclipse (downloaded from [here](https://www.eclipse.org/downloads/packages/)). Also there is no code completion.
I have setup the ENV variables (ESP_HOME and SMING_HOME) in Eclipse.
For the debugging configuration in Eclipse I followed this [tutorial](https://github.com/SmingHub/Sming/wiki/Debugging#debugging-with-visual-debuggers-like-eclipse-cdt).
When I start debugging in Eclipse, the debugger stops in a unknown program and never reaches the break point at my application.app (Basic_Blink).
Did something changed since Sming 3? Can someone point how to configure Eclipse under Linux so that code completion and debugging is working?

Thanks
wesli

mikee47, 2021-02-03:
@vesley Hi there and welcome back! You can chat here https://gitter.im/SmingHub/Sming but it's fine to open an issue.

Since version 4 we have a new documentation portal, see here https://sming.readthedocs.io/en/latest/information/debugging.html for up to date guidance.

Setting up debugging in eclipse is quite fiddly so please let us know how you get on, happy to help.

mikee47, 2021-02-03:
Incidentally, I've been using [VS Code](https://code.visualstudio.com/Download) quite a lot recently - it's a lot more responsive than eclipse and has loads of plugin support. I develop on Windows (with WSL) but it's also available for linux so worth a go!

closed 2021-02-04 by slaff