SmingHub/Sming #2242: Esp32 project with WPS does not compile

joerg65, 2021-03-05:
Hello,
I try to connect my Esp32 board with WPS, but I get compiling errors.
```Building /home/joerg/Development/esp32/Sming/Sming/out/Esp32/debug/lib/clib-sming-arch-218e68e8aa01d3e681db8bb1e09ca42e.a
C+ /home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp
In file included from /home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp:11:
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.h:70:25: error: 'wps_cb_status' has not been declared
  void internalWpsConfig(wps_cb_status status);
                         ^~~~~~~~~~~~~
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp:427:37: error: variable or field 'internalWpsConfig' declared void
 void StationImpl::internalWpsConfig(wps_cb_status status)
                                     ^~~~~~~~~~~~~
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp:427:37: error: 'wps_cb_status' was not declared in this scope
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp:427:37: note: suggested alternative: 'WpsStatus'
 void StationImpl::internalWpsConfig(wps_cb_status status)
                                     ^~~~~~~~~~~~~
                                     WpsStatus
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp: In member function 'virtual bool StationImpl::wpsConfigStart(WPSConfigDelegate)':
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp:465:2: error: 'wifi_station_disconnect' was not declared in this scope
  wifi_station_disconnect();
  ^~~~~~~~~~~~~~~~~~~~~~~
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp:465:2: note: suggested alternative: 'esp_wifi_disconnect'
  wifi_station_disconnect();
  ^~~~~~~~~~~~~~~~~~~~~~~
  esp_wifi_disconnect
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp:466:26: error: 'wifi_get_opmode' was not declared in this scope
  wifi_set_opmode_current(wifi_get_opmode() | STATION_MODE);
                          ^~~~~~~~~~~~~~~
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp:466:26: note: suggested alternative: 'esp_wifi_get_mode'
  wifi_set_opmode_current(wifi_get_opmode() | STATION_MODE);
                          ^~~~~~~~~~~~~~~
                          esp_wifi_get_mode
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp:466:46: error: 'STATION_MODE' was not declared in this scope
  wifi_set_opmode_current(wifi_get_opmode() | STATION_MODE);
                                              ^~~~~~~~~~~~
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp:466:46: note: suggested alternative: 'STAILQ_REMOVE'
  wifi_set_opmode_current(wifi_get_opmode() | STATION_MODE);
                                              ^~~~~~~~~~~~
                                              STAILQ_REMOVE
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp:466:2: error: 'wifi_set_opmode_current' was not declared in this scope
  wifi_set_opmode_current(wifi_get_opmode() | STATION_MODE);
  ^~~~~~~~~~~~~~~~~~~~~~~
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp:466:2: note: suggested alternative: 'wifi_ant_mode_t'
  wifi_set_opmode_current(wifi_get_opmode() | STATION_MODE);
  ^~~~~~~~~~~~~~~~~~~~~~~
  wifi_ant_mode_t
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp:468:22: error: 'WPS_TYPE_PBC' was not declared in this scope
  if(!wifi_wps_enable(WPS_TYPE_PBC)) {
                      ^~~~~~~~~~~~
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp:468:22: note: suggested alternative: 'ICMP6_TYPE_PE4'
  if(!wifi_wps_enable(WPS_TYPE_PBC)) {
                      ^~~~~~~~~~~~
                      ICMP6_TYPE_PE4
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp:468:6: error: 'wifi_wps_enable' was not declared in this scope
  if(!wifi_wps_enable(WPS_TYPE_PBC)) {
      ^~~~~~~~~~~~~~~
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp:468:6: note: suggested alternative: 'wifi_ps_type_t'
  if(!wifi_wps_enable(WPS_TYPE_PBC)) {
      ^~~~~~~~~~~~~~~
      wifi_ps_type_t
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp: In lambda function:
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp:472:65: error: 'wps_cb_status' was not declared in this scope
  if(!wifi_set_wps_cb([](int status) { station.internalWpsConfig(wps_cb_status(status)); })) {
                                                                 ^~~~~~~~~~~~~
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp:472:65: note: suggested alternative: 'WpsStatus'
  if(!wifi_set_wps_cb([](int status) { station.internalWpsConfig(wps_cb_status(status)); })) {
                                                                 ^~~~~~~~~~~~~
                                                                 WpsStatus
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp: In member function 'virtual bool StationImpl::wpsConfigStart(WPSConfigDelegate)':
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp:472:6: error: 'wifi_set_wps_cb' was not declared in this scope
  if(!wifi_set_wps_cb([](int status) { station.internalWpsConfig(wps_cb_status(status)); })) {
      ^~~~~~~~~~~~~~~
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp:472:6: note: suggested alternative: 'wifi_ant_gpio_t'
  if(!wifi_set_wps_cb([](int status) { station.internalWpsConfig(wps_cb_status(status)); })) {
      ^~~~~~~~~~~~~~~
      wifi_ant_gpio_t
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp:477:6: error: 'wifi_wps_start' was not declared in this scope
  if(!wifi_wps_start()) {
      ^~~~~~~~~~~~~~
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp:477:6: note: suggested alternative: 'dhcps_start'
  if(!wifi_wps_start()) {
      ^~~~~~~~~~~~~~
      dhcps_start
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp: In member function 'virtual void StationImpl::wpsConfigStop()':
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp:486:6: error: 'wifi_wps_disable' was not declared in this scope
  if(!wifi_wps_disable()) {
      ^~~~~~~~~~~~~~~~
/home/joerg/Development/esp32/Sming/Sming/Arch/Esp32/Platform/StationImpl.cpp:486:6: note: suggested alternative: 'wifi_csi_cb_t'
  if(!wifi_wps_disable()) {
      ^~~~~~~~~~~~~~~~
      wifi_csi_cb_t
make[2]: *** [/home/joerg/Development/esp32/Sming/Sming/component-wrapper.mk:170: Arch/Esp32/Platform/StationImpl.o] Fehler 1
make[1]: Verzeichnis „/home/joerg/Development/eclipse-workspace/fps-mqtt“ wird verlassen
make[1]: *** [/home/joerg/Development/esp32/Sming/Sming/project.mk:430: sming-arch-build] Fehler 2
make: *** [/home/joerg/Development/esp32/Sming/Sming/project.mk:22: all] Fehler 2
"make SMING_ARCH=Esp32 ENABLE_SSL=1 ENABLE_WPS=1" terminated with exit code 2. Build might be incomplete.
```
In Init I do:
```
	WifiAccessPoint.enable(false);
	WifiStation.enable(true);
	WifiStation.wpsConfigStart(WPSConfigCallback);
```
BTW. 
Without WPS my project compiles o.k.
I give`ENABLE_WPS=1` with make.
I am on devlop branch.

slaff, 2021-03-05:
@joerg65 Your observation is correct. WPS support is still missing for the Esp32 architecture. If your C/C++ skills are good you can try to add it or you should wait for a contributor to help here.

joerg65, 2021-03-05:
I see. I have made this implementation, based on what I have found in the example of esp-idf:
```diff --git a/Sming/Arch/Esp32/Platform/StationImpl.cpp b/Sming/Arch/Esp32/Platform/StationImpl.cpp
index e82ef32d..9ab8d745 100644
--- a/Sming/Arch/Esp32/Platform/StationImpl.cpp
+++ b/Sming/Arch/Esp32/Platform/StationImpl.cpp
@@ -14,6 +14,9 @@
 #include <esp_wifi.h>
 #include <esp_netif.h>
 #include <esp_event.h>
+#ifdef ENABLE_WPS
+#include <esp_wps.h>
+#endif
 
 static StationImpl station;
 StationClass& WifiStation = station;
@@ -423,69 +426,37 @@ void StationImpl::smartConfigStop()
 #endif // ENABLE_SMART_CONFIG
 
 #ifdef ENABLE_WPS
-
-void StationImpl::internalWpsConfig(wps_cb_status status)
-{
-	bool processInternal = true;
-	if(wpsConfigCallback) {
-		processInternal = wpsConfigCallback(WpsStatus(status));
-	}
-	if(processInternal) {
-		switch(status) {
-		case WPS_CB_ST_SUCCESS:
-			debugf("wifi_wps_status_cb(): WPS_CB_ST_SUCCESS\n");
-			wpsConfigStop();
-			connect();
-			break;
-		case WPS_CB_ST_FAILED:
-			debugf("wifi_wps_status_cb(): WPS_CB_ST_FAILED\n");
-			wpsConfigStop();
-			connect(); // try to reconnect with old config
-			break;
-		case WPS_CB_ST_TIMEOUT:
-			debugf("wifi_wps_status_cb(): WPS_CB_ST_TIMEOUT\n");
-			wpsConfigStop();
-			connect(); // try to reconnect with old config
-			break;
-		case WPS_CB_ST_WEP:
-			debugf("wifi_wps_status_cb(): WPS_CB_ST_WEP\n");
-			break;
-		default:
-			debugf("wifi_wps_status_cb(): unknown wps_cb_status %d\n", status);
-			wpsConfigStop();
-			connect(); // try to reconnect with old config
-		}
-	}
-}
-
-bool StationImpl::wpsConfigStart(WPSConfigDelegate callback)
+bool StationImpl::wpsConfigStart()
 {
 	debugf("WPS start\n");
-	wpsConfigCallback = callback;
-	wifi_station_disconnect();
-	wifi_set_opmode_current(wifi_get_opmode() | STATION_MODE);
-	debugf("WPS stationmode activated\n");
-	if(!wifi_wps_enable(WPS_TYPE_PBC)) {
-		debugf("Station::wpsConfigStart() : wps enable failed\n");
-		return false;
-	}
-	if(!wifi_set_wps_cb([](int status) { station.internalWpsConfig(wps_cb_status(status)); })) {
-		debugf("Station::wpsConfigStart() : cb failed\n");
-		return false;
-	}
 
-	if(!wifi_wps_start()) {
-		debugf("Station::wpsConfigStart() : wifi_wps_start() failed\n");
-		return false;
+	esp_wifi_disconnect();
+	wifi_mode_t mode;
+	ESP_ERROR_CHECK(esp_wifi_get_mode(&mode));
+	switch(mode) {
+	case WIFI_MODE_STA:
+	case WIFI_MODE_APSTA:
+		break; // No change required
+	case WIFI_MODE_AP:
+		mode = WIFI_MODE_APSTA;
+		break;
+	case WIFI_MODE_NULL:
+	default:
+		mode = WIFI_MODE_STA;
 	}
+	ESP_ERROR_CHECK(esp_wifi_set_mode(mode));
+	ESP_ERROR_CHECK(esp_wifi_start());
+	debugf("WPS stationmode activated\n");
+	esp_wps_config_t config = WPS_CONFIG_INIT_DEFAULT(WPS_TYPE_PBC);
+	ESP_ERROR_CHECK(esp_wifi_wps_enable(&config));
+	ESP_ERROR_CHECK(esp_wifi_wps_start(60000));
+
 	return true;
 }
 
 void StationImpl::wpsConfigStop()
 {
-	if(!wifi_wps_disable()) {
-		debugf("Station::wpsConfigStop() : wifi_wps_disable() failed\n");
-	}
+	ESP_ERROR_CHECK(esp_wifi_wps_disable());
 }
 
 #endif // ENABLE_WPS
diff --git a/Sming/Arch/Esp32/Platform/StationImpl.h b/Sming/Arch/Esp32/Platform/StationImpl.h
index e7852087..fd7b6898 100644
--- a/Sming/Arch/Esp32/Platform/StationImpl.h
+++ b/Sming/Arch/Esp32/Platform/StationImpl.h
@@ -54,7 +54,7 @@ public:
 #endif
 
 #ifdef ENABLE_WPS
-	bool wpsConfigStart(WPSConfigDelegate callback) override;
+	bool wpsConfigStart() override;
 	void wpsConfigStop() override;
 #endif
 
@@ -67,7 +67,7 @@ private:
 	void internalSmartConfig(sc_status status, void* pdata);
 #endif
 #ifdef ENABLE_WPS
-	void internalWpsConfig(wps_cb_status status);
+	//void internalWpsConfig(wps_cb_status status);
 #endif
 
 private:
```
When I start the WPS process with my FritzBox, pushbutton methode, no PIN, it gives disconnect error 204 after some seconds:
```
I (50) wifi:wifi driver task: 3ffbd4c8, prio:23, stack:6656, core=0
I (677) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (687) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (717) wifi:wifi firmware version: bcf771b
I (717) wifi:config NVS flash: enabled
I (717) wifi:config nano formating: disabled
I (717) wifi:Init data frame dynamic rx buffer num: 32
I (717) wifi:Init management frame dynamic rx buffer num: 32
I (727) wifi:Init management short buffer num: 32
I (727) wifi:Init dynamic tx buffer num: 32
I (727) wifi:Init static rx buffer size: 1600
I (737) wifi:Init static rx buffer num: 10
I (737) wifi:Init dynamic rx buffer num: 32
I (747) wifi_init: rx ba win: 6
I (747) wifi_init: tcpip mbox: 32
I (747) wifi_init: udp mbox: 6
I (757) wifi_init: tcp mbox: 6
I (757) wifi_init: tcp tx win: 5744
I (767) wifi_init: tcp rx win: 5744
I (767) wifi_init: tcp mss: 144ð£+ê768042 WPS start

I (877) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
                                                                    I (877) wifi:mode : sta (ac:67:b2:38:ea:a0)
863028 WPS stationmode activated

863190 event WIFI_EVENT|2
I (2927) wifi:new:<8,0>, old:<1,0>, ap:<255,255>, sta:<8,0>, prof:1
I (2937) wifi:state: init -> auth (b0)
I (2937) wifi:state: auth -> assoc (0)
I (2947) wifi:state: assoc -> run (10)
I (12947) wifi:state: run -> init (cc00)
I (12947) wifi:new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1
12934373 event WIFI_EVENT|5

12934689 disconnect from ssid (hidden), reason 204

120869177 event WIFI_EVENT|9
```
As of sdk it means: `HANDSHAKE_TIMEOUT`.

I will try it with Arduino, to see how it is doing there.

joerg65, 2021-03-05:
Need to implement an event handler ...

joerg65, 2021-03-05:
I have implemented an event handler. What I don't understand is that with this code, only for one time the process worked and the Esp32 was connected to my network. I pressed the connect button on the Fritz Repeater. Later it never connected again, doesn't matter if started WPS by webpage or button at Box or Repeater.
```
diff --git a/Sming/Arch/Esp32/Platform/StationImpl.cpp b/Sming/Arch/Esp32/Platform/StationImpl.cpp
index e82ef32d..ecd0faf1 100644
--- a/Sming/Arch/Esp32/Platform/StationImpl.cpp
+++ b/Sming/Arch/Esp32/Platform/StationImpl.cpp
@@ -14,6 +14,14 @@
 #include <esp_wifi.h>
 #include <esp_netif.h>
 #include <esp_event.h>
+#ifdef ENABLE_WPS
+#include <esp_wps.h>
+#define MAX_RETRY_ATTEMPTS  5
+static wifi_config_t wps_ap_creds[MAX_WPS_AP_CRED];
+static int s_ap_creds_num = 0;
+static int s_retry_num = 0;
+static esp_wps_config_t wps_config = WPS_CONFIG_INIT_DEFAULT(WPS_TYPE_PBC);
+#endif
 
 static StationImpl station;
 StationClass& WifiStation = station;
@@ -424,68 +432,115 @@ void StationImpl::smartConfigStop()
 
 #ifdef ENABLE_WPS
 
-void StationImpl::internalWpsConfig(wps_cb_status status)
+void wifi_event_handler(void* arg, esp_event_base_t event_base,
+                                int32_t event_id, void* event_data)
 {
-	bool processInternal = true;
-	if(wpsConfigCallback) {
-		processInternal = wpsConfigCallback(WpsStatus(status));
-	}
-	if(processInternal) {
-		switch(status) {
-		case WPS_CB_ST_SUCCESS:
-			debugf("wifi_wps_status_cb(): WPS_CB_ST_SUCCESS\n");
-			wpsConfigStop();
-			connect();
-			break;
-		case WPS_CB_ST_FAILED:
-			debugf("wifi_wps_status_cb(): WPS_CB_ST_FAILED\n");
-			wpsConfigStop();
-			connect(); // try to reconnect with old config
-			break;
-		case WPS_CB_ST_TIMEOUT:
-			debugf("wifi_wps_status_cb(): WPS_CB_ST_TIMEOUT\n");
-			wpsConfigStop();
-			connect(); // try to reconnect with old config
-			break;
-		case WPS_CB_ST_WEP:
-			debugf("wifi_wps_status_cb(): WPS_CB_ST_WEP\n");
-			break;
-		default:
-			debugf("wifi_wps_status_cb(): unknown wps_cb_status %d\n", status);
-			wpsConfigStop();
-			connect(); // try to reconnect with old config
-		}
-	}
+
+    static int ap_idx = 1;
+
+    switch (event_id) {
+        case WIFI_EVENT_STA_START:
+        	debugf("WIFI_EVENT_STA_START");
+            break;
+        case WIFI_EVENT_STA_DISCONNECTED:
+        	debugf("WIFI_EVENT_STA_DISCONNECTED");
+            if (s_retry_num < MAX_RETRY_ATTEMPTS) {
+                esp_wifi_connect();
[... 124 lines cut ...]
 #ifdef ENABLE_WPS
-	bool wpsConfigStart(WPSConfigDelegate callback) override;
+	bool wpsConfigStart() override;
 	void wpsConfigStop() override;
 #endif
 
@@ -66,9 +66,6 @@ private:
 #ifdef ENABLE_SMART_CONFIG
 	void internalSmartConfig(sc_status status, void* pdata);
 #endif
-#ifdef ENABLE_WPS
-	void internalWpsConfig(wps_cb_status status);
-#endif
 
 private:
 	bool runScan = false;
```
If someone here can try it with that code?
```
WifiAccessPoint.enable(false);
WifiStation.enable(true);
WifiStation.wpsConfigStart();
```

BTW. there is a conflict with `base64_decode`, fast fix is to rename to `base64_decode_local`  it in `Sming/Components/axtls-8266/axtls-8266/crypto/crypto_misc.c`.

joerg65, 2021-03-06:
So far no success. I tried with the wps-example from esp-idf and this is working. I have a log :
```
I (588) wifi:wifi driver task: 3ffc0bd8, prio:23, stack:6656, core=0
I (588) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (588) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (618) wifi:wifi firmware version: bcf771b
I (618) wifi:config NVS flash: enabled
I (618) wifi:config nano formating: disabled
I (618) wifi:Init data frame dynamic rx buffer num: 32
I (628) wifi:Init management frame dynamic rx buffer num: 32
I (628) wifi:Init management short buffer num: 32
I (638) wifi:Init dynamic tx buffer num: 32
I (638) wifi:Init static rx buffer size: 1600
I (638) wifi:Init static rx buffer num: 10
I (648) wifi:Init dynamic rx buffer num: 32
I (648) wifi_init: rx ba win: 6
I (658) wifi_init: tcpip mbox: 32
I (658) wifi_init: udp mbox: 6
I (658) wifi_init: tcp mbox: 6
I (668) wifi_init: tcp tx win: 5744
I (668) wifi_init: tcp rx win: 5744
I (678) wifi_init: tcp mss: 1440
I (678) wifi_init: WiFi IRAM OP enabled
I (678) wifi_init: WiFi RX IRAM OP enabled
I (778) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (778) wifi:mode : sta (ac:67:b2:38:ea:a0)
I (788) example_wps: start wps...
I (788) example_wps: WIFI_EVENT_STA_START
I (24618) wifi:new:<8,0>, old:<1,0>, ap:<255,255>, sta:<8,0>, prof:1
I (25388) wifi:state: init -> auth (b0)
I (25398) wifi:state: auth -> assoc (0)
I (25408) wifi:state: assoc -> run (10)
I (26988) wifi:state: run -> init (17a0)
I (26988) wifi:new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1
I (26988) example_wps: WIFI_EVENT_STA_DISCONNECTED
I (27118) wifi:new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1
I (27898) wifi:state: init -> auth (b0)
I (27908) wifi:state: auth -> assoc (0)
I (27918) wifi:state: assoc -> run (10)
I (27988) example_wps: WIFI_EVENT_STA_WPS_ER_SUCCESS
I (27988) wifi:state: run -> init (0)
I (27998) wifi:new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1
I (27998) wifi:new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1
I (27998) wifi:state: init -> auth (b0)
I (28028) wifi:state: auth -> assoc (0)
I (28038) wifi:state: assoc -> run (10)
I (28068) wifi:connected with hidden, aid = 11, channel 8, BW20, bssid = dc:39:6f:1e:bb:e3
I (28068) wifi:security: WPA2-PSK, phy: bgn, rssi: -89
I (28138) wifi:pm start, type: 1
I (28238) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (29568) esp_netif_handlers: sta ip: 192.168.1.60, mask: 255.255.255.0, gw: 192.168.1.1
I (29568) example_wps: got ip: 192.168.1.60
```
And from not working implementation:
```
I (50) wifi:wifi driver task: 3ffbda64, prio:23, stack:6656, core=0
I (675) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (685) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (715) wifi:wifi firmware version: bcf771b
I (715) wifi:config NVS flash: enabled
I (715) wifi:config nano formating: disabled
I (715) wifi:Init data frame dynamic rx buffer num: 32
I (715) wifi:Init management frame dynamic rx buffer num: 32
I (725) wifi:Init management short buffer num: 32
I (725) wifi:Init dynamic tx buffer num: 32
I (735) wifi:Init static rx buffer size: 1600
I (735) wifi:Init static rx buffer num: 10
I (735) wifi:Init dynamic rx buffer num: 32
I (745) wifi_init: rx ba win: 6
I (745) wifi_init: tcpip mbox: 32
[... 66 lines cut ...]
I (37895) wifi:state: assoc -> run (10)
I (47895) wifi:state: run -> init (cc00)
I (47895) wifi:new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1
I (47895) wifi:new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1
I (47905) wifi:state: init -> auth (b0)
I (47915) wifi:state: auth -> assoc (0)
I (47925) wifi:state: assoc -> run (10)
I (57925) wifi:state: run -> init (cc00)
I (57935) wifi:new:<8,0>, old:<8,0>, ap:<255,255>, sta:<8,0>, prof:1
57915932 event WIFI_EVENT|5

57916240 disconnect from ssid hidden, reason 204

57919313 WIFI_EVENT_STA_DISCONNECTED
59972208 event WIFI_EVENT|5

59972436 disconnect from ssid hidden, reason 205

59972503 WIFI_EVENT_STA_DISCONNECTED
59972553 Failed to connect!
121149595 event WIFI_EVENT|9

121149736 WIFI_EVENT_STA_WPS_ER_TIMEOUT
```
Note that after `WIFI_EVENT_STA_START` this: `W (1265) wifi:sta_scan: STA is connecting, scan are not allowed!`

slaff, 2021-03-07:
@joerg65 Make a PR with your code changes so far. Make sure to put [WIP] at the start of the name of the PR. This way we can easier look together and change it to something working.

mikee47, 2021-03-07:
OK, I've tidied the code up a bit then modified the `Basic_Wifi` sample, replacing `WifiStation.startScan()` with `WifiStation.wpsConfigStart()`. Here's what happens when I hit `connect` on my router:

```
Sming. Let's do smart things!
I (844) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (844) wifi:mode : softAP (3c:71:bf:59:00:5d)
I (854) wifi:Total power save buffer number: 16
I (854) wifi:Init max length of beacon: 752/752
I (854) wifi:Init max length of beacon: 752/752
I (854) wifi:mode : sta (3c:71:bf:59:00:5c) + softAP (3c:71:bf:59:00:5d)
836331 event WIFI_EVENT|12

859868 event WIFI_EVENT|2

860174 Station IP successfully updated
863326 WPS start

I (874) wifi_init: rx ba win: 6
I (874) wifi_init: tcpip mbox: 32
I (874) wifi_init: udp mbox: 6
I (874) wifi_init: tcp mbox: 6
I (884) wifi_init: tcp tx win: 5744
I (884) wifi_init: tcp rx win: 5744
I (884) wifi_init: tcp mss: 1440
I (884) wifi_init: WiFi IRAM OP enabled
I (884) wifi_init: WiFi RX IRAM OP enabled
867943 WPS stationmode activated

READY!
AP. ip: 192.168.4.1 mac: 3c:71:bf:59:00:5d
I (11674) wifi:ap channel adjust o:7,1 n:4,1
I (11674) wifi:new:<4,1>, old:<7,1>, ap:<4,1>, sta:<4,1>, prof:7
I (12454) wifi:state: init -> auth (b0)
I (12464) wifi:state: auth -> assoc (0)
I (12464) wifi:state: assoc -> run (10)
I (14224) wifi:state: run -> init (c0)
I (14224) wifi:new:<4,0>, old:<4,1>, ap:<4,1>, sta:<4,1>, prof:7
14214075 event WIFI_EVENT|5

14214386 disconnect from ssid HouseHarkonnen, reason 0

Disconnected from "HouseHarkonnen", reason: Unknown reason (0)
14215582 WIFI_EVENT_STA_DISCONNECTED
I (14354) wifi:new:<4,1>, old:<4,0>, ap:<4,1>, sta:<4,1>, prof:7
I (15154) wifi:state: init -> auth (b0)
I (15154) wifi:state: auth -> assoc (0)
I (15164) wifi:state: assoc -> run (10)
15211460 event WIFI_EVENT|7

15211786 WIFI_EVENT_STA_WPS_ER_SUCCESS
I (15224) wifi:state: run -> init (0)
I (15234) wifi:new:<4,0>, old:<4,1>, ap:<4,1>, sta:<4,1>, prof:7
15218286 event WIFI_EVENT|5

15218434 disconnect from ssid HouseHarkonnen, reason 8

Disconnected from "HouseHarkonnen", reason: Disassociated: STA is leaving, or has left
15219106 WIFI_EVENT_STA_DISCONNECTED
I (15654) wifi:new:<4,1>, old:<4,0>, ap:<4,1>, sta:<4,1>, prof:7
I (15664) wifi:state: init -> auth (b0)
I (15664) wifi:state: auth -> assoc (0)
I (15664) wifi:state: assoc -> run (10)
I (15684) wifi:connected with HouseHarkonnen, aid = 5, channel 4, 40U, bssid = ec:08:6b:9c:93:b5
I (15684) wifi:security: WPA2-PSK, phy: bgn, rssi: -64
I (15704) wifi:pm start, type: 1

15686048 event WIFI_EVENT|4

15686336 connect to ssid HouseHarkonnen, channel 4

15688496 event IP_EVENT|0

15688584 ip:192.168.1.171,mask:255.255.255.0,gw:192.168.1.1

I'm CONNECTED to 192.168.1.171
I (15704) esp_netif_handlers: sta ip: 192.168.1.171, mask: 255.255.255.0, gw: 192.168.1.1
I (15734) wifi:AP's beacon interval = 102400 us, DTIM period = 1

```

joerg65, 2021-03-07:
I wonder that you don't have `W (1265) wifi:sta_scan: STA is connecting, scan are not allowed!`. I don't have it with the wps-example of esp-idf, but with my Sming app, also with your latest commits.

mikee47, 2021-03-07:
Have you commented-out the 'startScan' call?

```
// Print available access points
// WifiStation.startScan(listNetworks); // In Sming we can start network scan from init method without additional code

WifiStation.wpsConfigStart([](WpsStatus status) { return true; });
```

joerg65, 2021-03-07:
I have started from MQTT example. As I use since years, but now with Esp32. There is no startScan call. But I can try with now the same Basic_Wifi example as you. Meanwhile I tried again the wps-example from /opt/esp-idf and this is working always I push the connect button of the near repeater, like in your log.

joerg65, 2021-03-07:
OK, when I try the Basic_WiFi, it works!
When I change my app init to:
```
WifiAccessPoint.enable(true);
WifiAccessPoint.config(_F("Sming InternetOfThings"), nullptr, AUTH_OPEN);
WifiAccessPoint.setIP(IpAddress(192, 168, 2, 1));

WifiStation.enable(true);
WifiStation.wpsConfigStart([](WpsStatus status) { return true; });
WifiEvents.onStationGotIP(onConnected);
```
it works too!
And I don't have `scan are not allowed!` warning this way.
So the WifiAccessPoint must be switched on to make WPS work. I haven't thought so.

mikee47, 2021-03-07:
That's an interesting thought! I've tested it by explicitly disabling the AccessPoint (fixing a nullptr bug), but it still works for me.

In `Basic_WiFi`:

```
WifiAccessPoint.enable(false);
// WifiAccessPoint.config(_F("Sming InternetOfThings"), nullptr, AUTH_OPEN);
```

Log shows:

```
I (837) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (847) wifi:mode : sta (3c:71:bf:59:00:5c)

...

15694576 WIFI_EVENT_STA_WPS_ER_SUCCESS
...
16661722 ip:192.168.1.171,mask:255.255.255.0,gw:192.168.1.1

I'm CONNECTED to 192.168.1.171
```

Clearly something else is going on...

mikee47, 2021-03-07:
Searching for `scan are not allowed` warning includes this https://esp32.com/viewtopic.php?t=8788.
The message is inside `components/esp_wifi/lib/esp32/libnet80211.a` so if you're not seeing a problem in your code then it could be an ESP-IDF quirk.

joerg65, 2021-03-08:
I have tried with several settings.
Works:
```
	WifiAccessPoint.enable(false);
	WifiAccessPoint.config(_F("Sming InternetOfThings"), nullptr, AUTH_OPEN);
	WifiAccessPoint.setIP(IpAddress(192, 168, 2, 1));

	WifiStation.enable(true);
	WifiStation.wpsConfigStart(WPSConfigCallback);
	WifiEvents.onStationGotIP(onConnected);
```
Bootloop:
```
	WifiAccessPoint.enable(false);
	//WifiAccessPoint.config(_F("Sming InternetOfThings"), nullptr, AUTH_OPEN);
	WifiAccessPoint.setIP(IpAddress(192, 168, 2, 1));

	WifiStation.enable(true);
	WifiStation.wpsConfigStart(WPSConfigCallback);
	WifiEvents.onStationGotIP(onConnected);
```
Works:
```
	WifiAccessPoint.enable(false);
	WifiAccessPoint.config(_F("Sming InternetOfThings"), nullptr, AUTH_OPEN);
	//WifiAccessPoint.setIP(IpAddress(192, 168, 2, 1));

	WifiStation.enable(true);
	WifiStation.wpsConfigStart(WPSConfigCallback);
	WifiEvents.onStationGotIP(onConnected);
```
Don't work, `scan are not allowed!`:
```
	WifiAccessPoint.enable(false);
	//WifiAccessPoint.config(_F("Sming InternetOfThings"), nullptr, AUTH_OPEN);
	//WifiAccessPoint.setIP(IpAddress(192, 168, 2, 1));

	WifiStation.enable(true);
	WifiStation.wpsConfigStart(WPSConfigCallback);
	WifiEvents.onStationGotIP(onConnected);
```
Don't work, `scan are not allowed!`:
```
	WifiAccessPoint.enable(true);
	//WifiAccessPoint.config(_F("Sming InternetOfThings"), nullptr, AUTH_OPEN);
	//WifiAccessPoint.setIP(IpAddress(192, 168, 2, 1));

	WifiStation.enable(true);
	WifiStation.wpsConfigStart(WPSConfigCallback);
	WifiEvents.onStationGotIP(onConnected);
```
So `WifiAccessPoint.config` is the one that makes it work. But I don't have idea why, also after looking into code of AP.
BTW. I saw already from where does come the warning. I saw also some of conversations about it from Arduino users.

mikee47, 2021-03-08:
Have you pulled the latest changes from this PR? I added a fix for that bootloop (nullptr in setIp)

joerg65, 2021-03-08:
No, but that is not the case. I wanted to show when the warning does show up. Anyway I won't use it this way.

closed 2021-03-10 by slaff