SmingHub/Sming #2246: /opt/sming/Tools/install.sh fails on clang-format-6.0 \

frankdownunder, 2021-03-07:
ubuntu 20.04 doesnt have  clang-format-6.0 but it does have clang-format.

mikee47, 2021-03-07:
Strange. Here's what I get:

```
mike@ATHLONME:~/sandboxes/sming/Sming/Libraries/google-cast/samples/basic$ cat /etc/os-release
NAME="Ubuntu"
VERSION="20.04.2 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.2 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal

mike@ATHLONME:~/sandboxes/sming/Sming/Libraries/google-cast/samples/basic$ apt search clang-format
Sorting... Done
Full Text Search... Done
arcanist-clang-format-linter/focal 0.git20161021-2 all
  clang-format linter for Arcanist

clang-format/focal 1:10.0-50~exp1 amd64
  Tool to format C/C++/Obj-C code

clang-format-10/focal 1:10.0.0-4ubuntu1 amd64
  Tool to format C/C++/Obj-C code

clang-format-11/focal-updates 1:11.0.0-2~ubuntu20.04.1 amd64
  Tool to format C/C++/Obj-C code

clang-format-6.0/focal,now 1:6.0.1-14 amd64 [installed]
  Tool to format C/C++/Obj-C code

clang-format-7/focal 1:7.0.1-12 amd64
  Tool to format C/C++/Obj-C code

clang-format-8/focal 1:8.0.1-9 amd64
  Tool to format C/C++/Obj-C code

clang-format-9/focal 1:9.0.1-12 amd64
  Tool to format C/C++/Obj-C code

mike@ATHLONME:~/sandboxes/sming/Sming/Libraries/google-cast/samples/basic$
```

slaff, 2021-03-07:
> ubuntu 20.04 doesn't have clang-format-6.0

@frankdownunder Try the following command: `apt install clang-format-6.0`

frankdownunder, 2021-03-08:
I'm actually using 20.10 (GROOVY GORILLA). Is that version of ubuntu supported?

mikee47, 2021-03-08:
According to https://packages.ubuntu.com/groovy/allpackages the oldest version is `clang-format-8`. I have a feeling that it produces (slightly) different results to version 6 so probably won't work at the moment. I cannot see `clang-format-6` in the backports library.

frankdownunder, 2021-03-15:
Is there a workaround for this ? What impact would vsn 8 (or better still the latest) have?
I tried an earller version of ubuntu, 16.04,using a minimal docker file, but got a problem with "source" then "sudo"
/opt/sming/Tools/install.sh: line 61: sudo: command not found

mikee47, 2021-03-15:
Version 8 does produce slightly different (arguably slightly better!) results. I've been unsuccessful attempting to find a setting to get version 8 to look like version 6. However, it might be possible to get version 6 to look like version 8.

Ubuntu 16.04 has gcc-8 and clang-format-8 available so yes I think updating to clang-format 8 would be worth doing.

I'll take another look...

slaff, 2021-03-26:
We moved to clang-8 with the latest develop version.

closed 2021-03-26 by slaff