SmingHub/Sming #2253: [Feature request]: Use kconfiglib to manage the different configuration settings

slaff, 2021-03-09:
Sming has multiple configuration settings that allow the framework to be tweaked to our needs. At the moment these configuration settings can be found by looking at the documentation or/and at the component.mk files. It would be better to have easier visual selection of those features. For example running in your application a command like the one below
```
make config
```

will take care to fetch all configuration settings and present a nice visual dialog. 

[image]

The above can be implemented: 
- using python's [Kconfiglib](https://github.com/ulfalizer/Kconfiglib) library which can be included in Sming.
- adding `.config` file to every component providing adjustable parameters. The `.config` file will contain a menu definition with parameters, explanation, default values, etc.
- adding `.config` to sample applications that provide adjustable parameters
- storing and using the final generated config in `$(OUT_BASE)/.config` and integrating it with the existing Sming's configuration system.

mikee47, 2021-03-09:
Should this be in #1712?

slaff, 2021-03-09:
> Should this be in #1712?

I guess we should start working on this :).  At least have a PoC that can be polished later on.

slaff, 2021-03-09:
Closing as duplicate of #1712.

closed 2021-03-09 by slaff