SmingHub/Sming #2256: ScreenOLED_SSD1306 sample fails to build

etmmahi, 2021-03-10:
Hi guys,

I just tried to build the sample ScreenOLED_SSD1306 from the develop branch.
It fails at:
```
.../Sming/Sming/Libraries/Adafruit_SSD1306/Adafruit_SSD1306.cpp: In member function 'void Adafruit_SSD1306::drawFastHLineInternal(int16_t, int16_t, int16_t, uint16_t)':
.../Sming/Sming/Libraries/Adafruit_SSD1306/Adafruit_SSD1306.cpp:635:21: error: ISO C++17 does not allow 'register' storage class specifier [-Werror=register]
  635 |   register uint8_t *pBuf = buffer;
```

It seems the patch for the Adafruit_SSD1306 Library needs to be upgraded to remove all the register keywords.
This change needs to be backported to the lts versions.

Best
Martijn

mikee47, 2021-03-10:
Yes, probably needs fixing. You can build the sample with `make STRICT=1` and the errors will be converted to warnings.

etmmahi, 2021-03-10:
The STRICT=1 makes compilation work again.

However fix is still needed. When I remove all the 'register' keywords, the code also compiles without problems/warnings

slaff, 2021-03-10:
@etmmahi If you want create a PR with the latest `Adafruit_SSD1306` library version and updated sample.

etmmahi, 2021-03-10:
With PR #2259 the sample application ScreenOLED_SSD1306 builds without any issues. pse merge

PR #2259 (2021-03-11): ScreenOLED_SSD1306 sample fails to build #2256
Update patch file to remove 'register' keywords.

Closes #2256.
closed 2021-03-11 by slaff