SmingHub/Sming #2261: Reading Vcc on ESP8266 using system_get_vdd33()

mostorer, 2021-03-12:
On a battery operated ESP8266 module I need to read the power voltage. 
Before start using Sming I was working with Espressif NONOS SDK, getting battery voltage with `system_get_vdd33()`.

As from the SDK documentation, to be able to read the power voltage instead of using the ADC pin, a change within `esp_init_data_default.bin` is needed.
Byte 107 (0 - 127) need to be set to 0xff.

Now, looking around, I have seen that in the past this topic was already discussed, but I have not understood if, at the end, something has been done within the framework to manage in some way this topic.

In Arduino (I really do not use nor know Arduino) seems that you have to put within the sketch this:
`ADC_MODE(ADC_VCC);`, while in Sming ... ?

Now I have edited the file`/Sming/Sming/Arch/Esp8266/Components/esp8266/ESP8266_NONOS_SDK/bin/esp_init_data_default.bin`, that is flashed to the ESP8266 and all is working as expected, but in this way I suppose it will impact all other projects I will create and flash.

Is there a flag that can be configured into the component.mk to instruct the builder that there is the need of a modified esp_init_data_default.bin file ? Could something like RBOOT_TWO_ROMS be added to manage this ?

In case a way is already there, please, just let me know where I can find it.

Thank you in advance.

mikee47, 2021-03-12:
I'd suggest taking of copy of your modified `esp_init_data_default.bin` file and renaming it appropriately, e.g. `powervdd.bin`. You can then provide a custom hardware config for your project to use it. See https://sming.readthedocs.io/en/latest/_inc/Sming/Components/Storage/index.html#custom-configurations for information about creating custom configs.

Your custom config might look something like this:

```
{
	"name": "Customised VDD init",
	"comment": "Override the default RF initialisation config to support reading of VDD",
	"base_config": "standard", // Or whatever suits
	"partitions": {
		"phy_init": {
			"filename": "powervdd.bin" // Relative path to your project directory
		}
	}
}
```

mikee47, 2021-03-12:
The new initdata will be used when you run a `make flash`. If you need to flash the initdata separately, do this:

```
make flashpart PART=phy_init
```

mostorer, 2021-03-12:
> I'd suggest taking of copy of your modified `esp_init_data_default.bin` file and renaming it appropriately, e.g. `powervdd.bin`. You can then provide a custom hardware config for your project to use [...]

Working really well. Thank you so much @mikee47

mikee47, 2021-03-12:
The power of partition tables :-)

mostorer, 2021-03-12:
Clarified as to leverage on Sming power.

closed 2021-03-12 by mostorer

slaff, 2021-03-15:
@mostorer With the merge of the latest PRs it should be [easier now to enable vdd support](https://sming.readthedocs.io/en/latest/information/tips-n-tricks.html#reading-vcc-on-esp8266) in your application.