SmingHub/Sming #2275: Compiler command on FreeBSD

kmihaylov, 2021-03-22:
Hello, on FreeBSD they use the GCC version as a suffix, e.g. gcc9. When trying to build some of the samples I stuck here:
```
Building /opt/Sming/Sming/out/Esp8266/debug/tools/esptool2 /opt/Sming/Sming/out/Esp8266/debug/lib/clib-rboot-MD5.a
CC esptool2.c
gcc -O2 -Wall -std=c99 -c esptool2.c -o /opt/Sming/Sming/out/Esp8266/debug/build/rboot/rboot-MD5/esptool2/esptool2.o
gmake[3]: gcc: No such file or directory
```

I tried ```export COMPILE=gcc9``` or ```COMPILE=gcc9 gmake``` w/o success. Would you please point me how to instruct (g)make for the proper compiler command?

mikee47, 2021-03-22:
Linux does the same but there should be links (see below) created either as part of the install or using a tool such as `update-alternatives`. Unfortunately I don't have a FreeBSD system to check this on...

```
/usr/bin
lrwxrwxrwx  1 root   root           5 Mar 20  2020  cpp -> cpp-9
lrwxrwxrwx  1 root   root          22 Aug  8  2020  cpp-9 -> x86_64-linux-gnu-cpp-9
lrwxrwxrwx  1 root   root           5 Mar 20  2020  g++ -> g++-9
lrwxrwxrwx  1 root   root          22 Aug  8  2020  g++-9 -> x86_64-linux-gnu-g++-9
lrwxrwxrwx  1 root   root           5 Mar 20  2020  gcc -> gcc-9
lrwxrwxrwx  1 root   root          22 Aug  8  2020  gcc-9 -> x86_64-linux-gnu-gcc-9
lrwxrwxrwx  1 root   root           8 Mar 20  2020  gcc-ar -> gcc-ar-9
lrwxrwxrwx  1 root   root          25 Aug  8  2020  gcc-ar-9 -> x86_64-linux-gnu-gcc-ar-9
lrwxrwxrwx  1 root   root           8 Mar 20  2020  gcc-nm -> gcc-nm-9
lrwxrwxrwx  1 root   root          25 Aug  8  2020  gcc-nm-9 -> x86_64-linux-gnu-gcc-nm-9
lrwxrwxrwx  1 root   root          12 Mar 20  2020  gcc-ranlib -> gcc-ranlib-9
lrwxrwxrwx  1 root   root          29 Aug  8  2020  gcc-ranlib-9 -> x86_64-linux-gnu-gcc-ranlib-9
-rwxr-xr-x  1 root   root        3477 Aug  4  2020  gcore
lrwxrwxrwx  1 root   root           6 Mar 20  2020  gcov -> gcov-9
lrwxrwxrwx  1 root   root          23 Aug  8  2020  gcov-9 -> x86_64-linux-gnu-gcov-9
lrwxrwxrwx  1 root   root          11 Mar 20  2020  gcov-dump -> gcov-dump-9
lrwxrwxrwx  1 root   root          28 Aug  8  2020  gcov-dump-9 -> x86_64-linux-gnu-gcov-dump-9
lrwxrwxrwx  1 root   root          11 Mar 20  2020  gcov-tool -> gcov-tool-9
lrwxrwxrwx  1 root   root          28 Aug  8  2020  gcov-tool-9 -> x86_64-linux-gnu-gcov-tool-9
```

slaff, 2021-03-22:
> I tried export COMPILE=gcc9 or COMPILE=gcc9 gmake w/o success. 

Try export CC=gcc9 and  CXX=gcc9++. Or whatever the name of the C++ compiler is. You can also test it from a sample application by passing the compilers as arguments to make. Example:
```
cd $SMING_HOME/../samples/Basic_Blink
make CC=gcc9 CXX=gcc9++
```

kmihaylov, 2021-03-22:
@mikee47 
I thought that ```update-alternatives``` tool would be best on FreeBSD, but it doesn't exist. https://forums.freebsd.org/threads/linux-update-alternatives-equivalent-on-freebsd.79316/

I'll go with symlink for now, however I believe it would be better if ```make``` could be instructed for the proper compiler command.
PS: I did the link and the sample compiles fine.

@slaff 
I think I tried that (I passed the parameters in front of ```gmake``` not after it), w/o success (I'll re-check it again in a while). Also I think ```gcc``` is needed only once, at initializing Sming tools, since after that the xtensa compiler is used for the actual work. Give me few tens of minutes to figure it out and test my build system on FreeBSD 12.2 since it took me some 2 weeks trying to static-build EQT's binutins (w/o success, I omitted the LDFLAGS=-static).

https://github.com/earlephilhower/esp-quick-toolchain/issues/28

mikee47, 2021-03-22:
I don't suppose the MacOS/darwin tools work on clean FreeBSD ?

kmihaylov, 2021-03-22:
Although MacOS is a fork of FreeBSD I think the tools and also some system internals are different, for example on FreeBSD the USB serial is ```/dev/ttyU0```, but on MacOS (judging from the documentation) is ```/dev/tty.usbserial```. Therefore I think the differences are more than expected.

I'm not aware with XCode and Homebrew. On FreeBSD I use binary packages. I'll try to make a quick PR for the docs.

With ESP8266 Sming and FreeBSD are good together, but I'm scared of what would be with esp-idf. :(

mikee47, 2021-03-22:
> Also I think `gcc` is needed only once, at initializing Sming tools, since after that the xtensa compiler is used for the actual work.

It's also required for Host emulator builds - I consider this an essential part of my development and testing process :-)

May I suggest (for now at least) simply editing the `Sming/Arch/Host/build.mk` file and adjust the tool specs. as required?

kmihaylov, 2021-03-22:
Thanks for the suggestion. I don't use the Host arch, but if in need I'll know the place!

kmihaylov, 2021-03-22:
I'm closing this as insignificant issue.

closed 2021-03-22 by kmihaylov