SmingHub/Sming #2276: Latest develop: Exception SMTP sample

bolzenwolf, 2021-03-22:
Hi together,

I'm currently facing some problems running Sming SMTP sample.
It compiles okay, but always runs into exception after sending the test mail.
`client.send(mail)`

Please see attachement: [stack-trace.txt](https://github.com/SmingHub/Sming/files/6184775/stack-trace.txt)

Does anybody has an idea what may cause this exception?

slaff, 2021-03-23:
Paste here also the decoded stacktrace `make decode-stacktrace`.

bolzenwolf, 2021-03-23:
Sure, forgot to do so.

Attached decoded stacktrace:
```
0x4022fba0: system_partition_get_item at ??:?
0x40101096: malloc at ??:?
0x40101053: pvPortRealloc at ??:?
0x402399a2: MemoryDataStream::write(unsigned char const*, unsigned int) at /opt/sming/Sming/Core/Data/Stream/MemoryDataStream.cpp:57
 (inlined by) MemoryDataStream::write(unsigned char const*, unsigned int) at /opt/sming/Sming/Core/Data/Stream/MemoryDataStream.cpp:48
0x402384f6: TcpClient::send(char const*, unsigned short, bool) at /opt/sming/Sming/Core/Network/TcpClient.cpp:68
0x402348fe: String::reserve(unsigned int) at /opt/sming/Sming/Wiring/WString.cpp:199
 (inlined by) String::reserve(unsigned int) at /opt/sming/Sming/Wiring/WString.cpp:173
0x4022fba0: system_partition_get_item at ??:?
0x40234a2a: String::concat(char const*, unsigned int) at /opt/sming/Sming/Wiring/WString.cpp:383
0x402348fe: String::reserve(unsigned int) at /opt/sming/Sming/Wiring/WString.cpp:199
 (inlined by) String::reserve(unsigned int) at /opt/sming/Sming/Wiring/WString.cpp:173
0x40234e78: String::concat(char const*, unsigned int) at /opt/sming/Sming/Wiring/WString.cpp:389
0x40236bdc: TcpClient::sendString(String const&, bool) at /opt/sming/Sming/Core/Network/TcpClient.h:108
0x402381c3: String::~String() at /opt/sming/Sming/Wiring/WString.h:207
 (inlined by) StringSumHelper::~StringSumHelper() at /opt/sming/Sming/Wiring/WString.h:878
 (inlined by) SmtpClient::onReadyToSendData(TcpConnectionEvent) at /opt/sming/Sming/Core/Network/SmtpClient.cpp:193
0x40234a2a: String::concat(char const*, unsigned int) at /opt/sming/Sming/Wiring/WString.cpp:383
0x402348fe: String::reserve(unsigned int) at /opt/sming/Sming/Wiring/WString.cpp:199
 (inlined by) String::reserve(unsigned int) at /opt/sming/Sming/Wiring/WString.cpp:173
0x40238578: TcpClient::pushAsyncPart() at /opt/sming/Sming/Core/Network/TcpClient.cpp:163
0x401010af: free at ??:?
0x4023870c: TcpClient::onReadyToSendData(TcpConnectionEvent) at /opt/sming/Sming/Core/Network/TcpClient.cpp:136
0x40234878: String::invalidate() at /opt/sming/Sming/Wiring/WString.cpp:161
0x40236bdc: TcpClient::sendString(String const&, bool) at /opt/sming/Sming/Core/Network/TcpClient.h:108
0x402383d2: SmtpClient::onReadyToSendData(TcpConnectionEvent) at /opt/sming/Sming/Core/Network/SmtpClient.cpp:274
0x40234a2a: String::concat(char const*, unsigned int) at /opt/sming/Sming/Wiring/WString.cpp:383
0x402348fe: String::reserve(unsigned int) at /opt/sming/Sming/Wiring/WString.cpp:199
 (inlined by) String::reserve(unsigned int) at /opt/sming/Sming/Wiring/WString.cpp:173
0x40238578: TcpClient::pushAsyncPart() at /opt/sming/Sming/Core/Network/TcpClient.cpp:163
0x401010af: free at ??:?
0x4023870c: TcpClient::onReadyToSendData(TcpConnectionEvent) at /opt/sming/Sming/Core/Network/TcpClient.cpp:136
0x40234878: String::invalidate() at /opt/sming/Sming/Wiring/WString.cpp:161
0x40236bdc: TcpClient::sendString(String const&, bool) at /opt/sming/Sming/Core/Network/TcpClient.h:108
0x402383d2: SmtpClient::onReadyToSendData(TcpConnectionEvent) at /opt/sming/Sming/Core/Network/SmtpClient.cpp:274
0x40234a2a: String::concat(char const*, unsigned int) at /opt/sming/Sming/Wiring/WString.cpp:383
0x402348fe: String::reserve(unsigned int) at /opt/sming/Sming/Wiring/WString.cpp:199
 (inlined by) String::reserve(unsigned int) at /opt/sming/Sming/Wiring/WString.cpp:173
0x40238578: TcpClient::pushAsyncPart() at /opt/sming/Sming/Core/Network/TcpClient.cpp:163
0x401010af: free at ??:?
0x4023870c: TcpClient::onReadyToSendData(TcpConnectionEvent) at /opt/sming/Sming/Core/Network/TcpClient.cpp:136
0x40234878: String::invalidate() at /opt/sming/Sming/Wiring/WString.cpp:161
0x40236bdc: TcpClient::sendString(String const&, bool) at /opt/sming/Sming/Core/Network/TcpClient.h:108
0x402383d2: SmtpClient::onReadyToSendData(TcpConnectionEvent) at /opt/sming/Sming/Core/Network/SmtpClient.cpp:274
0x40234a2a: String::concat(char const*, unsigned int) at /opt/sming/Sming/Wiring/WString.cpp:383
0x402348fe: String::reserve(unsigned int) at /opt/sming/Sming/Wiring/WString.cpp:199
 (inlined by) String::reserve(unsigned int) at /opt/sming/Sming/Wiring/WString.cpp:173
0x40238578: TcpClient::pushAsyncPart() at /opt/sming/Sming/Core/Network/TcpClient.cpp:163
0x401010af: free at ??:?
0x4023870c: TcpClient::onReadyToSendData(TcpConnectionEvent) at /opt/sming/Sming/Core/Network/TcpClient.cpp:136
0x40234878: String::invalidate() at /opt/sming/Sming/Wiring/WString.cpp:161
0x40236bdc: TcpClient::sendString(String const&, bool) at /opt/sming/Sming/Core/Network/TcpClient.h:108
0x402383d2: SmtpClient::onReadyToSendData(TcpConnectionEvent) at /opt/sming/Sming/Core/Network/SmtpClient.cpp:274
0x40234a2a: String::concat(char const*, unsigned int) at /opt/sming/Sming/Wiring/WString.cpp:383
0x402348fe: String::reserve(unsigned int) at /opt/sming/Sming/Wiring/WString.cpp:199
 (inlined by) String::reserve(unsigned int) at /opt/sming/Sming/Wiring/WString.cpp:173
0x40238578: TcpClient::pushAsyncPart() at /opt/sming/Sming/Core/Network/TcpClient.cpp:163
0x401010af: free at ??:?
0x4023870c: TcpClient::onReadyToSendData(TcpConnectionEvent) at /opt/sming/Sming/Core/Network/TcpClient.cpp:136
0x40234878: String::invalidate() at /opt/sming/Sming/Wiring/WString.cpp:161
0x40236bdc: TcpClient::sendString(String const&, bool) at /opt/sming/Sming/Core/Network/TcpClient.h:108
0x402383d2: SmtpClient::onReadyToSendData(TcpConnectionEvent) at /opt/sming/Sming/Core/Network/SmtpClient.cpp:274
[... cut ...]
0x40238578: TcpClient::pushAsyncPart() at /opt/sming/Sming/Core/Network/TcpClient.cpp:163
0x401010af: free at ??:?
0x4023870c: TcpClient::onReadyToSendData(TcpConnectionEvent) at /opt/sming/Sming/Core/Network/TcpClient.cpp:136
0x40234878: String::invalidate() at /opt/sming/Sming/Wiring/WString.cpp:161
0x40236bdc: TcpClient::sendString(String const&, bool) at /opt/sming/Sming/Core/Network/TcpClient.h:108
0x402383d2: SmtpClient::onReadyToSendData(TcpConnectionEvent) at /opt/sming/Sming/Core/Network/SmtpClient.cpp:274
```

slaff, 2021-03-25:
@bolzenwolf The reported problem should be fixed now. Can you check if it is working for you?

Get the latest `develop` version, recompile the library and your application and try again. On Linux you can use the following commands:
```
cd $SMING_HOME
git checkout develop
git pull
make dist-clean
cd ../samples/SmtpClient
make
```

bolzenwolf, 2021-03-25:
Works for me.
Thanks to both of you.

closed 2021-03-25 by bolzenwolf